helgeerbe/picframe #403: Idea for a new feature

meirmark, 2024-09-09:
It might be an interesting feature if the frame could display a certain photo on a certain date (nice for birthdays, anniversaries, etc.)
For example, if the software checks every time the date is changed to a new day if there is a file for that date and displays it.
The file name contains the date for example #1002.jpg
Will be displayed 2nd October.

Meir.

meirmark, 2024-10-03:
Hi all.
I (with Paddy's help) added an addition to the picframe app.
Now under the pictures directory there are some added directories:
Mein (holds the pictures that are displayed in a normal way)

More directories with names that consist of four digits that represent day and month (for example 0510 represents the date of the fifth of October)
Every day at 00:01 the date is checked and if there is a directory that matches the date then only the pictures from that directory are displayed for the next 24 hours.
this addition is grate for Special events such as birthdays, holidays, and wedding anniversaries, etc.
Note: no pictures under pictures directory.

meirmark, 2024-10-03:
This is a Python script (Meir_check_date.py) for checking the if there is a directory that mech the date, I run it every night at 00:01 using a line at crontab. 
=================================================================
import datetime
import os
from urllib import request

ROOT_DIR = "/home/pi/Pictures" # could make a sub subdirectory at extra coomplication
URL = "http://localhost:9000"

date_now = datetime.datetime.now()
subdirectory = f"{date_now.month:02}{date_now.day:02}"
full_url = f"{URL}?subdirectory="
if subdirectory in os.listdir(ROOT_DIR): # there is no verification or error catching here!!
    full_url += subdirectory
request.urlopen(full_url) # set back to all only works with the mod I will add to the dev branch
with open("/home/pi/picframe_data/PicFrame_log.txt", "a") as log_file:
    log_file.write(f"directory:{full_url}\n")
==================================================================

The line added to crontab is:
01 00 * * * python3 /home/pi/picframe_data/Meir_check_date.py

===================================================================

You need also to change the line:   subdirectory: ""     to    subdirectory: "main"   in the file configuration.yaml so the every day pictures will be display from the main directory.

sapnho, 2024-11-14:
Full directions here: https://www.thedigitalpictureframe.com/celebrate-birthdays-and-anniversaries-with-auto-themed-photos-on-your-digital-frame/

meirmark, 2024-11-14:
Very nice, Thank You.