helgeerbe/picframe #405: PI5 running at 86C because of the 100% CPU

sdegner999, 2024-09-22:
Based on Wolfgang's article (2024 Edition), I installed Picframe. It worked great, and I was so happy until I discovered the PI5 was running at 86C because of the 100% CPU. That’s a show-stopper for me. 
He mentioned this issue in his article.  Have you ever found a solution or workaround for PI5 CPU spiking?  
Thanks, Steve

sapnho, 2024-09-23:
Hi @paddywwoof , do you think there is any way of running the Pi5 using X Server one day? 

Not being able to turn off the monitor costs a lot of energy which most people don't want. Do you have a guess what the problem of the CPU utilisation could be and how it could be fixed? 

Thanks, Wolfgang

paddywwoof, 2024-09-23:
Wolfgang, I think the general feeling in the linux desktop community is that wayland is the way forward and X has had its day. However that talk has been going on for many years so I think it won't happen soon. I imagine there are various utilities that have been written to allow wayland to cope with X server instructions, as a kind of wrapper. For some reason part of what we are doing with picframe is causing something to run flat out, it seems likely to be something to do with one of the wrapper utilities.

If I could run picframe on my RPi5 (Someday I will find a beefier power supply) then I might have a go at finding out what function is running to use up all the CPU, top might be a start. But I can't believe we're the only people using X server emulation for something and finding high CPU usage, so maybe some feedback will find its way to the maintainers of wayland and they might fix it. Alternatively we could try asking on a wayland forum!

The other hope is that we can find a way to turn the screen off with wayland. That seems such an obvious requirement that I feel it must be possible. The odd thing we do is run wayland without the normal desktop, and that seems to prevent the session re-starting itself (we *can* in fact turn the screen off, just not back on again!). So I wonder if it's worth trying to set up picframe with the full desktop and accept that there's a bit more load and some unwanted intermediate desktop views while loading?

Paddy

sapnho, 2024-09-23:
Thanks, @paddywwoof. Yes, betting on the new technology would definitely be my preferred option, and as you say, it's a basic requirement that should work at some point.

Will keep my eyes open for anything to pop up in forums!

paddywwoof, 2024-09-23:
@sdegner999 if you try switching to wayland window manager by just upgrading picframe as I suggested, you will need to update you configuration.yaml file. The `use_sdl2=True` is mentioned in @helgeerbe's instructions so you should catch it!

sdegner999, 2024-09-23:
Excellent, thank you Paddy for the help and fantastic software. I will give it a try. I'll probably use a smart plug to power down the monitor and PI5 during off hours for now.   I didn't find a wake-and-sleep schedule in the config.  I assume turning the monitor off/on is done with MQTT.  Thank you, Wolfgang, for your help and brilliant site.  I'm excited to get it working.

sdegner999, 2024-09-23:
Solved! The temperature is 52C, and the CPU is less than 1%. Thanks so much. Wifey is happy, too. Now, on to load about 3000 pictures.

paddywwoof, 2024-09-23:
Ah, well done. I use HTTP to turn the screen off and on (actually I've had to swap to pausing and turning brightness down and up) see https://github.com/helgeerbe/picframe/issues/321

It would be possible to use the same code (with an extra couple of lines to output to gpio) to turn a relay on and off. Solid state relays are quite cheap and can be turned on and off with RPi gpio. I don't know what the power draw, inductance etc of your screen, but you might be able to just use one directly (I've used them to switch 3kW heating loads and had to use an  extra contactor to stop the stop the solid state relays cutting out).

Paddy

sapnho, 2024-10-08:
@paddywwoof I got this solution on the RPI forum: https://forums.raspberrypi.com/viewtopic.php?t=374678&e=1&view=unread#unread

Can't try it at the moment as I don't have a spare external monitor but maybe someone here can. Curious to know if it works.

paddywwoof, 2024-10-09:
I thought I would try turning the screen on and off with the full desktop system. Installed the default 32 bit OS and a python script seems to blank and unblank it. However when I moved on to try installing picframe etc. following https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland I find that picframe fails to install because of an error setting up pyheif. Also the default numpy doesn't seem to work without `sudo apt install libgfortran5 libopenblas0-pthread` (non of the heif related python wheels say they need an `.so` library installing so don't know how to fix that yet)
I don't know if this is just with 32 bit version of the OS but I noticed that if you buy a pre-loaded SD card that's the version you get.
I'm away from home at the moment but I will continue the experiment when I get back tomorrow. All very frustrating!

Paddy

paddywwoof, 2024-10-13:
OK, I seem to have this working on RPi5 using 64bit desktop bookworm with low CPU work load, as expected.

I basically used @helgeerbe instructions as ref above but **critically** had to comment out the line in the bash script `start_picframe.sh` to enforce wayland `#export SDL_VIDEODRIVER=wayland # sdl has to use wayland
` I put lines in `viewer_display.py`
```python
    @display_is_on.setter
    def display_is_on(self, on_off):
        self.__logger.debug("Switch display (display_power=%d).", self.__display_power)
        wlr_randr = ["wlr-randr", "--output", "HDMI-A-1"]
        if on_off:
            wlr_randr.append("--on")
        else:
            wlr_randr.append("--off")
        subprocess.call(wlr_randr)
```
and I modified `pi3d/util/DisplayOpenGL.py` to make SDL2 use a desktop fullscreen rather than just fullscreen (not sure what that means!)
```python
      if (w == self.width and h == self.height) or (self.display_config & DISPLAY_CONFIG_FULLSCREEN):
        sdl2.SDL_SetWindowFullscreen(self.window, sdl2.SDL_WINDOW_FULLSCREEN_DESKTOP)
```
I'm not sure that the pi3d mod is needed as I did this before turning off the wayland SDL_VIDEODRIVER. I will check this more thoroughly, write up the full instructions and put the mods into the repositories (and update pi3d on PYPI if it turns out that step was needed)

The weird thing was that I could turn the wayland version on and off providing I started picframe (or a super simple pi3d application) in a terminal manually. If I tried to do it with a bash script or automatically after boot then I could turn the screen off, but that made the SDL2 application crash and it (obviously) couldn't be turned back on.

Paddy

sapnho, 2024-10-13:
@paddywwoof That would be a very nice breakthrough! I will try it out right away once you are finished. Thanks!

paddywwoof, 2024-10-13:
Good. This seems to be working... But please do check, I might have missed something or just happened upon a quirk that made it work for me but not anybody else!!

1. I followed @helgeerbe instruction from https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland
1.1 but I used the 64bit full desktop
```
Raspberry Pi OS with desktop
Release date: July 4th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)
Size: 1,142MB
Show SHA256 file integrity hash:
[Release notes](https://downloads.raspberrypi.com/raspios_arm64/release_notes.txt)
```
1.2 in `raspi-config` leave the booting as auto login but to the desktop **not** command line (I think it's default)

1.3 python3-venv git wayfire wlr-randr are already installed but you need to `sudo apt install libsdl2-dev` (doesn't really matter if you include the other libraries)

1.4 at the moment the changes are only on github and not uploaded to pypi.org so the line for installing picframe becomes
```
pip install git+https://github.com/helgeerbe/picframe
```
NB still use the venv_picframe as in the instuctions
NB you don't need to do the uninstall and re-install steps as you installed from github in the first place

1.5 you need to edit `configuration.yaml` and change at least two lines
```yaml
  display_power: 2                        # default=0. choices={0, 1}, ...
  use_glx: False                          # default=False. ...
  use_sdl2: True                         # default=True. ... (default is actually False!)
```
I also set `  use_http: True` as I used that for testing and for turning the screen on and off from other python apps.

1.6 **critically** `wlr-randr` doesn't cope with the specific way we use the display if SDL_VIDEODRIVER uses wayland drivers, so the `start_picframe.sh` file needs to look like this
```bash
#!/bin/bash
source /home/pi/venv_picframe/bin/activate # activate phyton virtual env
picframe &  #start picframe
```

1.7 to get picframe running as soon as the desktop loads you need to edit the existing `.config/wayfire.ini` and simple add on the end
```ini
[autostart]
picframe = /home/pi/start_picframe.sh
```

1.8 the desktop starts automatically after boot so you don't need to use systemd, which might seem like an advantage but it means I don't know of a way to start and stop picframe over SSH apart from `sudo reboot`!

PS @helgeerbe when I ran picframe just now I got a weird warning message from the logger
` WARNING:viewer_display.ViewerDisplay:Unsupported setting for display_power=2.` which is caused by the else clause following the elif that is certainly being run as the screen goes off and on! See https://github.com/helgeerbe/picframe/blob/0e2c8eb362ed4555ece40e82b2dc679cb782a6c5/src/picframe/viewer_display.py#L157

paddywwoof, 2024-10-13:
PPS
@sdegner999 I've not tried it yet (next job!) but you might find the changes above also work with the lite version of the OS. i.e. you could try:
```bash
cd ~
source venv_picframe/bin/activate
pip install --upgrade git+https://github.com/helgeerbe/picframe
pip install --upgrade pi3d
```
and comment out the line in `start_picframe.sh` that forces `SDL_VIDEODRIVER=wayland`

If it works you would then be able to turn the monitor on and off with MQTT or HTTP

sapnho, 2024-10-13:
Hi @paddywwoof Just followed yours and Helge's instructions. Picframe starts, shows one image and then I get this 
[image]

paddywwoof, 2024-10-13:
That doesn't sound good. I wonder why it shows one image, very odd! Does the login popup happen when the next image would be starting to load? If you set the time between images to something very long, can you toggle the screen on and off OK with HTTP or MQTT? If you run top in an ssh terminal does the CPU or memory look odd when the crash happens? Is it always the same image first? To be honest I can't think of anything in these mods that would make it do this!

sdegner999, 2024-10-13:
Wolfgang and Paddy,

I hesitate to give solutions, but I had that problem and solved it by toggling the Console Autologin and Desktop Autologin, in raspi-config

I just did it again to test it and thought I broke permanently because, One, I don't know what I'm doing and Two, I got the same login screen and then couldn't get it back.  I did get it back!

I broke it to boot to login by selecting. 
sudo raspi-config
1 System Options -> S5 Boot -> B5 Desktop Autologin as 'steve' user  (this broke it to the login screen after reboot)
But when I selected the same line again, to bring it back, it didn't fix it. (fear and panic)  

Then I selected
sudo raspi-config
1 System Options -> S5 Boot -> B2 Console Autologin as 'steve' user
It worked.  So I'm not sure what the secret sequence is.  It doesn't appear to give an option when I select the line. I'm almost too afraid to attempt it again.

I'm loving the frame and will attempt the new install.  I'm getting another SD card so I can keep this one protected.  I keep snapping pictures of my pictures. The matting and side-by-side pictures is phenomenal.  Let me know how I can show my appreciation and contribute to the project. I’ve been looking for something like this for years.

Thanks,
Steve

paddywwoof, 2024-10-14:
Steve, that sounds quite plausible, and the kind of quirky behaviour this seems to be. Worth trying anyway.

I have had that popup appear myself when I've been experimenting. Generally it's been when I've gone into a different full screen terminal (ie CtrlAltF2) to stop the picframe process, then tried to get back to the desktop by whatever means. Usually typing wayfire or wayfire-pi. None of which sounds relevant to Wolfgang's problem.

Thanks for you feedback, fingers crossed...

Paddy

sdegner999, 2024-10-18:
Paddy.

HTTP display is_on didn't work for me.  That's ok. I'm using a smart plug to shut everything down. I just wanted to see if the new method would work for me.  To be fair if I power down the monitor by hand and then start it, the screen remains dark.  This seems like a problem with how Raspberry5 works with HDMI.

I installed picframe as described 5 days ago on another SD card. 
pip install git+https://github.com/helgeerbe/picframe

I used the HTTP website and clicked display is_on, it turns off the monitor and the box goes orange. A reboot is required to get it back.

I'm adding pictures to the frame daily.  Too much fun! 

Thanks,
Steve

paddywwoof, 2024-10-18:
That's **_annoying_**! I wonder what I did differently that made it work. At least you didn't get the login popup after the first image like Wolfgang did.

Just to double check: You did install the standard 64bit Desktop version of the OS, i.e. the desktop shows for a second or two before picframe? Also if you look in pi3d DisplayOpenGL file i.e.
`nano venv_picframe/lib/python3.11/site-packages/pi3d/util/DisplayOpenGL.py`
line 209 has the _DESKTOP constant i.e.
```python
        sdl2.SDL_SetWindowFullscreen(self.window, sdl2.SDL_WINDOW_FULLSCREEN_DESKTOP)
```
Do you have any useful message if you run `journalctl` then go to the end (End key)?

On another related issue, I am trying writing things to `/dev/shm/clock.txt` (i.e. reading emails and posting up messages as well as adding to the Pictures directory). I seem to have a memory leak, and eventually I see errors in journalctl `[drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from GEM DMA helper:` I did come across a [drm:vc4 error before](https://forums.raspberrypi.com/viewtopic.php?p=2176624&hilit=drm%3Avc4#p2176624) bit of a nightmare keeping track of and freeing GPU allocated buffers.

helgeerbe, 2024-10-22:
I tried this with the light version. It does not work.

@paddywwoof How does your `wayfire.ini` from the desktop version looks like ?

paddywwoof, 2024-10-22:
Hi, I think I tried with the lite version previously.

I'm away from the RPi for a couple of days but will do a fresh install and write everything down.

helgeerbe, 2024-10-22:
Thanks. I start alacritty and picframe from console. If  I turn off, sleep 15, turn on the screen. Only Alacritty will survive.

helgeerbe, 2024-10-24:
OK, I did a clean install of 64 bit bookworm desktop on my Pi 3B+.

Wondering that I was not able to activate Wayland. Found https://forums.raspberrypi.com/viewtopic.php?t=359880

Showing:

````txt
               .-----------.------------.
               |  Default  |  Optional  |
.--------------|-----------|------------|
|  Pi 5        |  Wayland  |  X11       |
|  Pi 400      |  Wayland  |  X11       |
|  Pi 4B 2GB+  |  Wayland  |  X11       |
|  Pi 4B 1GB   |  X11      |  Wayland   |
|  Pi 3A+/3B+  |  X11      |  -         |
|  Pi 3B       |  X11      |  -         |
|  Pi 2B       |  X11      |  -         |
|  Pi A+/B+    |  X11      |  -         |
|  Pi A/B      |  X11      |  -         |
|  Pi Zero 2W  |  X11      |  -         |
|  Pi Zero W   |  X11      |  -         |
|  Pi Zero     |  X11      |  -         |
`--------------^-----------^------------'
````

 You can get it working by adding `wayland=on` to the end of the `/boot/firmware/cmdline.txt` file.

Ensure that `.config/wayfire.ini` does **not** contain:

```ìni
[core]
xwayland = false
````

and `start_picframe.sh`does not contain:

```ìni
export SDL_VIDEODRIVER=wayland
````

This works for me. My interpretation is, that sdl is using internally the x-emulation of wayland. Pure wayland seems not to work.

Next test:
1. boot to console as user pi
2. /boot/firmware/cmdline contains still `wayland=on`
3. mv wayfire.ini wayfire.ini.old
4. create new wayfrie.ini file

```ìni
[core]
plugins = autostart

[autostart]
picframe = /home/pi/start_picframe.sh
````

5. run /usr/bin/wayfire from console

This works as well.

Next steps. Test again with bookworm lite.

paddywwoof, 2024-10-24:
Helge, well done. At this stage I don't think I'm worried about picframe running under true wayland or not. The objective is to get the CPU usage at a reasonable level and enable the screen switch off to not kill the process! I will try following your recipe on this RPi5 for booting to cmdline (should that be `/boot/firmware/cmdline.txt`?)

paddywwoof, 2024-10-24:
Just to expand on @helgeerbe's experimentation, I *didn't* have the `wayland=on` in `/boot/firmware/cmdline.txt` but did the other suggestions and when I ran `wayfire` or `/usr/bin/wayfire` from the command line after rebooting, the picframe ran and I could turn the screen on and off fine.

I then put the standard `.config/system/user/picframe.service` i.e.
```ini
[Unit]
Description=picframe on pi5

[Service]
ExecStart=/usr/bin/wayfire

[Install]
WantedBy=default.targetvice`
```
and picframe started on boot and the screen could still be turned on and off

sdegner999, 2024-10-24:
Solved, display on/off works!  Thanks, I made the changes, and it switches for me.  This is crazy good.  I'm showing everyone who comes to the house.

sapnho, 2024-10-25:
Wow, that would be a break-through! So everything that worked previously, now works with Wayland?

paddywwoof, 2024-10-25:
Wolfgang, I think this setup is still using X fiction calls at a lower level, but they are wrapped in SDL. I feel that the difference in screen blanking is an inconsistency that should be patched at some point, and when it does we can toggle SDL to use the Wayland back end.

I think the high CPU usage is something to do with the graphics driver emulation with the new hardware running older OS. It's probably also fixable but I can't get my RPi5 to stay on with the weak USB plugs that I have here, unless I use the latest OS!

Paddy

helgeerbe, 2024-10-25:
So I run the setup on 64 bit lite version on my 3B. It crashes, when I blank the screen. So there seems something missing in the lite version. Unfortunately, I don't see any logs.

helgeerbe, 2024-10-25:
OK, found it. xwayland was missing.

````bash
sudo apt install xwayland
````

I will update the wiki, and create a new release next week.

sapnho, 2024-10-29:
I will (try to) start a blog page with the clean instructions for the Pi 5 shortly. Let's see how far I get. :.)

paddywwoof, 2024-10-29:
@sapnho it sounds like Helge will update the Bookworm, Wayland wiki page and update the PYPI version of picframe this week. Might be worth waiting to see if he unearths any other quirks or dependencies!

The instructions should work for 3, 4 and 5 at least. Trying to set up the 32 bit OS was very frustrating as various dependencies were missing and had to be compiled from source. Yuk.

helgeerbe, 2024-10-30:
Just saw (https://www.heise.de/news/Raspberry-Pi-OS-Wayland-jetzt-fuer-alle-9997332.html) that wayland will be enabled on all raspberries. But they will switch the compositor. Instead wayfire, labwc will become default.

sapnho, 2024-10-30:
What does this change mean for our script?

helgeerbe, 2024-10-30:
Hi @sapnho I updated the wiki.

From the functional perspective, there is no different. Raspberry say, it has a better performance than wayfire.

https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland

Would you be so kind to test the setup?

Instead of 

````sh
pip install picframe
````

run 

````sh
pip install git+https://github.com/helgeerbe/picframe@dev
````

If this is working for you, I would create a new release.

sapnho, 2024-10-30:
I will gladly test it, but I can only do it tomorrow evening.

sapnho, 2024-10-31:
Hello @helgeerbe, just tried your latest instructions on a Pi 5 and it works fine with a CPU load of about 1%. Great!

sapnho, 2024-10-31:
You didn't specify in your new script how display_power is supposed to be set. My old Bookworm method with xrandr doesn't work anymore because it's part of the X11 utilities.

paddywwoof, 2024-10-31:
Wolfgang, I'm probably at fault for this. The mod I did https://github.com/helgeerbe/picframe/commit/0e2c8eb362ed4555ece40e82b2dc679cb782a6c5 went into main branch instead of dev branch.

Using main branch you will need to set display_power=2 in the config file. To test if the screen off works OK you can run a little python or bash application that runs the command line `wlr-randr --output HDMI-A-1 --off` or --on. It might be easiest to do `pip install --upgrade git+https://github.com/helgeerbe/picframe` i.e. without the @dev which should default to main branch. I would cede to Hege's knowledge of the workings of GitHub so maybe wait for his comment.

I will be out of reach of my RPi for a while so can't test stuff.

Automatic install scripts sound to be an excellent idea 💡

sapnho, 2024-10-31:
Actually, I accidentally had done `pip install picframe` instead of `pip install git+https://github.com/helgeerbe/picframe@dev`, so I guess I was using the main branch.

Just did another install, which worked as well.

`wlr-randr --output HDMI-A-1 --off` works as well. I guess it is no longer necessary to specify the resolution when turning it back on again?

sapnho, 2024-11-01:
One thing I noticed that `python3-venv is already the newest version (3.11.2-1+b1)` so might we remove it from `sudo apt install python3-venv git libsdl2-dev xwayland labwc wlr-randr`?

paddywwoof, 2024-11-01:
I think `venv` is shipped with python as pretty standard nowadays. But putting it in with other dependencies shouldn't do any harm as `apt` should spot it's already satisfied and not do any downloading. 

Did you try setting disply_power=2 in configuration.yaml?

helgeerbe, 2024-11-01:
I will set 

````yml
display_power=2
use_sdl2: True
````

as default options, in the next release

helgeerbe, 2024-11-01:
I published version 2024.11.01. I close this issue. Please open a new issue, if you find something new.

closed 2024-11-01 by helgeerbe