helgeerbe/picframe #406: Screen Blank when blur_edges set to True

PiSignage, 2024-09-23:
Hi, 

I seem to be having a problem when i have blur_edges set to True nothing shows just a black screen.

Any help would be great

paddywwoof, 2024-09-23:
Hi @PiSignage I think that might be due to a breaking change in the python image library (Pillow). Try upgrading picframe to the latest dev branch, something like (depending on what you did to originally set up your system pre or post Pi4 etc etc)
```
cd ~
source venv_picframe/bin/activate
pip install git+https://github.com/helgeerbe/picframe@dev --upgrade
```
I will try to pull the changes from the dev branch into main soon.

Paddy

paddywwoof, 2024-09-23:
I pulled the changes to main branch.  If you upgraded using the dev branch you might need to add a line to the viewer section of your configuration.yaml i.e.
``` yaml
  use_glx: False                          # default=False. Set to True on linux with xserver running. NB use_sdl2 might need to be False for this to work.
  use_sdl2: False                         # default=True. pysdl2 can use display without xserver, it should be installed as a dependency of pi3d
```

PiSignage, 2024-09-23:
Thank you for getting back to me, I can confirm all is working now with your suggestions

closed 2024-09-23 by PiSignage