helgeerbe/picframe #408: Black screen after boot

doblerguy, 2024-10-25:
Pulling my hair out here.  Had everything working great and now when I reboot, I see the pi go through it's boot sequence and then I am presented with a black screen and a "no signal" message. What's odd is that I had it working and then when I added MQQT into the mix it seemed to be when the black screen started.. But that could just be coincidence.

helgeerbe, 2024-10-25:
Hi, could be a problem, if picframe can‘t connnect to your broker initial. 
- add a logfile in your configuration.yaml and check log after reboot

doblerguy, 2024-10-25:
thanks for the response! total noob here.  How would I add the logfile to the YAML?  And how would it "break" if I had everything including MQTT setup and working in home assistant?

helgeerbe, 2024-10-25:
Check picframe configuration.yaml

''''Yaml
log_level: "WARNING"                    # default=WARNING, could beDEBUG, INFO, WARNING, ERROR, CRITICAL
  log_file: "/home/pi/picframe.log"                            # default="" for debugging set this to the path to a file. NB logging messages will
''''

This will create a logfile. 

Picframe might have a problem connecting to your broker. E. g. Typo in password.

doblerguy, 2024-10-26:
I think I cracked the code....  It looks like MQTT requires a username and password. If you try to go without it (as per the MQTT setup from Wolfgang's blog) you end up with the blackscreen.  I added a username and password for MQTT and then added that to the config and boom.  No more black screen.

closed 2024-11-01 by helgeerbe