helgeerbe/picframe #409: Building an installation script for picframe setup

sapnho, 2024-11-03:
I am building a script to include [all the necessary steps](https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/) to get picframe up and running. (Thanks Helge and Paddy!)

The script would reboot where necessary, remembering the next step in the installation sequence.

Everything works.... except I cannot get picframe -i /home/pi/ to run. That means, it runs, doesn't report any errors but doesn't do what it's supposed to do.

This is the header of script to manage the installation sequence.

```
#!/bin/bash

# Path to store progress and log file
PROGRESS_FILE="/home/pi/install_progress.txt"
LOG_FILE="/home/pi/install_log.txt"
SERVICE_NAME="install_script_service"

# Function to log messages
log_message() {
    echo "$1" | tee -a "$LOG_FILE"
}

# Function to update progress
update_progress() {
    echo "$1" > "$PROGRESS_FILE"
}

# Function to get the last completed step
get_last_completed_step() {
    if [ -f "$PROGRESS_FILE" ]; then
        cat "$PROGRESS_FILE"
    else
        echo "0"
    fi
}

# Function to add a systemd service to resume the script after reboot
add_systemd_service() {
    local script_path=$(realpath "$0")
    SERVICE_FILE="/etc/systemd/system/$SERVICE_NAME.service"
    sudo tee "$SERVICE_FILE" > /dev/null <<EOL
[Unit]
Description=Resume install script after reboot

[Service]
ExecStart=$script_path
Type=oneshot
RemainAfterExit=true

[Install]
WantedBy=multi-user.target
EOL
    sudo systemctl enable $SERVICE_NAME
    log_message "Added systemd service for reboot resume."
}

# Function to remove the systemd service after completion
remove_systemd_service() {
    sudo systemctl disable $SERVICE_NAME
    sudo rm /etc/systemd/system/$SERVICE_NAME.service
    log_message "Removed systemd service after completion."
}

# Function to reboot and resume
reboot_and_resume() {
    add_systemd_service
    update_progress "$1"
    log_message "Rebooting to complete the installation. The script will continue after reboot."
    sudo reboot
    exit 0
}
```
Then I have all the various parts like
```
# Main install script

# Get the last completed step
LAST_COMPLETED_STEP=$(get_last_completed_step)

# Step 1: Update the system (if not already done)
if [ "$LAST_COMPLETED_STEP" -lt 1 ]; then
    log_message "Step 1: Updating system..."
    sudo apt-get update && sudo apt upgrade -y
    reboot_and_resume 1
fi
```
Samba is Step 2 followed by:
```
# Step 3: Install additional packages
if [ "$LAST_COMPLETED_STEP" -lt 3 ]; then
    log_message "Step 3: Installing additional packages (xwayland, labwc, wlr-randr, etc.)..."
    sudo apt-get install git libsdl2-dev xwayland labwc wlr-randr -y
    reboot_and_resume 3
fi

# Step 4: Update raspi-config settings and reboot
if [ "$LAST_COMPLETED_STEP" -lt 4 ]; then
    log_message "Step 4: Configuring raspi-config settings for boot..."
    sudo raspi-config nonint do_boot_behaviour B2
    reboot_and_resume 4
fi

# Step 5: Set up virtual environment and install Picframe
if [ "$LAST_COMPLETED_STEP" -lt 5 ]; then
    log_message "Step 5: Setting up virtual environment and installing Picframe..."
    mkdir venv_picframe
    python -m venv /home/pi/venv_picframe
    source venv_picframe/bin/activate
    pip install picframe
    # Initialize Picframe and confirm default directories
    if (echo -e "\n\n\n" | picframe -i /home/pi/); then
        log_message "Picframe initialized with default directories."
    else
        log_message "Error: Failed to initialize Picframe."
        exit 1
    fi
    update_progress 5
    log_message "Picframe installation and initialization completed."
fi
```

I also added the autostart features in Steps 6-8 and all works fine, except for
    `picframe -i /home/pi/ `.

Interestingly, when I let the script run until picframe installation (without the initialisation), and then start a new script with only the picframe initialisation, everything works fine.

I tried so many things but couldn't come up with a solution.

Surely, there are brighter heads here than me.

sapnho, 2024-11-03:
Interestingly, it works fine when I put step 5 in a separate script.

sapnho, 2024-11-05:
Problem solved. The script installed as sudo, not a user pi.

sapnho, 2024-11-07:
Saves a bit of time: https://www.thedigitalpictureframe.com/the-six-minute-install-script-for-the-pi3d-pictureframe-software-for-your-raspberry-pi/

paddywwoof, 2024-11-09:
@sapnho this looks really comprehensive and easy to use, I'm looking forward to trying it out.
PS after a quick glance I just noticed you suggest installing a big, complex java application to check for IP addresses on your local network (for someone not used to logging into their router). You can do something fairly simple with `nslookup` i.e.
```python
import subprocess
for i in range(1, 255):
    try:
        print(subprocess.check_output(['nslookup', f'192.168.1.{i}']))
    except:
        pass
```

sapnho, 2024-11-09:
Hi @paddywwoof, the check_ip script is a neat idea. It probably makes sense to run it as the first step before a reboot, then the output is given in the Terminal. Will do a few checks. Thanks!

The script now runs fine on Pi 3-5. I had a few issues with the script being unreliable, which were due to the network not being available when the command started executing. Raspi-config used to have an option to wait for the network at boot, but that seems to be gone. So now I use a simple function to check for Google.

The other issue was using the correct permissions for certain commands, but that is solved.

It's quite a time saver, especially when you do so many reinstalls for testing on different Pi models as I do.

paddywwoof, 2024-11-09:
Hi @sapnho, you just need to run the script on your PC/Mac while the Rasbperry Pi has booted and done all its initial configuration, at which point it should be on the local network. `nslookup` seems to be a terminal command in linux, macOS and windows, so it should work on any of them.

sapnho, 2024-11-09:
It just found that `arp -a`  gives you a full list, making it even simpler. Could you check if it runs on your device?

I already updated the [article](https://www.thedigitalpictureframe.com/the-one-click-install-script-for-the-pi3d-pictureframe-software-2024-edition-raspberry-pi-3-4-5/) accordingly.

Thanks for the idea!

paddywwoof, 2024-11-09:
Hi, I don't have `arp` available on this ubuntu laptop but I can install it with `sudo apt install net-tools`. When I run `arp -a` it just lists the router on this network as that seems to be the only 'host' (no RPi here!). Presumably if other devices on this network were running server type applications then they would show up. I've not tried running it with windows OS but I think it's available by default.

Paddy

jgodfrey, 2024-11-09:
Local IP address. What about `hostname -I`? That returns expected results on my pi4s as well as a Linux Mint installation.

sapnho, 2024-11-09:
Yes, but for that you already need to be connected to the Raspberry Pi.

jgodfrey, 2024-11-09:
Ah, of course. I obviously didn't pay enough attention to this conversation. Sorry for the noise... :)

sapnho, 2024-11-09:
@jgodfrey, I wrote the same answer you gave to Paddy as well... :-)

sapnho, 2024-11-09:
> Hi, I don't have `arp` available on this ubuntu laptop but I can install it with `sudo apt install net-tools`. When I run `arp -a` it just lists the router on this network as that seems to be the only 'host' (no RPi here!). Presumably if other devices on this network were running server type applications (such as SSH server) then they would show up. I've not tried running it with windows OS but I think it's available by default.
> 
> Paddy

Hmm, so it works on a Mac. More digging needed. Ideally, I would like to avoid a script because it adds complexity.

sapnho, 2024-11-09:
> Hi, I don't have `arp` available on this ubuntu laptop but I can install it with `sudo apt install net-tools`. When I run `arp -a` it just lists the router on this network as that seems to be the [...]

Could you try `ip neigh` on your Ubuntu system?

sapnho, 2024-11-09:
If there is someone here on a Windows system, could you try if `arp -a ` works in Terminal? Thanks!

jgodfrey, 2024-11-09:
`arp -a` works under Win11 Pro...

jgodfrey, 2024-11-09:
Though, I will say that on my Linux Mint laptop, while `arp -a` returns a number of network devices, there are a lot of devices not shown - including my picture frame. Not sure why that is as I'm not really familiar with the details of `arp`.

sapnho, 2024-11-09:
Strange that it doesn't work on Linux. I commented on that in my instructions. But the Linux guys don't need me to tell them how to find an IP address anyway.. :-)

paddywwoof, 2024-11-09:
`ip neigh` gives the same address as `arp -a` i.e. just the router. I get that running `ip` or `arp` on windows 11 too, but there is only an IP address and MAC number - no device name.

Probably agree that you don't need too much details for folk doing all this from a linux machine. Also odd about the pictureframe not showing on linux mint laptop!

sapnho, 2024-11-14:
@helgeerbe, I guess we can close this thread.

sangh, 2024-11-19:
I wanted to comment saying that I read thedigitalpictureframe.com post and was a bit uncomfortable with the requirement that one has to autologin to start the picframe on rpi boot.  (I disabled USB so people can't just plug in a keyboard/mouse, but still, logging in shouldn't be needed).  So in the install script you can just completely not run Step 4 (or set raspi-config to boot to console without login).

Before wayland you could just enable linger and start xinit as a user service with picframe as the command.  But labwc won't start without a tty (and it's surprisingly difficult to mimc the Xwrap options to only give a non-root user access to start with a tty but not allow them to read _any_ tty). So, you can't use a user systemd service and have to use a root systemd service with a User= option to have it change to that user after start.  Then labwc will start.

Also I didn't want to depend on labwc config files in the user's home directory because by default it merges the system configs, so to get around that you can start labwc with the config directory option (and have the files in there with does not pull in the system configs).  And one can use the session (-S) option to avoid creating the startup .sh file and have labwc exit when picframe exits.