helgeerbe/picframe #411: Has anybody here connected Picture Frame to a network server?

sapnho, 2024-11-10:
I get this question quite often and I guess it can't be too hard.

Any knowledge sharing available?

helgeerbe, 2024-11-10:
https://darkwolfcave.de/nas-raspberry-netzwerkfreigabe-mounten/

sapnho, 2024-11-10:
> https://darkwolfcave.de/nas-raspberry-netzwerkfreigabe-mounten/

Thanks, Helge.

This must be the worst-designed website I have ever seen. :-) But it does have someful useful content.

Actually, mounting is not the issue, I wonder more if anything needs to be changed in PicFrame and how the app will respond to delays or interruptions that might be introduced through the network server.

helgeerbe, 2024-11-10:
On an unstable wifi network a network share mount wouldn't be the best idea. I would go for a sync client. Nextcloud or smaler and easy to setup https://docs.syncthing.net/

sapnho, 2024-11-11:
Yes, that makes sense. I use to sync with the photos on my Mac and it works like a charm.

adammhaile, 2024-11-17:
@sapnho I'm doing that with the frame I recently built and so far it's working fine.
All I'm doing is editing `/etc/fstab` to add a mount point at `~/Pictures` which points to a folder that has over 90GB of images on my NAS. The amount of images is why I didn't want to do any syncing to the Pi itself.
The shuffle option didn't seem to work at first, but I'm guessing just because it was still scanning images in the background. After about an hour it was definitely being shuffled as I would expect. Otherwise it's worked fine in the few days I've had the frame running.
Granted, I will admit that I'm quite a network infrastructure nerd so everything I have at home is basically enterprise grade and certainly not anything you'd call unstable.

Nerdiyde, 2024-11-17:
Hey guys, 
just an additional idea to cover this:
I use syncthing to distribute the pictures to the different picframes around my house. I have one folder with the "worth to put it on the frames"-pictures in a folder on my server. As soon as the picframe pis are powered up (I usually power them down when I'm at home or over night) they pull the latest state from the server.
This works since two years like a charm. This way I can also delete the pictures in a centralized way)
Only downside: On a RPI Zero 2 the additional load makes kenburn effects a bit laggy. But no problem so far on a RPi 4.
I run syncthing as docker-compose which makes configuration pretty easy. 
My recommendation is the following:

- Install docker
- Install portainer
- Finish portainer configuration
- Start new stack via Portainer Webinterface
- Use and adapt example configuration below:
- `version: "2.1"
services:
  syncthing:
    image: lscr.io/linuxserver/syncthing:latest
    container_name: syncthing
    restart: unless-stopped
    hostname: PiHostName ### Adapt this!
    environment:
      - PUID=1000    # put the correct user in here. must have rw permissions for all volumes bound in the container
      - PGID=100
      - TZ=Europe/Berlin
    volumes:
      - /home/fab/syncthing_config:/config  ### Adapt this!
      - /home/fab/Pictures:/Bilder  ### Adapt this!
      
      # add more volumes
    ports:
      - 8384:8384
      - 22000:22000/tcp
      - 22000:22000/udp
      - 21027:21027/udp`

- You need to do this on a server and in the picframe pi
- Connect two folders via information provided here: https://docs.syncthing.net/intro/getting-started.html

Let me know if you want to have any more information. :)