helgeerbe/picframe #416: Install on Raspberry Pi, Bullseye OS

ssnadra, 2024-11-16:
Hey y'all, starting an issue here in hopes someone might be able to help me with a problem I'm having during install. Specifically, the pi-heif wheel will not build on my raspberry pi 3B. 

I'm using Python3.11.2, the latest libheif library is installed, and I've run sudo apt-get update.

When trying to run `python3 -m pip install pi-heif` or any variation of it, I'm given the error codes :
```
error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
ERROR: Failed building wheel for pi-heif
```
 
I'm wondering if anyone has run into a similar problem or successfully installed pi-heif on their raspberry pi? It [seems to fail](https://www.piwheels.org/project/pi-heif/) building on the Bookworm OS but there are a number of accounts I can find of people using picframe on their raspberries. Any insight is appreciated!

sapnho, 2024-11-16:
Hi @ssnadra, Use Bookworm, pi-heif will install without issues

`Successfully installed IPTCInfo3-2.1.4 Pillow-11.0.0 PyYAML-6.0.2 defusedxml-0.7.1 ninepatch-0.2.0 numpy-2.1.3 paho-mqtt-2.1.0 pi-heif-0.20.0 pi3d-2.52 picframe-2024.11.1 pysdl2-0.9.16
`

Updated instructions are [here](https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/).

paddywwoof, 2024-11-16:
Hi, I tried using the 32 bit version of the OS a couple of weeks ago (as that's what's bundled on preinstalled SD cards) After a lot of messing around adding dependencies and trying to compile PILlow components, I just gave up... someone might sort this out sometime in the future as he if is becoming a standard and PIL and python are quite central to Raspberry Pi philosophy!!

paddywwoof, 2024-12-23:
@doubleO8 , that's a useful discovery on your part, though it does look odd on https://www.piwheels.org/project/pi-heif/ that the build has failed for all versions 0.7.0 to 0.21.0 for Bullseye and Bookworm.

If I get chance I will try pip installing older and older versions until I find where it stops breaking. After a quick look at github the last line of OP error listing `_pi_heif.c:1510:21` seems to be a change from about 10 months ago https://github.com/bigcat88/pillow_heif/commit/ed772d4a01ed64c4e40c8b453e98837c6968f6db#diff-891f96dddfd798f72d351f5cbc664c513bbcea8c83a610c82ea8f683340fa7e2R1316