helgeerbe/picframe #417: Overview of http commands available?

sapnho, 2024-11-23:
Do we have a primer on HTTP commands for PictureFrame like Helge did for [MQTT](https://github.com/helgeerbe/picframe/wiki/Control%20Interfaces)?

I am just playing around with a few ideas where I need to send HTTP commands to PictureFrame but somewhat my syntax isn't quite right yet.

paddywwoof, 2024-11-23:
Good idea, I will do something later. The commands are basically the same as the function names in the model or control file I think.Paddy

jgodfrey, 2024-11-23:
Hi Wolfgang,

Paddy's the expert here but until he chimes in you can always open Chrome dev tools, go to the network tab, and see what the existing HTML web page sends with each command. For example, here are a few for reference. 

[image]

Just provide the IP address and port of your frame along with the command. So, something like this:

`http://192.168.0.150:9000/?next={}`

paddywwoof, 2024-11-23:
Thanks Jeff, that's a really useful way of hacking all kinds of web accessible gadgets such as solar inverters etc etc. Chrome (or firefox) can basically give you the recipe for what you need to feed from your DIY app.

The HTTP interface gets a list of methods from the `Controller` object (see https://github.com/helgeerbe/picframe/blob/654006514513ddeb19f5ed16ba9736d0fb20f7f3/src/picframe/interface_http.py#L120) .. and here's the file listing https://github.com/helgeerbe/picframe/blob/main/src/picframe/controller.py

You can then send the method name = method argument as GET arguments to the HTTP server. i.e. like Jeff says
`http://192.168.0.150:9000/?next={}`
The methods are listed below (`true` can be upper or lower case and many things that seem plausible such as `true, on, yes, false, off, no`):
```
@property setters - these don't need a dict of arguments as they can 'just' set the value

?paused=true
?subdirectory=holiday2019
?date_from=2020/09/01
?date_to=2024/12/01
?display_is_on=true
?clock_is_on=true
?shuffle=true
?fade_time=10
?time_delay=30
?brightness=0.5
?matting_images=0.0
?location_filter=eldwick
?tags_filter=birthday

non @property methods must be passed arguments as dict of {"key1":val1, "key2":val2...   etc}
non argument methods
?back={}
?next={}
?delete={}
?refresh_show_text={}
?purge_files={}
?get_number_of_files={}
?get_directory_list={}
?get_current_path={}

methods with arguments
?set_show_text={"txt_key":"title", "val":true} // text_key can be "title", "caption", "name", "date", "location", "folder"
?text_is_on={"txt_key":"title"}
```
PS I'm doing this from the file listing as I'm away from an actual picframe to test on. I will double check the actual syntax when I'm home!!!

paddywwoof, 2024-11-24:
Hi @sapnho, it does look a little strange that you have to escape the curly brackets. How are you sending the instructions to the http server? It may be different when entering on the address bar of a browser. My list is what I would expect for python along the lines of:

```python
from urllib import request

URL = "http://localhost:9000"
...
            request.urlopen(f"{URL}?display_is_on=OFF")
            request.urlopen(f"{URL}?brightness=0.0")
            request.urlopen(f"{URL}?next={}")
            request.urlopen(f"{URL}set_show_text={\"txt_key\":\"title\", \"val\":true}")
```
I've escaped the quote marks in the last command but it should also work with `f"{URL}set_show_text={'txt_key':'title', 'val':true}"` as I think my code converts the `\'` characters to `\"` required by a json strings. As you can tell the issue of character encoding and quoting around strings is a complete nightmare in applications like this.

This is the HTML/js code for the simple web interface https://github.com/helgeerbe/picframe/blob/main/src/picframe/html/index.html

Also I might be wrong with some of this, I will check the details when I get home. So maybe hold off actually publishing your blog for a day or so!!

Paddy

sapnho, 2024-12-07:
@paddywwoof How would I send an HTTP command to show both date and location? I only manage to do one thing at a time.

paddywwoof, 2024-12-08:
Hi, Although the parts of the text to show are controlled by a cunning bit-field system, the method that changes each bit only works one bit at a time... with the exception of turning them ALL off, which can be done by calling `ViewerDisplay.set_show_text(txt_key=None)` (which is the default so could be done with `ViewerDisplay.set_show_text()` this method should be called via HTTP by, for instance:
```python
        request.urlopen(f"{URL}?set_show_text={{}}")
``` 
In order to turn on both date and location you would then need to send two requests:
```python
        request.urlopen(f"{URL}?set_show_text={{\"txt_key\":\"date\",\"val\":true}}")
        request.urlopen(f"{URL}?set_show_text={{\"txt_key\":\"location\",\"val\":true}}")
```