helgeerbe/picframe #418: Synching several Picture Frames

sapnho, 2024-11-24:
It's Sunday morning, a bit rainy outside.. time for some new ideas around PictureFrame!

I wonder how it could be achieved that two or three frames, running in the same house/network, all running Pi3D PictureFrame, would show the same pictures at the same time?

All the photos can be synched across the different Pis but it would take a Master Pi3D PictureFrame Pi telling the Slave installations which picture to show.

Is that even possible with our architecture?

paddywwoof, 2024-11-24:
At the moment the random list uses an sqlite function but if it was done using python then the seed could be stored and accessed by other RPis. I thought about this as a way to allow picframe to pick up from where it stopped but it could work for your Idea too. There would also need to be a sequence number and synchronisation time passed, possibly using v simple socket communication

helgeerbe, 2024-11-24:
I think picframe has nearly everything on board. I would use the mqqt home assistant interface.
[image]

1. Picframe slaves should subscribe to the master picframe image update message.
2. on message trigger next.
3. on image load, if slave read image from url
````python
from PIL import Image
import requests

im = Image.open(requests.get(url, stream=True).raw)
````
4. on metadata read, if slave, take data from mqtt message.

Advantage is, that you don't have to deal with synchronize your image pool. What's on master picframe will be displayed on slaves.

closed 2026-05-06 by github-actions[bot]