helgeerbe/picframe #421: What happens if a photo gets replaces with the same only with updated Exif data?

sapnho, 2024-12-04:
Probably a question for master @paddywwoof: I sometimes check where my photo library might have missing location data and add it. I then just replace the image in /Pictures.

Will PictureFrame read the updated Exif data even if the file name doesn't change?

And an add-on question: When the language for geodata retrieval is changed in configuration.yaml, will the database be updated or do you have to delete it to start afresh?

Thanks!

paddywwoof, 2024-12-04:
Hi, I'm, yet again, operating in limbo from memory (I could look at the source code in the GitHub repo but it's hard work making sense on my phone)... I seem to remember that the db building thread checks for folder change date times, which I think are updated when a component file changes. But I know Jeff and I messed around getting updates to happen, so this might be something that doesn't work properly. 

I don't know about the language. I would expect the basic info from the exif to be stored, but that might have added a repeated processing load that we wanted to avoid by making a one off translation when the images are added. 

I should be able to look at the code later when I have my laptop. 

Paddy

jgodfrey, 2024-12-04:
Also going from memory...

> Will PictureFrame read the updated Exif data even if the file name doesn't change?

I want to say that "it should" as long as the file's disk-based timestamp is newer than the one recorded in the db. However, I seem to remember case(s?) where that didn't seem to work (can't remember the details). However, I think if such cases do exist simply `touch`ing the updated file will cause the disk crawler to find it and the db will be updated.

> When the language for geodata retrieval is changed in configuration.yaml, will the database be updated

I don't think so. IIRC, if we have a location record for a given lat/lon combo, we just use it. If we don't, it's looked up and new a record is built. So, you'd either have kill the db or clear the `location` table.

sapnho, 2024-12-07:
It doesn't seem to update the information, but I probably need to make a dedicated test on a test Pi to be sure. 

Is there any drawback to automatically deleting the database one a week so that it refreshes all information?

paddywwoof, 2024-12-08:
@sapnho is that neither of Jeff's suggestions? With the first point, about changes to an existing file, you could try `touch`ing the directory in which the changed file exist. I think that the last modified date doesn't work it's way up in you simply `touch` a file within a folder, and picframe looks at the directory to see whether or not to work its way through all the contained files.

closed 2026-05-06 by github-actions[bot]