helgeerbe/picframe #422: Error with Pillow Install

jhopkins213, 2024-12-07:
When trying to install picframe I get this 

 pip install picframe
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picframe
  Using cached https://www.piwheels.org/simple/picframe/picframe-2024.11.1-py3-n                                                                          one-any.whl (3.3 MB)
Collecting Pillow>=10.2.0 (from picframe)
  Using cached pillow-11.0.0.tar.gz (46.7 MB)
  Installing build dependencies ... -Pillodone
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting defusedxml (from picframe)
  Using cached https://www.piwheels.org/simple/defusedxml/defusedxml-0.7.1-py2.p                                                                          y3-none-any.whl (25 kB)
Collecting pi3d>=2.52 (from picframe)
  Using cached https://www.piwheels.org/simple/pi3d/pi3d-2.52-py3-none-any.whl (                                                                          256 kB)
Requirement already satisfied: PyYAML in ./venv_picframe/lib/python3.11/site-pac                                                                          kages (from picframe) (6.0.2)
Collecting paho-mqtt (from picframe)
  Using cached https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-2.1.0-py3-non                                                                          e-any.whl (67 kB)
Collecting IPTCInfo3 (from picframe)
  Using cached https://www.piwheels.org/simple/iptcinfo3/IPTCInfo3-2.1.4-py3-non                                                                          e-any.whl (12 kB)
Collecting numpy (from picframe)
  Using cached https://www.piwheels.org/simple/numpy/numpy-2.1.3-cp311-cp311-lin                                                                          ux_armv7l.whl (6.0 MB)
Collecting ninepatch>=0.2.0 (from picframe)
  Using cached ninepatch-0.2.0-py3-none-any.whl.metadata (4.9 kB)
Collecting pi-heif>=0.8.0 (from picframe)
  Using cached pi_heif-0.21.0.tar.gz (16.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pysdl2 (from pi3d>=2.52->picframe)
  Using cached https://www.piwheels.org/simple/pysdl2/PySDL2-0.9.16-py3-none-any                                                                          .whl (580 kB)
Using cached ninepatch-0.2.0-py3-none-any.whl (9.1 kB)
Building wheels for collected packages: pi-heif, Pillow
  Building wheel for pi-heif (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for pi-heif (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [75 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/misc.py -> build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/constants.py -> build/lib.linux-aarch64-cpython-311/pi_hei                                                                          f
      copying pi_heif/_lib_info.py -> build/lib.linux-aarch64-cpython-311/pi_hei                                                                          f
      copying pi_heif/_version.py -> build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/_deffered_error.py -> build/lib.linux-aarch64-cpython-311/                                                                          pi_heif
      copying pi_heif/__init__.py -> build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/as_plugin.py -> build/lib.linux-aarch64-cpython-311/pi_hei                                                                          f
      copying pi_heif/options.py -> build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/heif.py -> build/lib.linux-aarch64-cpython-311/pi_heif
      copying pi_heif/HeifImagePlugin.py -> build/lib.linux-aarch64-cpython-311/                                                                          pi_heif
      copying pi_heif/AvifImagePlugin.py -> build/lib.linux-aarch64-cpython-311/                                                                          pi_heif
      running build_ext
      Package libheif was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libheif.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'libheif', required by 'virtual:world', not found
      Package heif was not found in the pkg-config search path.
      Perhaps you should add the directory containing `heif.pc'
      to the PKG_CONFIG_PATH environment variable
      Package 'heif', required by 'virtual:world', not found

      The headers or library files could not be found for libheif,
      a required dependency when compiling Pillow-Heif from source.

      Please see the install instructions at:
         https://pillow-heif.readthedocs.io/en/latest/installation.html

[... 248 lines cut ...]
        File "/tmp/pip-build-env-9kjeiiqx/overlay/lib/python3.11/site-packages/s                                                                          etuptools/build_meta.py", line 426, in _build
          return self._build_with_temp_dir(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-9kjeiiqx/overlay/lib/python3.11/site-packages/s                                                                          etuptools/build_meta.py", line 407, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-install-xdgywips/pillow_5582eaa966774792b8ce511f0a944703/                                                                          _custom_build/backend.py", line 20, in run_setup
          return super().run_setup(setup_script)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-9kjeiiqx/overlay/lib/python3.11/site-packages/s                                                                          etuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
        File "<string>", line 1042, in <module>
      RequiredDependencyException:

      The headers or library files could not be found for jpeg,
      a required dependency when compiling Pillow from source.

      Please see the install instructions at:
         https://pillow.readthedocs.io/en/latest/installation/basic-installation                                                                          .html

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem wit                                                                          h pip.
  ERROR: Failed building wheel for Pillow
Failed to build pi-heif Pillow
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based p                                                                          rojects (pi-heif, Pillow)

jhopkins213, 2024-12-07:
I appologize didnt install the 64 bit version this can be closed

closed 2024-12-07 by jhopkins213

paddywwoof, 2024-12-07:
@jhopkins213 it's really annoying that PIL insulation fails with the 32 bit OS because of missing heif headers. That's the 'default' version of the OS (ie if you buy a preinstalled SD card) to run on all versions of raspberry pi. I might see if I can raise it as an issue somewhere... not sure where though.

paddywwoof, 2024-12-10:
@jhopkins213 to add a little to this thread in case someone else happens across it with the same error messages. I don't think there is a `libheif` shared library compiled in the debian repositories for the 32 bit arm target. I've tried compiling it but ended up in dependency hell. BUT it might very well be possible to install picframe without heif support by cloning the repository to your 32 bit SD card, then taking the relevant out of the `pyproject.toml` file. i.e. follow the normal instructions on [Wolfgang's site](https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/) (you will have to do the manual version rather than use the one-shot script) up to `pip install picframe` but instead
```bash
cd ~
git clone https://github.com/helgeerbe/picframe.git
cd picframe
nano pyproject.toml # delete the line, around #45 "pi_heif>=0.8.0" .. Ctrl-S, Ctrl-X
python -m pip install -e .
```
NB the last line has a critical `.` on the end of it. This will install the code from the local repo in `/home/pi/picframe`. If there is an update in the future you should be able to navigate to the directory and `git pull` to update to the latest version then. `python -m pip install -e . --upgrade`