helgeerbe/picframe #423: Does overlay still work with Bookworm/Wayland/labwc?

sapnho, 2024-12-09:
Got a question regarding [this](https://github.com/helgeerbe/picframe-overlay), but user only got a black screen.

helgeerbe, 2024-12-09:
Havn't tested it. Actual I'm on vacation. Will Test it when I'm back.

paddywwoof, 2024-12-09:
Hi, the overlay works when I've tested it recently. But the link in your post is to @helgeerbe 's system using JavaScript and browser tech.

To test if the basic overlay works, copy a PNG with transparency to `/dev/shm/overlay.png`

helgeerbe, 2024-12-27:
Hi, I'm back. Just tested the overlay. For wayland you need to do the following changes:

My setup ist raspbian light and wayland

$ sudo apt-get install libgtk-3-dev

Change in `package.jason` to

````json
...
"scripts": {
    "start": "electron --enable-features=UseOzonePlatform --ozone-platform=wayland .
  }
...
````

Ensure, that picframe is already started, because it starts wayland.

closed 2025-03-31 by helgeerbe