helgeerbe/picframe #424: More Pictures

123Oliver123, 2024-12-11:
Hello everyone :)

I am setting up my first Picture Frame, and as I was testing it, I thought it would be nice to have more than one Image displayed at a time (or 2 if they are in Portrait), like a digital photo wall. The display is quite big, so I think that such a feature would look quite Interesting.

Is there a way to do that or a way I could try to implement it? 

Thanks in advance for your help and kind regards
Oliver

123Oliver123, 2024-12-11:
Hi @jgodfrey 

Thanks for the fast reply, i already did that and it worked fine, my question was more about beeing able to have 5,6 or more pictures displayed at one time, like a digital Foto wall.

Do you know if something like that is also supported ?

jgodfrey, 2024-12-11:
Ah, yeah. I misunderstood the question.

No, for multi-image display, the application currently only supports the 2-image "portrait pair" mode. Displaying more images would definitely complicate a number of integral features, including text overlay and matting.

123Oliver123, 2024-12-12:
Okay, thank you anyways :)