helgeerbe/picframe #426: MQTT and web "Display" off function automatically turns back on 20 seconds later.

mattbatt, 2024-12-13:
Sorry if I'm asking a dumb question, but it has fought me for days. 
I'm using a plain old Yard Sale HP 1080p computer monitor as my display and a RPi4, every time I toggle the "Display off" in MQTT or on the web page the display pops back on 20 seconds later. 
What am I doing wrong?  
I've resorted to plugging the monitor into a IOT outlet and powering it off with HA at night. 

Thanks for this project. 

Matt

paddywwoof, 2024-12-13:
@mattbatt I'm not sure how different HDMI devices can influence the behaviour of the OS control of the screen but I think it is certainly possible! We struggled to get the latest hardware (Pi5) and OS with Wayland to cope with the screen turning off, so just to check, are you using the latest instructions as on https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland or Wolfgang's equivalent?

When the screen comes back on does the slideshow restart or is it blank, or paused, or something else?

sangh, 2024-12-13:
@mattbatt Complete and total guess without any real information: When you turn off picframe the pi stops HDMI, but does not send the CEC command to tell the monitor to turn off (I'm pretty sure neither wlr-randr nor the other two methods picframe uses send CEC commands).  Then the monitor searches all it's inputs, one by one, to see if any are producing video output BUT as it is doing this it sends the CEC turn on command to the pi, which the xserver interprets as a wake up command and starts producing video output (even if it's a black screen).  So... maybe your old monitor (NOT the pi/picframe) has a menu option to "turn off auto-input switching", or disable "auto-scan", or "sleep immediately when a video signal is lost", or something else that would disable the searching for a video signal?