helgeerbe/picframe #428: Where to set the max time for time_delay in Home Assistant

sapnho, 2024-12-21:
This is, I guess, one for @helgeerbe: 

In the Home Assistant integration, there is the 
```
entity: number.picframe_time_delay
    name: Time_delay
```
setting, which is limited to 400 seconds. I could find any place where this can be increased beyond 400. Can you enlighten me?

helgeerbe, 2024-12-27:
Those ui elements are generated by mqtt from picframe. To change the values you have to tweak the code.

In `interface_mqtt.py'`look for:  

       # numbers
        self.__setup_number(client, "brightness", 0.0, 1.0, 0.1, "mdi:brightness-6", available_topic)
        self.__setup_number(client, "time_delay", 1, 400, 1, "mdi:image-plus", available_topic)
        self.__setup_number(client, "fade_time", 1, 50, 1, "mdi:image-size-select-large", available_topic)
        self.__setup_number(client, "matting_images", 0.0, 1.0, 0.01, "mdi:image-frame", available_topic)

sapnho, 2024-12-29:
Ah, thanks, @helgeerbe!