helgeerbe/picframe #429: Black background instead of blurred pictures

Fabian8112, 2024-12-23:
Hallo together,

I am struggeling to find the right configuration to fit pictures into the frame (= allow gaps), and fill the background in black color (no blurred, mirrored picture):

Want to have it like this:
https://www.thedigitalpictureframe.com/wp-content/uploads/IMG_2630.jpg

Current configuration:

blur_amount: 12   
blur_zoom: 1.0                    
blur_edges: False                     
edge_alpha: 0.5                       
fps: 20.0                         
background: [0, 0, 0, 1.0]        
fit: True

Can you help please? Thank you!
Fabian

paddywwoof, 2024-12-23:
@Fabian8112 , sorry, it's not very clear in `configuration.yaml` it should read:
```yaml
  edge_alpha: 0.0 # default=0.5, how much background colour shows through edge. 0.0 would be *all* background 1.0 would be *all* blurred reflection of image
```
i.e. you need to set the edge_alpha to 0.0

cheers,

Paddy

Fabian8112, 2024-12-23:
@paddywwoof  thank you!!! Now it's clear and working. Wish you wonderful Christmas. Fabian

closed 2025-03-31 by helgeerbe