helgeerbe/picframe #432: Touchscreen Problems

h0d3nt3uf3l, 2025-01-05:
Hi, yesterday I intalled the picframe and want to use it as "screensaver" for my HomeAssistant panel in the living room.

Hardware:
I'm using a RPI4 with Raspberry Pi OS (64-bit, Released: 2024-11-19). It's using Wayland with labwc as compositor.
Attached is a Iiyama North America PL2453MT in Portrait-Mode (works ot of the box with the rotation made on the Desktop, the touch function was rotated as well)

I was looking at the code today to get the touchscreen running. A few clicks did some things but it was never predictable.
I found out, that the mouse-postition and touch-position of the pointer where different and the touch position implausible as well (e.g. x= -23403, y= -2233)

That's because in pi3d.display Line 351 is delivering other values. I don't know it for sure, but I guess pi3d is broken on this point...
Btw. don't find where _mouse_relative is set for sure..

I'm not deep enough in that code to do some pull requests, but maybe I can help you with my findings.
It works for me with followed changed lines:
``` python 
pi3d.display Line 350..

        elif self.ev.type == sdl2.SDL_MOUSEMOTION:
          if self._mouse_relative: # default
->          self._mouse_x = self.ev.motion.x
->          self._mouse_y = -self.ev.motion.y
          else:
            self._mouse_x = self.ev.motion.x
            self._mouse_y = -self.ev.motion.y
```
``` python 
pifcrame.interface_peripherals Line 291..

    def __update_pointer_position(self) -> None:
        position_x, position_y = self.__mouse.position()
        if self.__input_type == "mouse":
            position_x -= self.__viewer.display_width // 2
 ->         position_y += self.__viewer.display_height // 2
        elif self.__input_type == "touch":
            # Workaround, pi3d seems to always assume screen ratio 4:3 so touch is incorrectly translated
            # to x, y on screens with a different ratio
 ->         position_x -= self.__viewer.display_width // 2
 ->         position_y += self.__viewer.display_height // 2
 ->         # position_y *= self.__viewer.display_height / (self.__viewer.display_width * 3 / 4)
        self.__pointer_position = (position_x, position_y)
```

Greetings

P.s. Maybe you could tell me why there is always a bisection of the display resolution... I don't get it :)

paddywwoof, 2025-01-05:
Hi @h0d3nt3uf3l thanks for feedback and big fixing. The pi3d mouse location is rather messy as it has adopted functionality from lots of different bits of software which has to be specified on start up. The original RPi worked well without desktop but had to get mouse input using fairly low level IO. Later, with an x11 desktop, or tk, sdl2 etc there ended up with more mechanisms.

I'm away from home at the moment but will have a look at this when I get home.

h0d3nt3uf3l, 2025-01-07:
@paddywwoof thanks for this explanation. Now I have a hint for a misbehaviour which is still present with my touchscreen.
The "os pointer" isn't in synch with the pi3d pointer atm. After reboot I have to "reset" it with touching in the lower right corner (getting the pi3d to the max border)
Not in synch means: If I touch after reboot the upper left corner, the pointer is showing somewhere at x  ~ 350 and y ~ -900 on a display resolution of 1920x1080

Do I understand you right that in OpenGL logic, the middle of the display (e.g. 1920 / 2 and 1080 / 2) is point (0, 0, 1)?

paddywwoof, 2025-01-07:
@h0d3nt3uf3l TL;DR yes. The longer answer is that there is vector math that takes a projection of the vertices representing polygons in 3D, as if viewed from the point of view of the camera, scales them for perspective and only renders the nearest ones to the camera. Obviously, for rendering a single quad for picframe, that's all a lot more complicated than is really needed. Also, within OpenGL texture lookup (i.e. UV coordinates) the origin is in the top left corner and for fragment sampling (i.e. pixels from a render) the origin is bottom left (I think, from memory).

h0d3nt3uf3l, 2025-01-07:
@paddywwoof yeah, just read about the opengl coordinate systems, the different spaces for them and transformation from 3D to the screen.

Will do some changes on my code these days and try to get the touch screen work properly.
Hope that my first thoughts about bugs in pi3d will be wrong because of my lack of understanding openGL, would be hard to work on both projects with missing knowledge if not.

paddywwoof, 2025-01-07:
@h0d3nt3uf3l there is almost certainly stuff that needs to be tidied (aka fixed) in the SDL2 mouse and keyboard code for in pi3d. I don't think that your patch of making the `_mouse_relative` work like an absolute mouse location can be correct as it will break all the relative mouse use (i.e. using the mouse to 'steer' around a 3D landscape you want the camera to rotate as you move the mouse without worrying where the mouse actually is). However, the fact that it worked better after you made the change, makes me think that `_mouse_relative` should be `False`.

This is set https://github.com/tipam/pi3d/blob/df0cc84aebc56fb8d62e2517aeb48e59da689681/src/pi3d/Mouse.py#L56 which means that for it to be `True` the `restrict` argument to `Mouse.__init__()` must have been `False`.

And the mouse is initialised here https://github.com/helgeerbe/picframe/blob/654006514513ddeb19f5ed16ba9736d0fb20f7f3/src/picframe/interface_peripherals.py#L135 i.e. for `restrict` to be `False`, `__input_type` must *not* be "mouse", and presumably you set it as "touch", so that makes sense... But maybe it needs to operate differently if you had to tweak your code as you did.

pyrced, 2025-01-27:
I've been having trouble getting the touchscreen functions to actually work when I set it up per the bookworm and Wayland guide. When I tried to enable touch in the config file, it never worked, but then after reading through this thread and saw issues with sdl2 being mentioned, I decided to set use_sdl2 to false in the config file and the touch input actually works now. I have also tried setting up picture frame using the bookworm and X guide and the touchscreen input works identically to what I have working on Wayland, and that's with use_sdl2 set to true on X. The thing I have noticed with both Wayland and C set-ups is though, is that I can tap at the top of my display to bring up the menu, but none of the menu items can be actioned no matter how you tap on them. I'm not sure why that is. As for the touch experience in general, I'm grateful for the work put in to enable this as I can tinker and experiment with things but I can't code (python is on my to learn list). I do find the whole touch input janky though, because you have to quickly tap the screen to get it to transition only one picture at a time and not many. I have read through the code on git and can see mentions for why this may be:
 "Due to pi3d not reliably detecting touch as Mouse.LEFT_BUTTON event
        when a touch happens at any position with x or y lower than previous touch,
        any pointer movement is considered a click event"
My experience is that if you hold your finger on the display for more than a brief second it detects that as repeated 'clicks'.

Has anyone else mentioned that mouse input doesn't seem to work? When I enable mouse input in the config file I only ever get a black display when picture frame starts.

j3tang, 2025-01-29:
Could someone help with the screen rotation?  I installed Pi3D onto a RPi4 with Bookworm and as advertised, the official Touch Display 2 runs in portrait mode:

https://www.raspberrypi.com/documentation/accessories/touch-display-2.html

The link also describes towards the bottom of the page how to rotate the screen by 90, 180 or 270 degrees.
I've tried editing the file as described, but it doesn't rotate the screen at all.

Prior to using the Touch Display 2, it was plugged in to a TV through HDMI and it worked as intended ... I only kept landscape photos on the device and they showed correctly on the TV.

pyrced, 2025-02-14:
I have been doing some trouble shooting and figure out why setting input to "mouse' in the config file would cause picture frame not to start. I decided to querry the status of the picture frame service using systemctl --user status picframe.service and returned the following: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/util/icons/pointer.png'
I have looked at the the pi3d tar files for previous version and can see that the pointer icon file is missing from at least version 2.52 of pi3d. It was present in the 2.49 tar of pi3d I dowloaded. I extracted the icon file and copied it to the expected directory and voila, picture frame started with the input set to "mouse' in the configuration file. I have since been able to test the onscreen menu with the mouse controls and it does what you expect it to do with mouse input. The touchscreen doesn't at all correlate with the pointer position on the screen though.
Although I don't know any python, I have been trying some script modifications out through google gemini to improve the touchscreen input. It's a few minor changes to detect when the touch starts and ends to make touch more predictable and expected so it doesn't skip through multiple pictures with a slightly long touch or slight movement of the finger.

paddywwoof, 2025-02-14:
I think most of these problems have arisen now because of the switch to using SDL as the main controller of windows, keyboards and mouse. The way events are collected and processed for SDL by pi3d is different from the X11 server system and results in multiple key or touch events.

If you discover any concrete fixes with your modifications please pass them on as I am in the process of revising these aspects of pi3d

Paddy

h0d3nt3uf3l, 2025-02-15:
Hi, I'm sorry to being not present and couldn't fix it til now but my work and family need my attention atm.

I think the whole problem is indeed the interpretation of coordinates delivered from SDL2. There is a switch needed if X or Wayland is used an based on that changing the behaviour.

Greetings

h0d3nt3uf3l, 2025-02-15:
> I have looked at the the pi3d tar files for previous version and can see that the pointer icon file is missing from at least version 2.52 of pi3d. It was present in the 2.49 tar of pi3d I dowloaded. [...]

I discovered this too. With the pointer enabled you can see how pi3d is interpreted the coordinates.

paddywwoof, 2025-02-15:
Maybe something changed in default gitignore https://stackoverflow.com/questions/39903374/why-does-git-ignore-png-files-and-how-do-i-un-ignore-them

paddywwoof, 2025-02-17:
It happened when I (had to) switch from the old setup.py system for packaging modules for pypi
https://stackoverflow.com/questions/79441934/python-venv-install-skips-component-file-pointer-png/79442061#79442061
When I push the next revision with mouse and keyboard harmonisation I will fix this

Paddy

paddywwoof, 2025-03-04:
Hi I've put the modifications to mouse and keyboard into development branches. They seem to work with mouse and keyboard but I don't have a touchscreen so can't check that out. If you get a chance you can have a go probably by doing something like
```bash
source /home/pi/venv_picframe/bin/activate
python -m pip install git+https://github.com/helgeerbe/picframe@dev
python -m pip install git+https://github.com/tipam/pi3d@develop
```

pyrced, 2025-03-04:
I'll get tinkering today and test it all out, thanks for putting in the work. I'll report back here once I have.

pyrced, 2025-03-05:
Hey Paddy,  I'm not sure that I'm going about this properly. I tried updating my pictureframe install using your instructions above. I already had picture framed installed from git source code. Do I run the commands to update my current install or is this to do a fresh install of the application. I have tried both and generally get the same outcome of not being able to run picture from. This is what happens when I try to run these commands on an already working picture frame installed from git source code:

I start by activating the virtual environment
source /home/pi/venv_picframe/bin/activate

This is what I get when I run the above command for picture frame:
python -m pip install git+https://github.com/helgeerbe/picframe@dev
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/helgeerbe/picframe@dev
  Cloning https://github.com/helgeerbe/picframe (to revision dev) to /tmp/pip-req-build-cz8opr2d
  Running command git clone --filter=blob:none --quiet https://github.com/helgeerbe/picframe /tmp/pip-req-build-cz8opr2d
  Running command git checkout -b dev --track origin/dev
  Switched to a new branch 'dev'
  branch 'dev' set up to track 'origin/dev'.
  Resolved https://github.com/helgeerbe/picframe to commit 1e45ff18f1178bb8d45bf18a48b599977d2d00e7
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: Pillow>=10.2.0 in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (11.1.0)
Requirement already satisfied: defusedxml in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (0.7.1)
Requirement already satisfied: pi3d>=2.52 in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (2.53)
Requirement already satisfied: PyYAML in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (6.0.2)
Requirement already satisfied: paho-mqtt in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (2.1.0)
Requirement already satisfied: IPTCInfo3 in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (2.1.4)
Requirement already satisfied: numpy in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (2.2.3)
Requirement already satisfied: ninepatch>=0.2.0 in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (0.2.0)
Requirement already satisfied: pi_heif>=0.8.0 in ./venv_picframe/lib/python3.11/site-packages (from picframe==0+untagged.569.g1e45ff1) (0.21.0)
Requirement already satisfied: pysdl2 in ./venv_picframe/lib/python3.11/site-packages (from pi3d>=2.52->picframe==0+untagged.569.g1e45ff1) (0.9.17)
Building wheels for collected packages: picframe
  Building wheel for picframe (pyproject.toml) ... done
  Created wheel for picframe: filename=picframe-0+untagged.569.g1e45ff1-py3-none-any.whl size=3252611 sha256=e6aac5927ca77f925c6a0ba89e7240fb3c2429f009bd47cc6d072472053c37d7
  Stored in directory: /tmp/pip-ephem-wheel-cache-_ibdahbq/wheels/bf/57/c4/17e14ac685ae0751fa5b1e6bba32c84f01654c6e3b730f76df
Successfully built picframe
Installing collected packages: picframe
  Attempting uninstall: picframe
    Found existing installation: picframe 0+untagged.565.g6540065
    Uninstalling picframe-0+untagged.565.g6540065:
      Successfully uninstalled picframe-0+untagged.565.g6540065
Successfully installed picframe-0+untagged.569.g1e45ff1

This is what I get when I run the command for pi3d:
python -m pip install git+https://github.com/tipam/pi3d@develop
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/tipam/pi3d@develop
  Cloning https://github.com/tipam/pi3d (to revision develop) to /tmp/pip-req-build-7wq4w_4b
  Running command git clone --filter=blob:none --quiet https://github.com/tipam/pi3d /tmp/pip-req-build-7wq4w_4b
  Running command git checkout -b develop --track origin/develop
  Switched to a new branch 'develop'
  branch 'develop' set up to track 'origin/develop'.
  Resolved https://github.com/tipam/pi3d to commit 09868403f4812e3e3b2a01b22c3349cc3cb4929b
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: Pillow in ./venv_picframe/lib/python3.11/site-packages (from pi3d==2.53) (11.1.0)
Requirement already satisfied: numpy in ./venv_picframe/lib/python3.11/site-packages (from pi3d==2.53) (2.2.3)
Requirement already satisfied: pysdl2 in ./venv_picframe/lib/python3.11/site-packages (from pi3d==2.53) (0.9.17)

The picture frame application just won't run after running these commands. This is what I get if I query the status of the picture frame service:
 systemctl --user status picframe.service
● picframe.service - PictureFrame on Pi
     Loaded: loaded (/home/pi/.config/systemd/user/picframe.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-03-05 15:00:26 AEDT; 21s ago
   Main PID: 1249 (labwc)
      Tasks: 3 (limit: 8742)
        CPU: 1.700s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/picframe.service
             └─1249 /usr/bin/labwc

Mar 05 15:00:30 raspberrypi labwc[1267]: Traceback (most recent call last):
[... cut ...]
Mar 05 15:00:30 raspberrypi labwc[1267]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py", line 10, in <module>
Mar 05 15:00:30 raspberrypi labwc[1267]:     from picframe.video_streamer import VideoStreamer
Mar 05 15:00:30 raspberrypi labwc[1267]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/video_streamer.py", line 1, in <module>
Mar 05 15:00:30 raspberrypi labwc[1267]:     import mpv
Mar 05 15:00:30 raspberrypi labwc[1267]: ModuleNotFoundError: No module named 'mpv'

pyrced, 2025-03-05:
Okay, I figured it out based on the pictureframe service query and noting that No module named 'mpv', I looked at the video_streamer.py code and noted you had the comments
#TODO at the moment mpv isn't included as a dependency `python -m pip install mpv`
# it depends on having libmpv.so installed, which proabaly needs `sudo apt install libmpv-dev`
I ran these commands in the virtual environment to install mpv and now pictureframe works.
However there is still the persistent issue with sdl2 using a wayland backend.
If you have use_sdl2: True the touchscreen doesn't work.
If you have use_sdl2: False the touchscreen does work, and the menu items can be triggered as expected. I did read another thread about touchscreen problems you provided the same advice for and can confirm I am seeing the PAUSED prompt showing up at the bottom of the display to notify the slideshow is paused. The touch is still really sensitive and unpredictable with any touch more than a brief quick touch, registering multiple 'clicks'.

paddywwoof, 2025-03-05:
Thanks for testing this. I'm really sorry to muddle things with the video playing funtionality, that really needs wrapping in a `try:` block. I will do that straight away.

When you say the touchscreen doesn't work with SDL2, is there no video output, or is it just the touch functionality?

I obviously need to do some more investigation into collecting and parsing touch events anyway, but what is the behaviour like if you use the mouse option in configuration.yaml?

pyrced, 2025-03-05:
If you would like some log data or anything like that, I would be happy to provide this, I just don't know how to do that. If you point me in the right direction on how to collect and retrieve log data, I will certainly collect this for you.

pyrced, 2025-03-05:
Well I just set up another sd card with pictureframe using x instead of wayland. I get the exact same issues as I do with wayland. If I use sdl2 with x, the two mouse cursors appear in sync in opposite quadrants of the display.

paddywwoof, 2025-03-05:
@pyrced that's fantastic info, thanks.

The issue with the mouse position being offset by half a screen height and width was the problem with the previous way that pi3d.Mouse and pi3d.Gui worked. I thought I had fixed that in the latest develop branch. I also thought the missing `pointer.png` problem had been fixed, so that's a bit odd. I thought the addition of `*.png` in the setup file had fixed that - and, indeed, it does exist on this laptop where I pip installed from github branch. But it's missing on the RPi, very odd.

Anyway I will double check what's going on. One thing that should be fixable is the mechanism in picframe touch where it interprets *every* pointer movement as a click. That seems OTT.

paddywwoof, 2025-03-05:
I've spent a long time going round in circles with this. TL;DR version is that pip installing from a github branch as per my comment above doesn't entirely work (it does some things some of the time but I don't know what the critical factors are) I managed to get it working by copy pasting from my laptop to the venv source!

I will post more later, but I would leave off banging your head against this particular brick wall for a while!

paddywwoof, 2025-03-05:
Maybe the best solution will be for me to commit the pi3d changes to the main branch and move it to v2.54 on pypi. That way the pip --upgrade should work.

I have found, when testing with the `mouse` option just now, that the menu works OK but if the display is turned off it doesn't wake up when the mouse is moved. SDL2 doesn't seem to register any mouse  movements after that! picframe is still running and the http or mqtt interface can be used to wake the screen up. I think with the X11 mouse (and hopefully touch) getting events directly from `/dev/input/` this can still be used to wake the screen.

PS the alternative pip install from a branch (way I used to do it)
```bash
cd ~
git clone https://github.com/tipam/pi3d
cd pi3d
git checkout develop
source /home/pi/venv_picframe/bin/activate
python -m pip install . #NB there's a full stop on the end of this line to mean 'this directory'
```
or `pip uninstall` prior to running the command to install from a specific github branch.

pyrced, 2025-03-06:
I tested out the display off and on issue with my installs.
When using wayland I have noted the same this as you, the screen won't turn back on with mouse movement with sdl2 set both true or false. With sdl2 set true, to simulate the display going off from the menu command and given I can't use the menu due to mouse cursor location, I turn off the display executing wlr-randr --output HDMI-A-1 --off over ssh as I believe this is the same command executed looking at the code, and then the display won't turn back on with mouse movement either. If I have sdl2 set false I can use the menu with the mouse and turn off the display and again it won't turn back on with mouse movement. It does turn back on executing wlr-randr --output HDMI-A-1 --on over ssh.

I tried this with x instead of wayland and found something different. If I have sdl2 set false I can turn off the display with the mouse and it will turn back on with mouse movement. If I set sdl2 to true because I can't use the menu due to the pointer location issue, I tried turning off the display executing the command xset -display :0 dpms force off over ssh as this appears to be the command in the code. The display turns off but soon turns itself back on without any mouse input.

paddywwoof, 2025-03-06:
The screen turns back on because there is usually minor single pixel movement of the mouse. I will add some filtering to only switch back on if there is more determined movement. I will also make the touch events "debounce" so they can't happen more often than once per second.

closed 2026-05-06 by github-actions[bot]