helgeerbe/picframe #436: Frame not changing pictures automatically

haloeight, 2025-01-12:
I am not sure what is causing this but it started just recently. Attached is my configuration file if relevant but not sure how to troubleshoot this and would appreciate any guidance or pointers to existing resources. Thank you. 

viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 5.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: True                      # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 60.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 20                        # default=40, text character size
  show_text: "caption name folder"        # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "C"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: True                               # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: 1080                         # width of display surface (null->None will use max returned by hardware)
  display_h: 1920                         # height of display surface
  display_power: 2                        # default=0. choices={0, 1, 2}, 0 will use legacy `vcgencmd`, 1 will use `xset`, 2 will use 'wlr-randr' to blank the display
  use_glx: False                          # default=False. Set to True on linux with xserver running. NB use_sdl2 might need to be False for this to work.
  use_sdl2: True                          # default=True. pysdl2 can use display without xserver, it should be installed as a dependency of pi3d
                                          # but might need `sudo apt install libsdl2-dev` if picframe gives errors about missing sdl2

  mat_images: False                       # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: False            # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: True                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 60                      # default=120, clock character size
  clock_format: "%-I:%M"                  # default="%-I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay
  clock_top_bottom: "T"                   # default="T" ("T", "B"), whether to display clock at top or bottom of screen
  clock_wdt_offset_pct: 3.0               # default=3.0 (1.0-10.0), used to calclate pixels between clock text and side of screen
  clock_hgt_offset_pct: 3.0               # default=3.0 (1.0-10.0), used to calclate pixels between clock text and top/bottom of screen
                                          # If text is found in the ramdisk /dev/shm/clock.txt then it will be displayed under the time

  menu_text_sz: 40                        # default=40, menu character size
  menu_autohide_tm: 10.0                  # default=10.0, time in seconds to show menu before auto hiding (0 disables auto hiding)
  geo_suppress_list: []                   # default=None, substrings to remove from the location text

model:
  pic_dir: "/home/pi/Pictures"                   # default="/home/pi/Pictures", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow_links to True to visit directories pointed to by symlinks, on systems that support them.
  no_files_img: "/home/pi/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 7                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
  time_delay: 60.0                       # default=200.0, time between consecutive slide starts - can be changed by MQTT
  fade_time: 2.0                         # default=10.0, change time during which slides overlap - can be changed by MQTT"
  update_interval: 2.0                    # default=2.0, time in seconds to wait between two consecutive scans for new files
  shuffle: True                           # default=True, shuffle on reloading image files - can be changed by MQTT"
  sort_cols: 'fname ASC'                  # default='fname ASC' can be any columns in the table with optional ASC or DESC separated by commas
                                          # fname, last_modified, file_id, orientation, exif_datetime, f_number,
[... 53 lines cut ...]
  certfile: "path/to/cert.pem"            # server certificate

peripherals:
  input_type: null                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
[... cut ...]
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"

paddywwoof, 2025-01-12:
Hi, do you get a different image each time you start? Can you change the image manually (ie with http)?

I would probably set the `log_file: "/home/pi/log.txt"` and look in that after the bad behaviour. You might need to ser log_level to DEBUG if you don't see anything useful.

haloeight, 2025-01-12:
> Hi, do you get a different image each time you start? Can you change the image manually (ie with http)?
> 
> I would probably set the `log_file: "/home/pi/log.txt"` and look in that after the bad behaviour. You might need to set log_level to DEBUG if you don't see anything useful at first. NB remember to turn off logging to file once you're sorted!

Thank you for your help. I do get a different picture on restart. I will try what you suggest and report back.