helgeerbe/picframe #438: Multiple Monitors

StonyManager, 2025-01-25:
I'd like to run two monitors/picture frames from on RPi 5, each retrieving pictures from different directories. Would it be reasonable to run two copies of the picframe service, one for each monitor? Or is there a better/more elegant solutions? Thanks.

paddywwoof, 2025-01-25:
I have run pi3d in two processes OK but not with the latest versions of the OS or with Wayland compositor. I would be interested to hear how you get on. Let us know of any problems you hit.

David53fowler, 2025-01-27:
Hello,

    I sent the attached following email  to Wolfgang Maennel hoping he could help me get out of a usage hole I created while attempting to run 2 independent picframe processes on 2 HDMI monitors attached to a Raspberry Pi4B.  He recommended I contact you for help by attaching my email.  I got close to achieving my goal, but failed.  Hopefully, you help me resolve my issues and generate success.

 Hello Wolfgang,

     Using your amazing blogs on building a digital picture frame with Pi3D and Raspberry Pi4, I have made substantial progress in achieving my ultimate goal: to drive 2 independent monitors with a single Pi4 where each is running a digital slide show with Pi3D/Picframe using different image folders stored within "Pictures" on the micro SD card file manager.  
      What I seem to have done correctly:

    I installed Samba and transferred images from my PC to the SD folder, "Pictures" on the Pi4 and created subfolders for the 2 image streams.
    I installed Pi3D/Picframe successfully and got one digital picture frame functioning after creating venv_picframe folder and picframe_data folder, where I modified the configuration.yaml file to get the display to my desired liking. I ran the program by using CLI with ./start_picframe.sh
    I was able to connect the 2nd HDMI monitor and run 2 versions of the picframe, one on each monitor.  Since the slider show was selecting randomly, I could verify that the 2 instances of the program were running independently.
    I set up a new version following your blog directions (as if I was doing it for the first time) by naming things venv_picframe2 and creating a new configuration2.yaml in the picframe_data/config folder.  I modified the Model.py file in the Python3.11 folder of this new virtual environment to look for this new configuration file instead of the original one. This second configuration file referenced the second subfolder of images for picframe2, as opposed to the first subfolder of images used for picframe.
    I successfully could start both picframe running images from folder1 and picframe2 running images from folder2; however, they ran on top of each other on HDMI1 monitor, with HDMI2 monitor sitting idle.  This was true whether I set the x-axis image start pixel to 0 and 0, or 0 and 3841 (my screen pixel size), respectively, in configuration.yaml and configuration2.yaml.
    I was initially successful at setting up the labwc/autostart script for start_picframe.sh and creating picframe.service within systemd/user on Pi4.  I also was able to include start_picframe2.sh as a second command in the labwc/autostart file.  Upon reboot both versions, picframe and picframe2, started; however, they still ran on top of each other on HDMI1 monitor.
    Next, I tried to build a second version of the service, using picframe2.service.  This is where everything kind of went to hell, and I am stuck.  I could really use your help and guidance to dig my way out of this hole.

    When I attempted a reboot with this last setup of autostart with 2 service files defined, the Pi4 came up with only black screens and no mouse or keyboard functionality (I have keyboard and mouse directly connected to the Pi4 and run the CLI directly. By removing power (without appropriate shutdown or reboot) and restarting, Pi4 came back to life, but would no longer seemed to attempt autostart script (more in a minute).
    I removed the 2nd service file, picframe2.service and commanded DISABLE for both picframe2.service and picframe.service. Then, I re-ENABLE picframe.service.
    Now, autostart for picframe.service attempts to run (previously I think it was, but did not see error code that time), but fails with the following error:

           DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
            Failed to create scanout resource

I could start over from scratch, but I am hoping a recovery can be made that is less onerous.  Once I recover the original function of running a single digital picture frame on the Pi4 (hopefully with your help), then I will still not have achieved my original goal.  So, how do I tell the code to put one version of picframe on HDMI1 and put a second independent version o picframe on HDMI2?

Sincerely,

David Fowler

David53fowler, 2025-01-31:
Hoping I can get a response/help on solving my issues with running 2 "Picframe" processes, one on HDMI1 and one on HDMI2 for Pi4B.

I resolved the error described in my first post to this issue case.  It occurs when specifying a pixel value greater than the size of the monitor.  Keeping the x-start and y-start values to < my monitor resolution removes the error and the "picframe" process runs appropriately.

I reconfigured a single virtual environment in Python for running 2 processes, "picframe" and "picframe2", each having a unique yaml configuration file.  Within this environment I am able to start both processes, manually.  They will run on different monitors if the mouse is residence on each monitor when the different picframes are started.

I can use an "autostart" file within Labwc  to start both picframe processes upon rebooting the Pi4B unit.  However, they always both show up on HDMI1, one ontop of the other, with HDMI2 only showing the desktop background.

Can you provide any guidance for getting my "picframe" to start on HDMI1 and my "picframe2" to start on HDMI2 upon booting the Pi4B up?

Thanks for any help you can provide.

David Fowler

paddywwoof, 2025-01-31:
@David53fowler well done getting it *nearly* working. Is the HDMI set as part of the systemd labwc autostart setup. Can you get both outputs to go to HDMI2?

David53fowler, 2025-02-01:
I don't know anything about labwc other than what is provided in Wolfgang's and Helge's blogs on setting up picframe.  Thus, I have defined no settings related to HDMI.

My systemd labwc autostart file is 

    /home/david/start_picframe.sh
    /home/david/start_picframe2.sh

My start_picframe.sh is

     #! /bin/bash
     source /home/david/venv_picframe/bin/activate
     picframe &

My start_picframe2.sh is

     #! /bin/bash
     source /home/david/venv_picframe/bin/activate
     picframe2 &

Previously, I have been able to put both outputs on either monitor if I start them from the CLI and have the mouse located on the respective monitor.  In the process of responding to you I went through a complete shutdown, unplug, and repowering of the Pi4B.  Unfortunately, this has brought back the original error I was seeing related to memory allocation.  I changed no code in between, so I have to assume the recycling of the Pi4B caused a corruption of code somewhere, and I really don't understand what the actually cause of this error is.

helgeerbe, 2025-02-01:
Hi David, never tested it with multiple monitors.
First I would use different config files for Picframe. Use different db files, log files, picframe ids.

Try something like this.

Under Wayland with labwc, you can launch a program on a specific monitor using environment variables or labwc-specific methods. Here are some ways to do it:

1. Set XDG_SCREEN (if supported)
Some applications respect the XDG_SCREEN environment variable:

XDG_SCREEN=HDMI-A-1 firefox
Replace HDMI-A-1 with the correct name of your desired screen (you can check with wlr-randr).

2. Use WAYLAND_DISPLAY (for multiple sessions)
If you're running multiple Wayland sessions, you can specify which one to use:

WAYLAND_DISPLAY=wayland-1 someapp

For xwayland you could try in the start bash:
#!/bin/bash
labwc &  
sleep 2  # Wait for labwc to start  
firefox &
sleep 1  
xdotool search --onlyvisible --name "Firefox" windowmove 1920 0

I have no idea, if something of this will work.

paddywwoof, 2025-02-01:
Another thing that occurred to me is that with the desktop running, there have already probably been lots of clever bits of program executed to detect the monitors and map the desktop over them  both. I think that was the mechanism behind the process of setting the X value of the pi3d `Display` for the second monitor to the right hand side of the first monitor.

It might be worth trying that approach. picframe would probably have to be running as a window rather than full screen. I haven't got a second screen (well, I haven't got a way to connect it to the micro HDMI) but I will see if I can get the display to overflow the first display. Then I will try starting two instances in different locations, then I will see if I can remember how to start each process after the desktop boots!

paddywwoof, 2025-02-04:
So, the good news is that I found a micro hdmi to vga adapter and now can run two monitors from the RPi5. So far I have done this:
1. vanilla 64 bit full desktop OS
2. messed around running pi3d on it's own with virtual env during which I `sudo apt install libgfortran5 libopenblas0-pthrea` in order to get numpy working, but this might have been a red herring as I'm not sure it's needed for normal picframe setup. I also `apt install libsdl2-dev` which, again, I'm not sure if it's needed for normal picframe setup, could try without and fix after if there are error messages.
3. with the venv source selected I `python -m pip install picframe` then ran default setup with `picframe -i /home/pi`
4. I then copied `/home/pi/picframe_data` to `/home/pi/picframe_data2` as well as `Pictures` and `DeletedPictures` to `Pictures2` and `DeletedPictures2`
5. in `/home/pi/picframe_data2/config/configuration.yaml` I changed the paths for the locations of pic_dir, deleted_pictures and db_file. And anything else you don't want to be shared between both instance
6. CRITICALLY I changed
```yaml
  display_x: 1920                         # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: 1280                         # width of display surface (null->None will use max returned by hardware)
  display_h: 1080                         # height of display surface
``` 
7. to match the two monitor sizes
8. I created a bash file
```bash
#!/usr/bin/env bash
picframe /home/pi/picframe_data/config/configuration.yaml &
picframe /home/pi/picframe_data2/config/configuration.yaml &
```
see video at https://youtu.be/yPUaVW0Q8qY

I just need to check out running the bash straight after the desktop loads, and I suspect we will be back to square one with the screen blanking, but maybe something will work more easily because the whole desktop is running.

Paddy

David53fowler, 2025-02-04:
Update from my efforts:

1. I tried your suggestion to make windows instead of full screen.  This does work, but leaves title bars on the screen, which I consider unacceptable for a digital picture frame.  Thus, I dropped pursuing this approach.
2. I did not really attempt to pursue the environment variable setting that you suggested (XDG_SCREEN) because I could not determine where to actually build this in to the code.  I did attempt some variations trying to set the environment variable in the "Environment" file of Labwc, which locked the entire system up, causing me to have to reload the PiOS system from scratch.  Very frustrating.
3. I asked ChatGPT to write the code to auto start 2 python processes directed to the 2 monitors.  In seconds, it came back with a "beautiful" complete code to make it happen.  It even looked like it might work, at least to my near nil Linux experienced eyes.
4. ChatGPT provided 3 service files, one for each picframe process and a 3rd one that defined the monitor setup.  It ran them out of the systemd root folders instead of the systemd user folder, as prescribed by your original setup blogs.  I tried it both ways with the same outcome, as far as I can tell.  This forced me to actually learn some Linux.  This code did not utilized an Autostart file in Labwc.
5. The ChatGPT code as provide threw several errors upon running it.  I resolved a couple through my own by trudging through internet comments and posts.  I got additional feedback from ChatGPT that resolved an error for XAuthority keys, but I could not get code to successfully work in the end.  Although I could see from the journalctl log (a ChatGPT suggestion) that the 2 processes were attempting to run from the service file and being directed to the correct monitor, but failed to actually get to the screen.
6. Here are some key snippets from ChatGPT (after my mods and GPT's mods):

**picframe.service:**

[Unit]
Description=Run Process 1 on HDMI-A-1
After=graphical.target

[Service]
ExecStart=/usr/bin/python3 /home/david/start_picframe.sh
Environment="DISPLAY=:0"
Environment="XAUTHORITY=/home/pi/.Xauthority"  
WorkingDirectory=/home/david
Restart=always
User=david

[Install]
WantedBy=default.target

**picframe2.service:**

[Unit]
Description=Run Process 2 on HDMI-A-2
After=graphical.target

[Service]
ExecStart=/usr/bin/python3 /home/david/start_picframe2.sh
Environment="DISPLAY=:0"
Environment="XAUTHORITY=/home/pi/.Xauthority"  
WorkingDirectory=/home/david
Restart=always
User=david

[Install]
WantedBy=default.target

**set_displays.service:**

[Unit]
Description=Set up HDMI Displays
After=graphical.target

[Service]
ExecStart=/home/david/set_displays.sh
Environment="DISPLAY=:0"
WorkingDirectory=/home/david
User=david
Restart=on-failure

[Install]
WantedBy=default.target

where **set_displays.sh:**

#!/bin/bash
# Set HDMI-A-1 to primary screen
xrandr --output XWAYLAND0 --primary --mode 3840x2160 --rate 30 &
# Set HDMI-A-2 to second screen
xrandr --output XWAYLAND1 --mode 3840x2160 --rate 30 &

8. ChatGPT originally stated output was to be labeled as HDMI-A-1 and HDMI-A-2.  This did not work, but I found the correct label by running xrandr --listmonitor.
9. The error that still exists has some to do with "authorization to open DISPLAY :0".  ChatGPT began to repeat old ideas/solutions when I asked it to fix this.  One of my "fixes" was to swap "multi-user.target" for "default.target" in WantedBy.  Even though this stopped an error from be thrown, I am not certain it does the right thing.

It sounds like you are having good progress with your approach, so further attempts to get ChatGPT approach to work seems to be unnecessary at this point. Consider it my "Ode to AI Dreams".

paddywwoof, 2025-02-04:
David, the window borders will only go away after you modify pi3d DisplayOpenGL.py or, probably better, wait for me to update the repository.

ChatGPT is pretty amazing with suggestions but I've gone round in circles trying all kinds of plausible routes with little success!

paddywwoof, 2025-02-04:
I've bumped pi3d to v2.53 with the `SDL_WINDOW_BORDERLESS` functionality, so that should work OK now. You will need to
```bash
source /home/pi/venv_picframe/bin/activate
python -m pip install pi3d --upgrade
```
Let me know what further issues raise their head!

Paddy

PS there are a few more significant changes (some probably breaking and maybe requiring fixes in picframe!!) due in pi3d around the harmonization and general improvement to the Keyboard and Mouse objects but I will hold back issuing those for a while.

David53fowler, 2025-02-05:
Hi Paddy,

  I did the Pi3d upgrade and I do get borderless windows, as promised.  With this change, I can now position a full screen picframe display on HDMI-1, and a picframe2 display window that essentially fills the screen on HDMI-2.  The full screen display is required to cover the desktop toolbar that is currently on HDMI-1 (see below).

The trouble starts/returns when I attempt to autostart the Pi4.  I returned all the autostart code back to your recommended blog installation files, so it should run in the normal way you have prescribed.  However, I am fairly certain that is an issue with the labwc package. 

 If I disable the picframe.service file, which calls labwc, then I can get the 2 picture frames started and running from boot up (see below), although there are some small issues with expected text not being displayed and the smoothness/jerkiness of the image transitions.

If I enable the picframe.service file and reboot, the first issue is that the autologin no longer functions and I have to type in my user password to start the displays.  If the display windows and full screen are as I used in the manual run, then the .Xsession.log throws my "old friend" error:

     DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
        Failed to create scanout resource

I now believe (can you really trust a Linux complete novice?) this is tied to the labwc package because the journalctl log of the picframe.service file throws the following error every few seconds out to infinity:

      "failed to commit frame"

I attempted to look this up on blogs and found only 2 references (actually one reference on RaspberryPi forum and repeated on GitHub) for a new issue raised on 12/24/2024.  My log file looks exactly like the one reported there.  There is some agreement that this is a critical flaw with the current labwc, but there is no indication of any resolution.  The indication is that the labwc service and the 2 picframe processes are competing for memory for my case (the reported case uses different processes of course), but the labwc memory grab is blowing them out of the water.  From my experience, I think I was not having any issue until I began trying to run the second picframe process on the 2nd monitor.

So, a question.  Since I get some desired 2 monitor functionality and the original picframe app on 1 monitor both seem to run without the initiation of the labwc through the service file, then what purpose does starting up this service fulfill?

In the meantime, you can see what I have been laboring to bring to life:

[image]

paddywwoof, 2025-02-05:
Wow those frames look very impressive!

My approach was to use the built in functionality of the Linux desktop system allowing two monitor real estate. The problem is that all the stuff with labwc and systemd evolved to run picframe as light weight as possible without needing the desktop so I suspect you may get a conflict if you mix them. The autostart.desktop method is very easy to set up but probably has other disadvantages (I've not tried stopping and restarting from SSH terminal for instance.)

I'm not sure what the text display issue is. Let me know as it might indicate some problem I've missed. For the jerkiness of transitions: that can happen when there is a large CPU or input output load happening at the same time as the image transition and inside picframe we delay the fade until after the image has loaded. (or vice versa) With two processes running this would need a semaphore system operating between them, maybe only a line or two of code with a ram disk file (rather than proper socket communication) Let me know if this is worth thinking about

David53fowler, 2025-02-10:
After setting up to 2 displays for the ultimate way I plan to use them; meaning images only transition ~ every 20 minutes and at different times described by 2 prime numbers, the jerkiness of transition completely goes away.  This part works great, now.

I am continuing to running the desktop, but without keyboard or mouse. I now only communicate to the Pi4 through SSH.  Early on in this development, I once tried to turn off the desktop, and in that one trial the picframe app did not start, so I thought I needed to have the desktop for things to work. In any case, I currently do not start labwc through systemd service file and only run the autostart file to start the 2 separate picframe processess at boot up.

So, at this point, although not critical to my usage, I never see text on the screen for the primary display that is running picframe in full screen mode (to avoid desktop toolbar overlay), and I only intermittently see it in the window mode of the secondary monitor running the 2nd picframe process w/different configuration file.  The text is sometimes very dim and grey (this seems to show for entire image display time) and once in a while bright and white(this shows for a very short time, much less than the configured 30 sec).  I am only configured for the filename and a caption name.

At present, I consider that I am in production mode for this project until I think of another change to make.

paddywwoof, 2025-02-10:
Hi, I'd like to sort out why the text is misbehaving. My testing was using a Pi5 and relatively small images so maybe I should check what happens with something nearer to your setup. I didn't set any dimensions or positions in the yaml file for the first screen, is that what you mean by full screen? There shouldn't be any borders either way.

David53fowler, 2025-02-11:
Yes.  I set up the full screen on the primary by configuring display_x: 0, display_y: 0, and display_w:3840, display_h:2160 (my full screen).  This yields the image without any borders or interference with the desktop, as desired.  However, if I define a window for the primary display by changing display_w: 3839, or display_h:2159 (like I do for the secondary display), then the toolbar for the desktop overlays on the image window.  Because I was getting the text display to work for a window (sometimes intermittently), but not full screen, this is why I was trying to use this path.

paddywwoof, 2025-02-12:
David, I've tested this on a Pi4 and it seems to work OK, apart from being a bit slower to start etc. In the process I've found a few quirks. The first thing is that the mod I made to pi3d had a typo in that the masking to test for `DISPLAY_CONFIG_NO_FRAME` should have been `&` instead of `|` (bitwise AND not OR). But that means that with the typo the condition will **never** be True so windows will always be BORDERLESS. When I correct this the line, creating the Display object in picframe will need to be changed too. I also noticed that the text can still show at a very low alpha value instead of fading away completely if the time is short. There is a minor fix for this too.

The desktop top bar will be concealed beneath a window if it is created with the FULLSCREEN flag set. This will happen in pi3d if x and y locations are 0 and the width and height are not specified. Alternatively if the width and height match the display width and height returned by the window manager. There doesn't seem to be an option to create sized windows that overlap the desktop top bar.

1. For the primary screen set configuration.yaml as (`display_x: 0, display_y: 0, display_w: null, display_h: null`)
2. For the secondary screen set `display_x` equal to the width of the primary screen and `display_w` and `display_h` to the dimensions of the secondary screen

I haven't managed to reproduce your text overlay issues. The text should fade from 0.0 to 1.0 alpha for 2.0s (in your case) stay at full alpha for 24.0s then fade to 0.0 for the last 2.0s. In theory, that's what it does on my screens here!

I will update pi3d and picframe once I've done some work on the `Keyboard` and `Mouse` classes

Paddy

David53fowler, 2025-02-14:
Paddy,  The issue with text not being displayed appears to be related to how I am running 2 unique picture frame processes.  When only one is running the text appears to work appropriately.  When both are running, it looks like some interference is created that corrupts the text display code.

To clarify how I am implemented:  I set up venv_picframe as prescribed in your blogs.  Then, I make a 2nd picframe folder within by making a copy of the picframe folder and renaming all relevant references I could find in the python code and in the naming convention of folders to picframe2.  I make a new configuration2.yaml file in picframe_data.  It is quite possible that I have missed something that needs to be changed.  Nevertheless, I initiate the 2 different picture displays, picframe and picframe2, from an autostart file in the labwc folder.

David53fowler, 2025-02-15:
Paddy,

    Is there a way that the picframe code can be modified such that you can start 2 instances of the same app, but allow a unique configuration file to be called upon the initiation of the process?  I know I can start 2 instances of picframe and have them run in parallel, but I think some change would be needed to be able to specify which configuration is to be used upon start.

paddywwoof, 2025-02-15:
David. Yes that's what I did. See my earlier post https://github.com/helgeerbe/picframe/issues/438#issuecomment-2634208619 I installed picframe **once** then copied the resulting picframe_data dirctory. I didn't do anything with labwc, all the autostart is done with the .desktop file.

David53fowler, 2025-02-16:
Paddy,

    I sincerely apologize for not paying closer attention to your previous comments/instructions.  I guess I just got onto a track that was kind of working when I was exploring solutions with "my local blinders" on. I have now implemented the your method using the "start_picframe.desktop" running 2 instances of the same picframe application, but with different picframe_data configurations, as you prescribed.  Everything has started up as expected.  So, now I will monitor it for total success.

You have been extremely helpful, and I am sorry my naivety in Linux caused this misunderstanding.

David

paddywwoof, 2025-06-01:
Josh, not sure what it could be. Did you say that you could get it to manually work fine to start with, but went wrong when you tried to get it to run on startup? If that's the case you need to try to return to that state. Worry about the autostart later. Does the desktop spread over both screens?

Paddy

joshcliffejones, 2025-06-01:
After some more fiddling around...I think it finally have it working correctly and consistantly! My sinceere apologies if this is a "yeah duh" moment...

But I switched from the Wayfire to X11 backend found in the raspi-config settings under advanced options > Wayland.

Th moment I did this and rebooted, I ran the start_picframe.sh script and it worked! So, re-introduced the autostart process referenced above, rebooted and it also worked!

I have rebooted several times so far and it has continued to work so far!

@David53fowler The above eurika moment of switching to X11 was down to your comment and my research into setting up a single output that spanned across both displays. I read that Wayland currently has a glitch with this practice and that it was better switching to X11. So just thought Id give this single change a go first before delving further and voila, it worked!

@paddywwoof Suddenly your comment of no labwc also makes sense as I noticed this in the options too which I hadnt explored previously, I was just working off a clean vanilla build of Rasperry PI OS.

A huge thank you to you both for your guidance and assistance, I will keep you posted if anything changes as I test further.

Now I just wish that I could output to more displays from a single Pi 😛

paddywwoof, 2025-07-08:
Probably simpler in the long run (hang the expense!).

When you want to split a picture into four and display each part as a super display you will have to work out coordination between them! Especially tricky for videos.