helgeerbe/picframe #439: New Installation Pains on RPi4 with Touch Display 2

j3tang, 2025-02-07:
Hi everyone,

I recently came upon this great project and quickly had my Pi4 setup up and running using the amazing "one click" method here:
https://www.thedigitalpictureframe.com/install-the-pi3d-pictureframe-software-with-one-click-2025-edition-raspberry-pi-2-3-4-5/

I connected the Pi via a mini-HDMI to HDMI cable to a spare TV and everything worked great.
Then I went and bought the official Raspberry Pi Touch Display 2 and had it connected to the Pi.

https://www.raspberrypi.com/documentation/accessories/touch-display-2.html

The thing with the Touch Display 2 is that it is advertised to run in portrait mode and to rotate it, there were some instructions towards the bottom of that page to edit a .txt file and to provide the desired degrees of rotation.  However, I've tried editing the file as described, but it doesn't rotate the screen at all with any of the numbers 90, 180 or 270.  

At the moment, the Pi currently is directed to a folder that just contains portrait photos to make it useable :)
Does anyone have any experience with the Touch Display 2 and getting it to display in landscape mode?

Another newb question is in regards to the article here:
https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/

The frame is currently connected to my HA instance that runs the MQTT broker (on a different Pi) and has all the basic integration/functionality.
However, the "current_image" part in the article above does not seem to work.  Basically, when I enter http://MY_IP:9000/current_image it is not able to load up the image on the picture frame.  Unfortunately, the article does not offer up any troubleshooting steps ... anyone have any suggestions?

helgeerbe, 2025-02-07:
Hi @j3tang,

see https://github.com/helgeerbe/picframe/issues/401 . Are you able to rotate your screen with `wlr-randr --output HDMI-A-1  --transform 270`

j3tang, 2025-02-07:
I gave that a try and the output was:

*unknown output HDMI-A-1*

I'm assuming it's because the touch display is connected through the DSI
cable and not via the mini HDMI ports?

helgeerbe, 2025-02-07:
Try `wl-randr` without parameter. It should list the available output devices.

j3tang, 2025-02-07:
ok, so the output device is 'DSI-1'.  I substituted that into the command above and does something funky.
Since the screen resolution is 720x1280 (in portrait), what I'm guessing it is doing is it rotates it to 1280x720, but the image is still only occupying the first 720 (in  width) on the left of the screen.  So the actual image size is 720x720.  And it does that by scaling the landscape photo to 720 width, but maintaining the original height (?) such that the top and bottom of the photo is not viewable.

And since a picture is worth a thousand words ... here are 2 thousand words ... the first being what the photo is when viewed through the MQTT integration on HA and the second is the view of the picture frame IRL

[image]

[image]

helgeerbe, 2025-02-07:
Have you set the paramenters in configuration.yaml? https://github.com/helgeerbe/picframe/issues/401#issuecomment-2476757292

j3tang, 2025-02-07:
ahh ... ok, so I set the display_w and display_h in the config file and then applied the rotation (manually in SSH) and it all works!

looking at the solution provided by adammhaile, it also indicates to put the rotation into the start_picframe.sh file.
when I go into my file, it is completely blank ... even though it sounds like he is suggesting that the sleep and rotate command should be appended to the end of the file.  Am I missing something because it feels like maybe it's not calling on that file when I start up my device?

ps. what is the "sleep" command supposed to accomplish in this instance? is that a delay?

paddywwoof, 2025-02-07:
Hi, sorry not to contribute earlier but didn't want to struggle with doing stuff on phone. If you work your way though Wolfgang's clever but rather opaque script you should see it generating start_picframe.sh as
```bash
#!/bin/bash
source /home/pi/venv_picframe/bin/activate  # Activate Python virtual environment
picframe &  # Start Picframe in the background
```
it seems odd that it's empty.

j3tang, 2025-02-07:
oops, user error ... it's not blank .. all is good now :)

paddywwoof, 2025-02-07:
Excellent. Once I've sorted the keyboard and mouse behaviour for the move to SDL we should group all these threads in some kind of ref document.

j3tang, 2025-02-07:
Does touchscreen functionality get lumped into that as well? :)
Would love to put the Touch Display 2 to full use!

Also, as I was testing out the commands above, I noticed that as long as the display_w and display_h were defined within configuration.yaml, whether or not I used the --mode parameter didn't seem to make a difference.  Is this correct behaviour? or is there something behind the scenes I'm not seeing?  Seems odd to have to provide this info twice and could be a potential source of conflict/error.

paddywwoof, 2025-03-04:
Hi I've put the modifications to mouse and keyboard into development branches. They seem to work with mouse and keyboard but I don't have a touchscreen so can't check that out. If you get a chance you can have a go probably by doing something like
```bash
source /home/pi/venv_picframe/bin/activate
python -m pip install git+https://github.com/helgeerbe/picframe@dev
python -m pip install git+https://github.com/tipam/pi3d@develop
```

closed 2026-05-06 by github-actions[bot]