helgeerbe/picframe #440: keyboard controls

Sachicon, 2025-02-07:
What am I doing wrong?

Everything works grand in my bookworm setup and configuration except for keyboard input.  With peripherals: input_type: "keyboard" enabled at the end of configuration.yaml, keystrokes a, d, space, e, or any other keypress freezes the slideshow and produces this traceback:

Traceback (most recent call last):
  File "/home/pi/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/usr/local/lib/python3.9/dist-packages/picframe/start.py", line 136, in main
    c.loop()
  File "/usr/local/lib/python3.9/dist-packages/picframe/controller.py", line 298, in loop
    self.__interface_peripherals.check_input()
  File "/usr/local/lib/python3.9/dist-packages/picframe/interface_peripherals.py", line 78, in check_input
    self.__handle_keyboard_input()
  File "/usr/local/lib/python3.9/dist-packages/picframe/interface_peripherals.py", line 251, in __handle_keyboard_input
    if len(code) > 0:
TypeError: object of type 'int' has no len()

Keyboard input worked after the bookworm related update in June 2024 but not in the earlier or later ones.  I might mention, I just did a fresh setup of Picture Frame on a 2022 image of bullseye I had archived and now get the same results with keyboard input where they worked in the past.

paddywwoof, 2025-02-07:
Hi, I think the problem is essentially due to the way SDL keyboard in pi3d does things differently from the X11 server. See https://github.com/tipam/pi3d/issues/264

This is top of my priority list at the moment, but asi think it will be a breaking change, I want to minimise destruction!

Paddy

Sachicon, 2025-02-08:
Thanks for that.  Good to know; I'll not spend more time on trying to work it out as if it's something I'm doing wrong.

closed 2025-02-08 by Sachicon