helgeerbe/picframe #443: Picture Frame Stops

YYC-hris, 2025-02-19:
I have three RPi Zero 2W's on three separate monitors.  They are all set up the same.  One of them has been running for a few days with no problems.  Another one will run for around a day before it stops (I get a black screen like the program has terminated).  The third will only run for an hour before the program appears to terminate.  I put this third one back to the default configuration.yaml file, and it still has issues.  The only change I made to the default is to change pictures every 30 second, so I could get to an error condition quicker.    

I've also set up logging in the configuration.yaml for debug.  I'm not seeing anything in the file to indicate what the issue is, but I'm not much of a programmer.  One odd thing I see is I get errors about image_cache.Imace Cache - Warning and then it says the image does not exist or is inaccessible.  The imagines it list are ones that used to be on the SD card, but haven't been there for days.  I've done many reboots, so I don't know why those would still exist in cache?  There are many errors like this, but the picture frame runs for many minutes after these last errors are generated.

I do have some pictures with a weird aspect ratio (very tall panorama's), and I'm wondering if that could be the problem.    I can't seem to find a way for Picture Frame to log which picture it is showing to see if it always stops at the same spot (I suspect it does).  

Any advise on troubleshooting appreciated......and thank you to the people who have put this great program together!

paddywwoof, 2025-02-20:
Hi, the Pi Zero has more limited memory and processing (than 3,4 or 5) so it might be causing a problem if an image is being resized or blended with a frame while another image is being loaded. Are the three instances running different pictures or does the same image show on one and crash another one.

What might be useful is if you look at the sequence that stops reliably at the same place when sorted alphabetically. Make a subfolder `~/Pictures/test/` and move into it the last two images that show and the two that should come after it. You might need to delete the database file (`rm ~/picframe_data/data/pictureframe.db3` I think but check what it's really called) so picframe will reindex the new locations correctly.

You can make picframe send copious debug info to a file by setting this `configuration.yaml` i.e. `log_level` and `log_file`. Remember to change it back after debugging to WARNING and the file back to `''` otherwise it will fill up your SD.

It's worth looking at the files before and after picframe stops to see if they are special in any way.

Also, assuming you have started picframe using systemd, if you type `journalctl` on the command line then End to the bottom of the file, you might see some kind of error message.

If the issue does turn out to be image processing on the fly, it's very easy to write a program to resize them all as a batch.

YYC-hris, 2025-02-20:
Thank you for the ideas for me to pursue!  

The 3 RPi's are running different pictures.  You asked if I'm running picframe using systemd.  I'm not sure.  I will have to dig into that.  I ran the scripts on [this page](https://www.thedigitalpictureframe.com/install-the-pi3d-pictureframe-software-with-one-click-2025-edition-raspberry-pi-2-3-4-5/) to set up pictureframe.   

I'm just starting to dig through the db3 to assess the sequence and see if it is a particular picture causing problems.  I will report back.   Is there any way to modify the yaml file so it logs the pictures it has displayed?  Maybe that info is buried in that file, but I have it set to DEBUG and I'm not seeing any image file names recorded.

paddywwoof, 2025-02-20:
Yes, Wolfgang's script does set up starting using systemd so `journalctl` (that's lower case L last character) might have some clues.

After a quick look it seems there isn't a logger debug in the section of code loading the images. Maybe we should add one for situations like this. To hack it yourself, you could add a line at the start of  `__tex_load()` around line 290 in `~/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py`
```python
    def __tex_load(self, pics, size=None):  # noqa: C901
        try:
            self.__logger.debug(f"loading images: {', '.join(pics)}") #<<<<<< This line added to debug badly behaved images
            if self.__mat_images and self.__matter is None:
                self.__matter = mat_image.MatImage(display_size=(self.__display.width, self.__display.height),
```
Of course it is very much frowned upon to edit installed python modules like this but it will be patched over in the near future anyway so worth a go to help solve this problem.

Paddy

paddywwoof, 2025-02-20:
Oops, I tried the debug line I typed earlier and saw that it wasn't going to work so fixed it... but forgot to edit my post.

Sorry if you tried hacking the code and it seemed to break picframe!

Paddy

YYC-hris, 2025-02-21:
Thanks for the info.  I must have typed this command wrong:  journalctl  It gives a wealth of information when I type it correct.  There are almost 200,000 lines in that file?!  Doesn't it ever erase? I saw stuff from beginning of February.  I see an  "Out of memory: Killed process 590 (picframe) total-vm:1477088kB, anon-rss:228288kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:1976kB oom_score_adj:200" at a time where I think the program quit working.  A second before this, I see "mosquitto invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0" and some stuff about ram swaps.  There are many lines of info, I just don't know what is relevant.  

I tried putting only a few pictures in the Pictures directory and only putting the ones I think that cause the problem (the ones before and after).  I was unsuccessful in making it fail.  Is there any way to know from the pictureframe.db3 what it was playing and intended to play?   In the file table, there is a "last displayed" column.  Can I trust that?  Is there any way to know what it intended to play next?  I'm not confident I've flagged the right files.  

I have Picture Frame set to change slides every 30 seconds, and I have noticed that it really lags on some pictures.  It might be two or three minutes on some pictures before it changes, and then it might change in 5 seconds.   

I don't see anything useful in the DEBUG log file created through YAML.

paddywwoof, 2025-02-21:
Hi, well that does sound like some useful info, crash likely caused by memory consumption. journalctl does have a method to clear itself as it gets too big!

There was a memory leak issue previously in the first version of the Mesa GPU software, I will remind myself what that was. But the fact that you're getting different behaviour with different sets of pictures makes me think it's something specific to some of those. Did you try hacking viewer_display.py as per my previous comment? That should give info on which file is about to be opened. You get 5 to 10 lines of text with each image so the log file wouldn't be impossibly big.

sapnho, 2025-02-21:
> If the issue does turn out to be image processing on the fly, it's very easy to write a program to resize them all as a batch.

I would definitely also try Paddy's point. I resize all my images to the maximum dimension of the display, so that no conversion is necessary. And never, did I have a black screen.

paddywwoof, 2025-02-21:
I suspect there is a more professional version on thedigitalpictureframe.com but I might do something like this. Making sure there was a copy of the files elsewhere, just in case!!
```python
from PIL import Image
import os

DIR = "/home/pi/Pictures"
(W, H) = (1920, 1200)

for (dirpath, dirname, filenames) in os.walk(DIR):
        for fname in filenames:
                fpath = os.path.join(dirpath, fname)
                im = Image.open(fpath)
                scale = min(W / im.width, H / im.height)
                if scale < 1.0:
                         im.resize((int(im.width * scale), int(im.height * scale))).save(fpath)
                         print(f"resized {fpath}")
```

YYC-hris, 2025-02-22:
Thank you.  I have the python code running, and it is logging which pictures are displayed......just waiting for a crash. 

I searched thedigitalpictureframe.com for info on resizing.  I found reference to cropolla.  I uploaded one picture, and it feels like this is a cropping tool (maybe that should have been obvious from the name).  I'm not too interested in cropping my pictures to fill the frame and I would rather have the blurred bars on the side like picture frame does when blur_edges:  true.  Thus, I don't think cropolla is what I want.  I don't understand the code Paddy provided for for resizing.  Is it a "cropping" solution as well?  Is there any way to preprocess my photos and add the blurred edge to them?  Maybe I'm getting ahead of myself as I'm not even sure what the problem is yet......

YYC-hris, 2025-02-22:
The python code Paddy did list the photos in the log file I identified in configuration.yaml.   It has been crashing about two pictures before this i.e. while loading the picture right before this.  The picture in question is 10162 x 3888 pixels (a iphone pano).  I've removed that picture and I'm going to see if there are any further issues.

sapnho, 2025-02-22:
It would be easy to have a small Python script to resize the content of the Pictures folder proportionally if that's what you are looking for. You could also add a watchdog function that checks for new photos to resize automatically.

Is that what you are looking for?

I always do it from within Adobe Lightroom.

paddywwoof, 2025-02-22:
10162x3888 is **too big** (for any practical purpose?). picframe will try to scale it down to the maximum texture size the GPU can process (4000ish pixels) but that's still a heavy load for Pi zero.

As Wolfgang says, in many ways it's better to do all the rescaling before putting the images onto the SD card as it can be done on a more powerful laptop. But doing it on the Pi isn't bad:

You must have done some editing to hack viewer_disply.py so in a similar way, in a terminal on the Pi edit a new file `\home\pi\resize.py` then copy paste my script from above. Change the screen dimensions and picture directory if needed then save. In the terminal run the script with
```bash
source venv_picframe/bin/activate
python resize.py
```
An additional concern is if there are apple related files with image extensions that aren't images or problems with heif compression. It might be a good idea to copy a small number of images (ideally largest) to a different directory to test the script before running it on the full set.

sapnho, 2025-02-22:
I also added an article to the blog. I vaguely remember that we had this issue in the past already, so it's probably worth documenting: https://www.thedigitalpictureframe.com/how-to-automatically-resize-images-to-fit-your-digital-picture-frame/

I am working on making the script a bit more foolproof but it works fine here.

sapnho, 2025-02-22:
Just modified the watchdog functionality a bit more. The issue is that it starts resizing before the file has been fully transferred, which, when you have a large number of files, results in plenty of errors.

Tested it with 300 files and no issues. Would be interesting to see if Chris' black screen issue is gone now.

YYC-hris, 2025-02-23:
Thanks Guys, your support it amazing!  

I tried sapnho's code.  It worked well, but it didn't convert the HEIC files I had on the SD.  Sorry, I don't think I mentioned I have JPG's and HEIC.  I consulted with AI and it told me the code would work with HEIC if I added a library.  I installed it, but I still couldn't get it working. 
 
I asked AI if the code Paddy supplied would work with HEIC.  It suggested I was going to have similar problems.  I asked what my options were, and it gave me a windows option ImageMagick.  The following script in Powershell resized both my JPG and HEIC files, and it seems to retain the EXIF data:

foreach ($i in Get-ChildItem *.jpg, *.heic) {
    magick $i.FullName -resize 1920x1080 -quality 95 -auto-orient "D:\My Pictures\downloads\TTTTemp\$($i.BaseName).jpg"
}

The resized images were uploaded to RPi.  So far, no issues and it has run past the files that were giving issues before.  I'm feeling fairly confident that resizing was the solution!  

I am tempted to mark this closed, but I will let the RPi run a bit longer as it hasn't got through all photos yet.  I would be game to give the supplied code another try, but I would need a bit of direction on how to handle the HEIC.  I could use ImageMagick to convert everything to JPG, but if I'm doing that already, then I might as well resize with it as well.

Thanks again

sapnho, 2025-02-23:
> I tried sapnho's code. It worked well, but it didn't convert the HEIC files I had on the SD. Sorry, I don't think I mentioned I have JPG's and HEIC. I consulted with AI and it told me the code would work with HEIC if I added a library. I installed it, but I still couldn't get it working.

Good point! I just added HEIC functionality to the script and tested it. I have updated [the article](https://www.thedigitalpictureframe.com/how-i-built-a-multi-photo-digital-frame-for-a-tall-narrow-space/) accordingly.

paddywwoof, 2025-02-23:
Sounds good. EXIF and heif both need to be catered for! At some point I imagine heif functionality will be included in pillow by default in the same way jpeg and PNG are. It was a little fragile at times but seems to have settled down. Who knows, in ten years...