helgeerbe/picframe #445: Digital pictureframe MQTT with Home Assistant, can't find "devices"

Stef-007, 2025-02-24:
Hello you all,

I'm trying to connect the digitalpictureframe to Home Assistant  with MQTT and made a successfull connection but I don't see the "picframe" devices and don't see how to add them. 

[image]

In mqtt explorer (see screenshot) I get also a connection but I guess not the result I need (see screenshot below). Any idea how this comes, only thing that comes to my mind that I forgot to make a change in the yaml-file but I would n't now what. Do any of you have ideas how to solve this?

thx, Stefan

paddywwoof, 2025-02-24:
Stefan there's an mqtt section in the config file https://github.com/helgeerbe/picframe/blob/654006514513ddeb19f5ed16ba9736d0fb20f7f3/src/picframe/config/configuration_example.yaml#L103
I'm sure Wolfgang has a bit about setting it up on his website (there's so much useful info there it's sometimes hard to remember where things are)

You look to have several connections, topics and  messages on your mqtt browser. Are those other HA devices?

helgeerbe, 2025-02-24:
Hi @Stef-007,

you schould find picframe unter MQTT. I see, there is 1 item.

helgeerbe, 2025-02-25:
First of all check, if picframe is connected to your broker and is sending image data.

Check log file or run picframe from console. Right after start, do you see something like:

````log
15 09:59:43,001 - interface_mqtt.InterfaceMQTT - INFO - creating an instance of InterfaceMQTT
2022-07-15 09:59:43,013 - simple_server.InterfaceHttp - INFO - creating an instance of InterfaceHttp
2022-07-15 09:59:43,018 - interface_mqtt.InterfaceMQTT - DEBUG - Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': None, 'EXIF DateTimeOriginal': 1657470591.7937217, 'Image Model': None, 'Image Make': None, 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
2022-07-15 09:59:43,019 - interface_mqtt.InterfaceMQTT - INFO - Send image state: {'image': 'Mimi-Rainbow.jpg'}
2022-07-15 09:59:43,020 - interface_mqtt.InterfaceMQTT - INFO - Send sensor state: {'directory': 'Pictures', 'image_counter': '39', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 200.0, 'fade_time': 10.0, 'brightness': 1.0, 'matting_images': 1}
````

Stef-007, 2025-02-25:
Ok, I did a reboot and did'nt see any mqtt entries on the pictureframe, I also used windows terminal command "journalctl -b" what resulted in a very long boot/log file but no mqtt log-files either I believe, or should I sheck another way?