helgeerbe/picframe #446: Display on/off cycle resets display resolution to default and messes up image scaling

ylexus, 2025-02-24:
(This issue is raised on the back off a short discussion that was started with https://github.com/helgeerbe/picframe/issues/386#issuecomment-2676227490, but since that issue is closed, and the problem still exists, decided to raise a new issue).

Raspberry Pi / bookworm (all latest software, including picframe, as of the time of writing) , 2K or 4K display connected to HDMI-1. The display off and on command work, but when the display turns back on, it's reset to 640x480 and so the screen is empty.  This is only served when

```yaml
use_glx=True
use_sdl2: False
```

Everything works as expected if 
```yaml
use_glx=False
use_sdl2: True
```
(but in this mode touchscreen does not work at all, so I can't use it).

When I simply invoke `wlr-randr --output HDMI-A-1 --on` while the display is off, the monitor turns on in 640x480@60Hz mode with the screen being grey, just like when turning in on with picframe. However, when I do wlr-randr --output HDMI-A-1 --on --mode 2560x1440@143.998001Hz (my 2K display is 2560x1440), then it turns on at 2560x1440, but the picframe photos are now scaled down and aligned to the lower left (see image below).

Seems at least 2 fixes are required:

1. ensure picframe adds --mode parameter, the value can be guessed from the output of wlr-randr.
2. attempt to fix the wrong scaling after monitor is turned back on in use_glx=True mode.

[image]

My configuration:
```yaml
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 10.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: "title caption date folder location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "C"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: True                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  display_power: 2                        # default=0. choices={0, 1, 2}, 0 will use legacy `vcgencmd`, 1 will use `xset`, 2 will use 'wlr-randr' to blank the display
  use_glx: True                          # default=False. Set to True on linux with xserver running. NB use_sdl2 might need to be False for this to work.
  use_sdl2: False                          # default=True. pysdl2 can use display without xserver, it should be installed as a dependency of pi3d
                                          # but might need `sudo apt install libsdl2-dev` if picframe gives errors about missing sdl2

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 20                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
[... 79 lines cut ...]

peripherals:
  input_type: "touch"                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
      label: "Location"                   # default="Location"
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"
```

paddywwoof, 2025-02-25:
Odd that GLX works but SDL2 doesn't. What happens with GLX False, i.e. just using X11?

I will try putting the mode argument into the wlr-randr command, don't think it will cause problems.

ylexus, 2025-02-25:
@paddywwoof if 
```yaml
  use_glx: False
  use_sdl2: False
```

Then everything works OK too.

paddywwoof, 2025-02-26:
Hi, I was really thinking about the touch screen. Can you get that to work with SDL2. The mouse location is provided by an SDL2 function but I can't get the X11 function to work so have to drop back to reading from relevant `/dev/input/` file. If you confirm that touch doesn't work with SDL2, I will search around and see if there is any touch screen specific functions or maybe let an X11 Mouse be used with SDL2 display surface if explicitly requested.

ylexus, 2025-02-27:
I have raised another issue for the touchscreen, this one is about display on/off. In any case I have now returned the touchscreen display (it was too small) and bought a bigger 4K screen, so can't unfortunately conduct any touch experiments any longer.

closed 2026-05-06 by github-actions[bot]