helgeerbe/picframe #448: MQTT struggles..

robzonbi, 2025-03-04:
I'm having issues with MQTT commands working consistently. I'm using an MQTT app on my android phone. I'd say 1 out of 4 or 5 times I press a control, picframe ACTUALLY responds. 

I installed Mosquitto as described on Wolfgang's site. Set up the MQTT section as true and localhost. My app is sending to the correct IP, as I have set up a test topic in terminal to listen, and everytime I press the control the message appears.

I'm struggling to figure out if it's my phone, the broker, or picframe or some other unseen anomaly I'm unaware of.

I'd appreciate any advice that anyone more experienced in the field might have.

jgodfrey, 2025-03-04:
Hi @robzonbi,

First, I'll say that I've never had any trouble with my MQTT setup (also controlling from an Android phone). It's absolutely solid and responds as expected 100% of the time. So, this CAN be made to work very reliably. That said, I'm not sure what might be causing your issues.

First, a few questions maybe...

- Is the MQTT broker running on the raspi that hosts the picframe app, or is it somewhere else?
- What app are you using on your phone? In my case, I've had good luck with both "Iot MQTT Panel" (free and Pro) as well as "MQTT Dashboard (though, seems to be no longer supported or even available on the play store).
- Are all MQTT-based commands equally unreliable, or only some?
- Might it be possible to show the Android configuration page of one of the problematic commands? 

It's strange that is seems to work for you "sometimes". I'd think it'd be more likely to work always or not at all... :)

jgodfrey, 2025-03-05:
Hmmm... That all sounds pretty reasonable I think. Looking at my "IoT MQTT Panel Pro" configuration for my "Next" button, I see a few minor differences, including:

- I'm using an earlier version of picframe, so my topic uses `/switch/` instead of `/button/`
- My button is defined to use QoS 0 instead of QoS 1
- I use a 300 second "keep alive" instead of 60, though I don't remember why (maybe no reason at all).

I might suggest doing a bit more snooping on the MQTT communications with a tool like [MQTT Explorer.](https://mqtt-explorer.com/) If you're unfamiliar, it's free, available for most platforms (win, mac, linux), and super handy for this sort of thing...

And, with any luck someone else will stop by with additional suggestions. :)

robzonbi, 2025-03-05:
Thanks! I'll look into that. 

I've tried all 3 QoS with no change in one from the other. Ive also tried multiple different "keep alive" times, again with no change.

Thanks for atleast confirming there's nothing whacky in my panel pro set up.

robzonbi, 2025-03-05:
@jgodfrey 

Ok! MQTT was exactly what I needed! I discovered that the picframe was cycling between online and offline rapidly. Like online for one second, off for half a second.

Upon searching this symptom, most common cause is multiple devices using the same client ID. I MUST have made a mistep in my MOSQUITTO setup, perhaps naming the broker itself "picframe" ??? Not sure

So anyway I renamed the client ID in the MQTT section of the picture frame config file to "picframe1", made the adjustments in my Panel Pro. And it seems to be seamless now.

I will report back tomorrow after I've had some sleep and more time to review my Mosquitto configuration.

jgodfrey, 2025-03-05:
Nice - glad to hear it! Hopefully, things will continue to work for you.