helgeerbe/picframe #449: If picframe starts when MQTT broker is unavailable, it hangs and does not respond to TERM signal

ylexus, 2025-03-06:
When this happens, the log looks like this:

```
Mar 06 15:34:50 pframe systemctl[847]: Environment variable $XCURSOR_SIZE not set, ignoring.
Mar 06 15:34:50 pframe systemctl[847]: Environment variable $XCURSOR_THEME not set, ignoring.
Mar 06 15:34:51 pframe labwc[851]: INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe']
Mar 06 15:34:51 pframe labwc[851]: INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Mar 06 15:34:52 pframe labwc[899]: The XKEYBOARD keymap compiler (xkbcomp) reports:
Mar 06 15:34:52 pframe labwc[899]: > Warning:          Unsupported maximum keycode 708, clipping.
Mar 06 15:34:52 pframe labwc[899]: >                   X11 cannot support keycodes above 255.
Mar 06 15:34:52 pframe labwc[899]: Errors from xkbcomp are not fatal to the X server
Mar 06 15:34:54 pframe labwc[851]: gl_id  b'GL' b'3.1 Mesa 23.2.1-1~bpo12+rpt3'
Mar 06 15:34:54 pframe labwc[851]: ERROR:interface_mqtt.InterfaceMQTT:MQTT not set up because of: [Errno 101] Network is unreachable
Mar 06 15:34:54 pframe labwc[851]: ERROR:controller.Controller:Can't initialize mqtt. Stopping picframe
```
(I could not find how to set logging to DEBUG to get more info)

The python process does not die though, but instead it stops responding to the graceful `kill` signal (TERM), only to `kill -9`.

My MQTT broker is on a different machine on a local network.

Ideally, picframe should retry connecting to MQTT for a small (configurable) period of time before dying. But at the very least it should not "hang" but actually quit, hopefully then labwc would restart it (or fail and exit, and then systemd would restart it).

My picframe service is defined like this
```
[Unit]
Description=PictureFrame on Pi
After=network-online.target var-storage.mount
Requires=var-storage.mount

[Service]
ExecStart=/usr/bin/labwc
Restart=always

[Install]
WantedBy=default.target
```
With `/home/pi/.config/labwc/` containing the line `/home/pi/start_picframe.sh` which contains:
```
#!/bin/bash
source /home/pi/venv_picframe/bin/activate  # Activate Python virtual environment
picframe &  # Start Picframe in the background
```

helgeerbe, 2025-04-01:
I know, this is not catched. Reconnect works later on, but initial connect will fail.  I don't know how to handle this:

- stop completly to indicate something is wrong on startup or
- start and ignore MQTT until a next restart.

ylexus, 2025-04-01:
My point was that the reconnect does not work. If it fails, the process hangs and nothing happens. No pictures are shown. To be honest, any other reaction would be better than this. A reliable app would just keep connecting forever (with expinential backoff) while still showing pictures. However, even a simple exit on the first failure is acceptable. It's the hanging that's a problem.

helgeerbe, 2025-04-03:
Should be fixed with PR #462 

Tries to connect to broker on start up.  If failed, a reconnect ist tried at every image change.

ylexus, 2025-04-03:
Thanks, I am now on some custom branch that you suggested I switch to when we were debugging the touch issues, when will this fix be publicly available and how do I switch back to the "official" branch?

helgeerbe, 2025-04-04:
Fix is now in dev.

@ylexus  activate your virtual environment like `source venv_picframe/bin/activate`

To install the dev branch: `pip install git+https://github.com/helgeerbe/picframe.git@mqtt-refact --upgrade`

To install the official branch: `pip install picframe --upgrade`

If I don't find any further errors in the MQTT handling I will create a release at the weekend.

helgeerbe, 2025-04-06:
Should be fixed with 2025.04.06

closed 2025-04-06 by helgeerbe