helgeerbe/picframe #450: Unable to get rid of "Image does not exists or is inaccessible" without deleting the database

ylexus, 2025-03-08:
Related to #351. So I deleted a large number of images from the underlying directory, and the log is now full of

```
Mar 08 12:30:26 pframe labwc[27988]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2012/27484FC0-F574-4AB2-83B3-D1F453929E8F-edited.JPG' does not exists or is inaccessible
```

This causes back/forwards not to work sometimes. i.e. often if I execute the "back" action it just refreshes the current image and I can see the WARNING in the log, so I am guessing it tries to go back to the non-existing image, fails and re-displays the next one, which is the current one.

I tried the purge action via the web interface but it does not seem to do anything. In the log I see that the HTTP request has been processed,  

```
Mar 08 12:35:22 pframe labwc[27988]: DEBUG:simple_server.InterfaceHttp:http request from: 192.168.1.10
Mar 08 12:35:22 pframe labwc[27988]: INFO:simple_server.InterfaceHttp:{}
Mar 08 12:35:22 pframe labwc[27988]: DEBUG:simple_server.InterfaceHttp:request finished in:  0.0009715557098388672 seconds
```

and it does not fix the WARNINGs, even after a restart.

I think ideally some smartness could be added here: if a file is missing for, say, a day, and there are other files from the same mount that are accessible, this is a reliable indicator that the file has been deleted by the user, and so it can be purged from the database.

Also, unsure why the PURGE action des not have any effect. #351 mentions a `picframe/purge_files` MQTT topic that does not exist anywhere on my system.

paddywwoof, 2025-03-08:
@ylexus I think this is an area that needs fixing. I can't remember the issue behind the decision to leave records on the db until purged. I will have a search though the original discussions. As you say, it must be possible to do something better here.

jgodfrey, 2025-03-08:
> I can't remember the issue behind the decision to leave records on the db until purged.

IIRC, this was to ensure that temporarily missing images (like, a failed mount for example) didn't cause an unnecessary removal of metadata that would then have to be rebuilt from scratch.

ylexus, 2025-03-08:
@jgodfrey @paddywwoof yes the images should not be deleted from the DB if they are temporarily missing - I use NFS and sometimes it's possible that the there's a temporary problem with the NFS mount. That's why I suggested some smarter logic. For now I suppose the deletion of the database is the only option for me, right?

jgodfrey, 2025-03-08:
Have you restarted the frame since mass deleting the images? It prebuilds the slideshow order when it starts up, and I wouldn't expect it to include images that are marked as missing in the db. If you haven't tried that, definitely give it a shot first.

Really, I wouldn't expect you to have to kill the db as the purge mechanism should* work. In fact, I use it periodically (though, via MQTT). However, killing the db is certainly an option. It should rebuild itself automatically...

ylexus, 2025-03-08:
@jgodfrey yes as I have restarted it multiple times since deleting the images and it's still giving house warnings. If this matters, the number of images is quite large - I don't have the count, but about 500GB out of 1.5TB of images were deleted.

When are the images marked as missing in the DB? During the pre-scan on startup? 

I don't see the purge option in HomeAssistant, so not sure how to invoke the action via MQTT.

jgodfrey, 2025-03-09:
> When are the images marked as missing in the DB? During the pre-scan on startup?

It looks like its done periodically while the frame is running - during otherwise idle time...

Though, it looks like only *folders* are marked as missing (and, thus the images they contained). So, if you removed some images from an otherwise still existing folder, I don't think such images are known to be missing - which might lead to the problem you mention. That's not how I remember this working. Maybe I missed something when poking through the code last night...

jgodfrey, 2025-03-09:
Ah, the images are verified to exist on disk just before being displayed, [here](https://github.com/helgeerbe/picframe/blob/654006514513ddeb19f5ed16ba9736d0fb20f7f3/src/picframe/model.py#L416):

jgodfrey, 2025-03-09:
It looks like:

-  Only folders (not individual files) are marked as missing
- If files are removed from still existing folders, they are not recognized as missing
- Such files will still be added to the "playlist" for the next pass of the slideshow
- If you trigger the purge action, such files should be removed from the db
- Ultimately, missing files should be skipped during the display cycle (in the code linked previously)
- When the slideshow attempts to display an image that's found to be missing from disk, the code skips that file and attempts to find another one. In your case, with thousands of files missing, I guess the code may be caught in a loop that selects the next file, finds it to be missing, selects the next file, finds it to be missing, selects the next file,, ... Maybe that iterative process takes too long and causes some unexpected delays in the slideshow processing (which is quite time dependent as you can image), which maybe leads to what you see.

Assuming the above is correct a "purge" should  straighten it out by removing the missing files from the db. But, even that might take a while when performed on the scale you mentioned - not sure.

It'd be nice to understand why purge isn't working for you. And, as I mentioned before, you could always kill the db, though it may take some time to rebuild.

ylexus, 2025-03-09:
@jgodfrey yes all the folders remain, it's just individual files that are deleted. RE the last item in your list - wouldn't I see a massive amount of warnings in the log when this is happening? I am definitely not seeing that. WARNINGs seem to be only happening when an image is about to be displayed, so not more often that once in a minute (my display period is 60 seconds), however sometimes in small batches of 2-3 images, like this:
```
Mar 08 11:20:19 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2014/IMG_0514.jpg' does not exists or is inaccessible
Mar 08 11:21:39 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2004/DSC01849.JPG' does not exists or is inaccessible
Mar 08 11:21:39 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2016/IMG_3190.JPG' does not exists or is inaccessible
Mar 08 11:21:39 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2006/IMG_2680.JPG' does not exists or is inaccessible
Mar 08 11:23:39 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2003/DSC04794.JPG' does not exists or is inaccessible
Mar 08 11:24:19 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2019/DSC08409.JPG' does not exists or is inaccessible
Mar 08 11:24:59 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2012/DSC_0198.JPG' does not exists or is inaccessible
Mar 08 11:24:59 pframe labwc[1091]: WARNING:image_cache.ImageCache:Image '/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2006/DSC_2254.JPG' does not exists or is inaccessible
```

I suspect once it eventually discovers all missing files, the warnings will disappear. So far not a single images had a duplicate warning:

```
pi@pframe:~ $ journalctl --user -u picframe.service | grep WARNING:image_cache | wc -l
6826
pi@pframe:~ $ journalctl --user -u picframe.service | grep WARNING:image_cache | uniq | wc -l
6826
```

However, this may take a while to discover all of them - there are 72,889 images. While this is happening, the usability issue with the "back" action described in the original report would exist.

As for the purge - it would be nice to have some more logging/stats, as at the moment it's not clear whether the purge code actually got executed when started via web interface.

jgodfrey, 2025-03-09:
The message you posted ("does not exist or is inaccessible") is generated while trying to access the data for the image that's about to be displayed but is found to be missing on disk. So, I'd expect you to see it at roughly the cadence of your set display period. 

The interactions here are somewhat complex, but it's fair to say that some things could / should be improved.

ylexus, 2025-03-16:
This is what happens when a directory is deleted from under a running picframe:
```
Mar 16 10:38:41 pframe labwc[943]: Exception in thread Thread-1 (__loop):
Mar 16 10:38:41 pframe labwc[943]: Traceback (most recent call last):
Mar 16 10:38:41 pframe labwc[943]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Mar 16 10:38:41 pframe labwc[943]:     self.run()
Mar 16 10:38:41 pframe labwc[943]:   File "/usr/lib/python3.11/threading.py", line 975, in run
Mar 16 10:38:41 pframe labwc[943]:     self._target(*self._args, **self._kwargs)
Mar 16 10:38:41 pframe labwc[943]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 55, in __loop
Mar 16 10:38:41 pframe labwc[943]:     self.update_cache()
Mar 16 10:38:41 pframe labwc[943]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 84, in update_cache
Mar 16 10:38:41 pframe labwc[943]:     self.__modified_folders = self.__get_modified_folders()
Mar 16 10:38:41 pframe labwc[943]:                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 16 10:38:41 pframe labwc[943]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 363, in __get_modified_folders
Mar 16 10:38:41 pframe labwc[943]:     mod_tm = int(os.stat(dir).st_mtime)
Mar 16 10:38:41 pframe labwc[943]:                  ^^^^^^^^^^^^
Mar 16 10:38:41 pframe labwc[943]: FileNotFoundError: [Errno 2] No such file or directory: '/var/storage/data/google-photos/Takeout/Google Photos/UK_ 2018_ 2018-02-03-inventory-checkout'
```

I am not a python dev (I am java), but this does not look like a graceful handling of a deleted directory. Is the process still usable after this happens?

paddywwoof, 2025-03-16:
Agree that the area of adding, removing, swapping files and directories needs a bit of fettling. I've often had to delete the db when I've been trying different file type (videos or large images etc) and the obvious course was to create a subdirectory and move files into or out of it.