helgeerbe/picframe #452: MQTT syntax

Duzeper, 2025-03-12:
I am a newcomer to picframe having been a PictureFrame2020 user for several years.
I have it installed and running, and now I would like to configure MQTT control. However, It seems that several MQTT options  from PictureFrame2020 are no longer available in picframe. In particular, I am interested in:-
	quit
	next
	back
	text_on		# toggle file name on and off.
	text_off		# turn all name, date, location off.
	date_on		# toggle date (exif if avail else file) on and off.
	location_on	# toggle location (if enabled) on and off.

I suspect that some of these might be available under the image_attr parameter. I just can't work out the syntax of the MQTT topics/payloads.

I have just read Paddy's 'How to use the DashMQTT Android app to remote control your Raspberry Pi photo frame', and it would seem that these MQTT commands are available. However, Looking at 'Examples of remote commands' in the article, there seems to be some overlap between DashMQTT and Home Assistant in the command syntax.
 
I do not use either of these, but I do use IoT MQTT Panel which is similar to DashMQTT (I just prefer it).

For example; 
	picframe/time_delay, number
uses the dashboard prefix - picframe, topic - time_delay and a numeric value.
I notice that 'time_delay' is a parameter in configuration.yaml  and sending 'picframe/time_delay, number' works fine from my MQTT app.

However, the 'next' command, for example, in Paddy's article is,
	homeassistant/button/picframe_next/set, ON
but there is no 'next' parameter in configuration.yaml 

Sorry if this is a bit rambling, and I suspect that I may be missing the obvious, but, as I say, I am new to picframe.

Any help would be most appreciated.

paddywwoof, 2025-03-14:
Hi, sorry not to reply sooner. I haven't used MQTT for ages but I will try out IoTMQTT next week and update my earlier writeup! Things have probably changed a bit over the years.

Paddy

helgeerbe, 2025-03-16:
Hi @Duzeper,

maybe https://github.com/helgeerbe/picframe/wiki/Control%20Interfaces will help you.

Duzeper, 2025-03-17:
Thanks, however this just seems to be a slightly easier to read version of Paddy's 'Examples of remote commands'.
The first 15 of these all refer to 'homeassistant', which I do not use, and only some of the next 10 'values' refer to configuration.yaml parameters so, although the MQTT options I need are all listed there, what I cannot work out is the syntax so that I can actually use them.

Martin

Duzeper, 2025-03-18:
OK, so I have decided to bite the bullet and install Home Assistant.
Under 'Installation', you say, 'The following assumes that you have completed the PictureFrame basic installation and have Home Assistant running'. Well I have PictureFrame installed and running, but as far as Home Assistant is concerned, I am stuck at the first hurdle. I have followed your link to `https://www.home-assistant.io/installation/raspberrypi` but this is to install a new OS.

How do I install Home Assistant on my RaspberryPi having already installed PictureFrame?

Martin

helgeerbe, 2025-03-21:
You don't have to use homeassistant. Just use your favorite MQTT Explorer and send the message on the specific topic.

You could also use the webinterface and send a rest call to picframe browser. That way, you don't even need the mqtt broker.

But if you like home assistant to manage your smart home. Feel free to use it. To run it on the same raspi like picframe, I would recommend that you use a docker install.

Duzeper, 2025-03-22:
Thanks, I'm pleased  to hear that. Having looked into Home Assistant a little, it seems to be overly complex for my needs. All I need to do is to send a few MQTT messages to PicFrame.
That brings me back to my original post, in that I can't work out the syntax for sending the MQTT messages for those 15 topics which include homeassistant.

I admit to almost total ignorance with regard to the workings of MQTT. However I have successfully configured IoT MQTT Panel on my Android phone to work with PictureFrame2020.
The app uses a Dashboard Prefix Topic (in this case - picframe), a Topic (eg /directory) and a payload (eg Holiday Photos). This would be fine for the 10 values listed in 'MQTT topics and messages' but, as I said, not the first 15.

Unfortunately, I now cannot even test the MQTT messages that I hoped will work. (see new problem below)

I was intending to just send the above response until, after trying a couple of MQTT messages `directory, fade_time, time_delay` which I am not sure actually worked (it was getting late last night), when I rebooted the Rpi, although it booted up, no images were displayed, just a black screen.

I have looked at the issue [picframe losing track of images #126](url), most of which I am afraid I don't understand. Also I do not know how to create or find log files. However, I have tried :

- Stop the picframe service
- Remove the existing cache database file (pictureframe.db3)
- Start the picframe service

but this has had no effect.
Unlike Jeff, I do not see any images at all, just a black screen from the start.

I also note that there could be an issue with `date_from` & `date_to` values, however I have never used those parameters and neither are present in `configuration.yaml`

Martin

Duzeper, 2025-03-28:
OK, so I have pretty much given up with this, having had no response at all for over a week, so I have done a clean install to a new sd card.

Clearly I will still have the same issues with reference to configuring MQTT, and I would still like this to be resolved, but this install has thrown up a completely different problem, in that it appears to totally **ignore** any changes I make to `configuration.yaml`!

Firstly it was showing the 'Have a break' image even though I had changed the main picture directory to `"/home/pi/usbdrive/Pictures"` having mounted a USB drive. At first I thought that this was because I may have made a mistake in mounting the drive, however, I did exactly as I have done in the past on a (fully working) PictureFrame2020, and the RPi does recognize `"/home/pi/usbdrive/Pictures"` and it's subdirectories.

Anyway, just in case, I copied some images to `"/home/pi/Pictures"` and restarted the frame and it displayed the images correctly, even though I had not reset `configuration.yaml` so it seemed to be ignoring the new value of `pic_dir`

Next I set `mat_images` to 'False' (I do not want Smart Matting) and, once again restarted. This setting was also ignored.

I have also reset `time_delay` to 30  (for test purposes), and this is also ignored. The time between slides is still 200 sec (I have timed it).

This install would therefore appear to be independent of `configuration.yaml`. How is this possible?

By the way, when I say 'restarted', I have tried both restarting the picframe service and restarting the RPi (it makes no difference)

Maybe I need to keep trying clean installs until it actually works!

Martin

paddywwoof, 2025-03-28:
Martin, it sounds odd that no changes to `configuration.yaml` do anything. As a kind of 'belt & braces' system we also set default values in `model.py` https://github.com/helgeerbe/picframe/blob/main/src/picframe/model.py You could check if those are the values being used by editing `/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/model.py` maybe change just one parameter. We then need to figure out why the config file isn't being read. You might be able to see error messages if you type `journalctl` the End to the bottom of the listing.

On the MQTT, with the picframe setup trying to cater for home assistant you won't be able to have a prefix topic I don't think. You need to put the whole topic in topic as in the list @helgeerbe pointed you at above. Leave the prefix empty.

Duzeper, 2025-03-29:
Thanks.
I have actually changed 4 parameters in `model.py` (just to save time)
I set:
 `mat_images` to False
`pic_dir` to '~/usbdrive/Pictures' (to confirm I have the drive mounting correct)
`time_delay` to 30.0 (to speed things up)
`fit` to True

and they were all successful.

I appended `journalctl` to the end of `configuration.yaml` and then, when picframe was running I typed `journalctl` in terminal. I'm not sure if that was what you were suggesting, however that did result in a very long list which I clearly don't understand, but I did see some errors toward the end which do relate to picframe:

```
Mar 29 11:33:39 pi labwc[742]: INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe']
Mar 29 11:33:39 pi labwc[742]: INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Mar 29 11:33:39 pi labwc[742]: ERROR:model.Model:Can't parse yaml config file: /home/pi/picframe_data/config/configuration.yaml: while >
Mar 29 11:33:39 pi labwc[742]:   in "/home/pi/picframe_data/config/configuration.yaml", line 56, column 3
Mar 29 11:33:39 pi labwc[742]: expected <block end>, but found '<block mapping start>'
Mar 29 11:33:39 pi labwc[742]:   in "/home/pi/picframe_data/config/configuration.yaml", line 86, column 4
Mar 29 11:33:39 pi labwc[742]: ERROR:model.Model:error trying to set locale to en_US.utf8
```
line 56, column 3 is `pic_dir: "/home/pi/Pictures"`
line 86, column 4 is `geo_key: "picture_frame_Duzeper"`
(I wonder if that is an abberant extra space at the beginning of the line, which I have corrected)

The 'locale' error is another  discrepancy between `model.py` and `configuration.yaml`. In `configuration.yaml` it is en_GB.UTF-8

Martin

Duzeper, 2025-03-29:
Hi, me again.
Just tried to run again, and all I get is a black screen

Running `journctl` I see the following errors:

```
Mar 29 14:22:58 pi labwc[736]: ERROR:interface_mqtt.InterfaceMQTT:MQTT not set up because of: [Errno 2] No such file or directory
Mar 29 14:22:58 pi labwc[736]: ERROR:controller.Controller:Can't initialize mqtt. Stopping picframe
```
I presume  `Stopping picframe` is the resaon for the black screen, but I don't understand. I have, at this stage, not attempted to configure MQTT

Martin

Duzeper, 2025-03-29:
Sorry - `journalctl`

paddywwoof, 2025-03-29:
Martin, Helge made a comment on another thread that's worth following up https://github.com/helgeerbe/picframe/issues/455#issuecomment-2763391373 It looks like, as the error message says, you have a malfformed yaml file. I don't know the details of how yaml works but it seems to use a python method of follow a colon with either a value or an indented block. Probably incorrect indent or tabs rather than spaces or any of the other horrible features of python will cause problems (If json would allow comments it would be 1,000,000 times better than yaml. I prefer it even without comments!)

Sometimes I've had difficulties restarting picframe.service even when it should be good to go. In those cases it's worth just checking if there is a python process still running. I use `pidof python` and if I get a pid 1234 `kill -9 1234`

It sounds like your configuration.yaml is being read now but a) you had a formatting error b) after you fixed that you have not set your mqtt properly
```yaml
mqtt:
  use_mqtt: False                         # default=False. Set True true, to enable mqtt
  server: "your_mqtt_broker"              # No defaults for server
  port: 8883                              # default=8883 for tls, 1883 else (tls must be "" then !!!!!)
  login: "name"                           # your mqtt user
  password: "your_password"               # password for mqtt user
  tls: "/path/to/your/ca.crt"             # filename including path to your ca.crt. If not used, must be set to "" !!!!
  device_id: "picframe"                   # default="picframe" unique id of device. change if there is more than one PictureFrame
  device_url: ""                          # if use_http==True, set url to picframe config page. Must be a valid url, or "" otherwise home assistant runs in an error.
```
Is the default setting and specifies tls. If you don't want to use tls then you need to change your port to 1883, possibly login and password to "" and `tls: ""` do those match up?

Paddy

Duzeper, 2025-03-29:
Thanks Paddy
I have reset the parameters in `model.py`

Set my desired parameters in `configuration.yaml` including the MQTT settings.

Checked validity of `configuration.yaml` in the site that Helge posted - file OK. (I had no idea that yaml was so picky about indentation)

Checked no python processes running.

Still have the same issue on restarting the RPi - no images displayed, just a black screen.

Can I clear `journalctl` so I just see output from last RPi restart (the current output includes stuff from November last year and there are over 38000 lines)?
I did see a couple of  instances of `TypeError`, but although they seemed to refer to `picframe`, they didn't reference `configuration.yaml`, so can I redirect `journalctl` output to a file? It would make searching the output much easier.

Martin

paddywwoof, 2025-03-29:
The errors you're interested in will always be near the end. Just press the `End` key to get there. `journalctl` with no arguments does just show the contents of a log file. It's trimmed automatically but to a larger size than is suitable for the RPi you can `--vacuum-time` or `--vacuum-size` I did `journalct --vacuum-size=50M`

Not sure why your screen is blank. When you say you "appended journalctl to the end of configuration.yam" that sounds like a bad thing to do. Try to undo it whatever you did.

I might try with a "clean" version of configuration.yaml copied from configuration_example.yaml and then **ONLY CHANGE ONE THING AT A TIME** restarting the RPi each time. That way you might be able to see where the problem arises.

Duzeper, 2025-03-29:
I really should have thought of that. I assume that is precisely why configuration_example.yaml is there!
Anyway, I have changed several parameters, one at a time, and, so far it is working fine.
Even though it was being reported as a valid yaml file, clearly there was something in my modified file that picframe didn't like.

I had already deleted `journalctl` from configuration.yaml but I wonder if it was that that had made picframe get it's knickers in a twist!

At the moment, the only error I can see in `journalctl ` is this:

`Mar 29 20:33:53 pi labwc[655]: 00:01:06.154 [ERROR] [types/wlr_linux_dmabuf_v1.c:1240] Failed to get backend DRM FD`

Whatever it means, it does not seem to be affecting picframe.

Thanks for all your help Paddy. Now I can get back to trying  to configure MQTT

I have no doubt I will be back in touch as my severely limited knowledge runs out.
In the meantime, I will be making regular backups of working versions of configuration.yaml.

All the best
Martin

paddywwoof, 2025-03-29:
No prob. If you run into a problem don't hesitate to post it. These issues threads are one of the main destinations for Google searches, so it could well help someone else out in the future.

Paddy

Duzeper, 2025-04-13:
Going back to MQTT syntax, albeit on a slightly different note.
I have successfully configured my desired MQTT topics, but I do have another question.
Is it possible to send more than one topic on the same button/switch?
For example, I have configured a switch to toggle image date - `homeassistant/switch/picframe_date_toggle/set`, and another to toggle image caption - `homeassistant/switch/picframe_caption_toggle/set` (both work OK), and I wonder if it is possible to set a single switch to send both of these topics at the same time.
I should, perhaps, point out that I am not actually using Home Assistant, but an android app (IoTMQTT)
I have tried concatenating the 2 topics separated by  `, ` (comma space) and  ` ,` (space comma), but neither work. In fact, when I try that, the original working topic stops working.

Martin

paddywwoof, 2025-04-15:
Martin, I can't see a simple way to do this with IoTMQTT (I don't think the pro version has this either) From picframe's side it would be possible, but require extra MQTT handling so not likely to happen unless other people spring up needing this functionality.

I'm just putting a little 'how to' for Wolfgang's site about using the M5Stack ASR unit for voice control and I will add a super simple MQTT control example as well as the HTTP control I've tested so far. If I get chance I will include a bit about subscribing and republishing that would suit the kind of thing you're trying to do here.

Paddy

paddywwoof, 2025-04-15:
Martin, I've just added a script to my article showing the MQTT alternative of voice control. I tried your commands in that and it's all pretty straightforward as Helge described.
```python
import time
import paho.mqtt.client as mqtt

def on_connect(client, userdata, flags, rc):
    client.subscribe("do_multiple_things", qos=0)

def on_message(client, userdata, message):
    msg = message.payload.decode("utf-8")
    if message.topic == "do_multiple_things": # NB entered as mqtt string "2016:12:25"
        client.publish('homeassistant/switch/picframe_date_toggle/set', msg)
        time.sleep(0.1)
        client.publish('homeassistant/switch/picframe_caption_toggle/set', msg)

client = mqtt.Client()
client.on_connect = on_connect
client.on_message = on_message
client.connect('localhost', 1883, 60)
client.loop_forever()
```
In IoTMQTT you need a panel with topic `do_multiple_things` (choose something more meaningful, obviously) and  payload `ON` or `OFF` [like this](https://drive.google.com/file/d/1MriSI6anjuSF-SyfODK0uMHSBf2pjHHJ/view?usp=drive_link)

paddywwoof, 2025-04-17:
Martin, you should probably make do_multiple_things.service Exec start `/home/pi/venv_picframe/bin/python` or wherever it's stored I'm out in the countryside so can't check. Alternatively have a bash script file to run `source` then `python`. I will an explicit instruction later tonight once I'm home.

Paddy

PS try it in a terminal manually *before* attempting to automate it with systemd

paddywwoof, 2025-04-17:
Martin, to follow up now I'm home:

1. I saved the python script above as `/home/pi/do_multiple_things.py`
2. I created a file `/home/pi/.config/systemd/user/do_multiple_things.service
```bash
[Unit]
Description=check for mqtt topic to trigger multiple other mqtt jobs

[Service]
ExecStart=/home/pi/venv_picframe/bin/python /home/pi/use_asr_mqtt.py
Restart=always

[Install]
WantedBy=default.target
```
4. enabled using `systemctl --user enable do_multiple_things.service`
5. started with `systemctl --user start do_multiple_things.service` # or will just happen after a reboot
6. In IoTMQTT on my phone I added two more panels like the image below. On has message ON and the other has message OFF. Note that in the python script the function `on_message()` passes the same message on to the two mqtt "publications". You can call this whatever you want **BUT** the lines you asked for clarification have to have the same text in them. i.e. the string in the double quotes must be the same as whatever you decide to use for the topic in your IotMQTT panel. Does that make sense?
7. When I said I created a python script as above, I actually have a python script that does lots of things and this is a stripped down version without the ASR, email attachment checking etc etc. Wolfgang has written up the first draft of my experiments [here](https://www.thedigitalpictureframe.com/how-to-voice-control-your-digital-picture-frame-using-a-cheap-m5stack-asr-voice-recognition-chip/) I don't know if it's officially published but it is available to read.

[image]

Duzeper, 2025-04-17:
Thanks, I'll have a go at that as soon as I can.
However, re. the script/topic names, I am clear that they all need to match. What I am not sure of is, wrt the double quoted strings, are the double quotes syntactically required or are they indicating 'suggested names'?
I am assuming the former, but I just wanted to confirm.

Also, I see that you suggest creating separate 'ON' and 'OFF' panels using 'Buttons'. I wonder if that should be a single 'Switch' panel (which incorporates both 'ON' and 'OFF' payloads), given that the topics I want to use are toggles, which is also indicated by the `client.publish` lines in your script.

Martin

paddywwoof, 2025-04-18:
Martin, the function `subscribe()` is expecting a *string*, which is also the type of variable held in `message.topic`. In python (and most languages) you can create a string by enclosing characters in quotes, single or double (or triple!)

Yes, a switch panel would be better for your application.

Duzeper, 2025-04-19:
Thanks a lot Paddy. That works a treat.
One question though.
I have read Wolfgang's article:
_'The ultimate guide on using systemd to autostart scripts on the Raspberry Pi'_, in which he is clear that service and timer files should be placed in `/etc/systemd/system/`, while you have put the service file in `/home/pi/.config/systemd/user/`. I also note that `picframe.service` is also in this folder.
As I say, it is working fine, so I wonder, if I was to also set a timer to turn picframe on & off, does it actually matter (within limits) where they are placed, as long as the service and related timer files are both in the same location?

Martin

paddywwoof, 2025-04-20:
I think the only difference is that using the `--user` option doesn't require `sudo`. That means there's less possible confusion over modules installed by one user not being available to another (root and pi). With the venv system this problem is probably much improved.

Duzeper, 2025-04-20:
Hi Paddy, thanks for that. Unfortunately, I have had another idea (sorry about that) which I will try to code myself, but I am falling at the first hurdle.
What I want to do is to interrogate image EXIF data for GPS data, along the lines of:
```
if exists 'GPS data' 
   #do this
else
   #do that
```
It's the `if exists 'GPS data'`that I am not sure about. I am not looking to find the actual GPS data, just whether or not it is present.

Also having just examined an image with, nominally, no GPS data, I see that the GPS section of the EXIF data is not actually blank, but contains zero values. I'm not sure if 0° 0' 0.00" would count as 'exists'.

Martin

paddywwoof, 2025-04-21:
Martin, without a bit of coding behind you it might be tricky but I would tap into the functionality in picframe GetImageMeta which will probably be available to python applications running on your Pi. I will show you the kind of thing to do later if I get chance.

Paddy

paddywwoof, 2025-04-21:
I just tried this and it seemed to work OK
```source venv_picframe/bin/activate```
python program with something like
```python
from picframe.get_image_meta import GetImageMeta
...
meta = GetImageMeta('/home/pi/Pictures/asr_serial.jpg')
if meta.has_exif():
    #do this
    print(meta.get_size(), meta.get_location(), meta.get_exif('EXIF ISO'))
else:
    #do that
...
```
You can see the functions available in GetImageMeta here https://github.com/helgeerbe/picframe/blob/main/src/picframe/get_image_meta.py

Duzeper, 2025-04-23:
OK, so that script works for a specific image.
```
from picframe.get_image_meta import GetImageMeta

meta = GetImageMeta('/home/pi/Pictures/test_image.JPG')
#if meta.has_exif():        Not sure if necessary. All my images have EXIF data
gpsdata = meta.get_location()
if "None" in gps_data:           #meta.get_location() returns {"latitude": None, "longitude": None}
    print("No GPS") 
else:
    print("Yes GPS")
```
However, what I wanted to do (I was just trying to help myself somewhat, and here I freely admit that I had rather over estimated my competance) was to further develop the `do_multiple_things` script, which works perfectly, along these lines:

```
import time
import paho.mqtt.client as mqtt

def on_connect(client, userdata, flags, rc):
    client.subscribe("date_caption", qos=0)

def on_message(client, userdata, message):
    msg = message.payload.decode("utf-8")
    if message.topic == "date_caption": 
...
         #if gps_data:
            #client.publish('homeassistant/switch/picframe_location_toggle/set', msg)
         #else:
            client.publish('homeassistant/switch/picframe_date_toggle/set', msg)
            time.sleep(0.1)
            client.publish('homeassistant/switch/picframe_caption_toggle/set', msg)
...

client = mqtt.Client()
client.on_connect = on_connect
client.on_message = on_message
client.connect('localhost', 1883, 60)
client.loop_forever()
```
How do I, or indeed, can I, apply `meta.get_location()` **(if gps_data:)** to the currently displayed image?

Martin

Duzeper, 2025-04-23:
Forgot to mention. Although the 'GetImageMeta' script above works fine, it does throw up a comment:
`xmp loading has failed: /home/pi/Pictures/Loc_Yes.JPG -> string indices must be integers, not 'str'`
This doesn't appear to cause any problems, I just thought I'd mention it.

Martin

paddywwoof, 2025-04-23:
Hi, To do what I think you are trying you would need to get hold of the image path for the image that's being shown currently. You can do that by subscribing to 'homeassistant/sensor/picframe_image/state' which gives you the file name. However you would need to add the directory then do all your GetImageMeta stuff. What you can do instead is get the image attributes directly. Because you want to 'hold onto' the messages and act accordingly in whatever order they arrive, you should probably put all your functions inside a class and set properties. Something like...
```python
import time
import json
import paho.mqtt.client as mqtt

class DateCaption:
    def __init__(self):
        self.date_caption = 'OFF'
        self.has_gps = False
        self.client = mqtt.Client()
        self.client.on_connect = self.on_connect
        self.client.on_message = self.on_message
        self.client.connect('localhost', 1883, 60)
        self.client.loop_start()
        
    def on_connect(self, client, userdata, flags, rc):
        client.subscribe("homeassistant/sensor/picframe_image/attributes", qos=0)
        client.subscribe("date_caption", qos=0)
    
    def on_message(self, client, userdata, message):
        msg = message.payload.decode("utf-8")
        if message.topic == "date_caption":
            self.date_caption = msg
        attr = json.loads(message.payload)
        if attr["latitude"] is None or attr["longitude"] is None:
            self.has_gps = False
        else:
            self.has_gps = True
        self.publish()

    def publish(self):
        if self.has_gps:
            self.client.publish('homeassistant/switch/picframe_location_toggle/set', self.date_caption)
        else:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.date_caption)
            time.sleep(0.1)
            self.client.publish('homeassistant/switch/picframe_caption_toggle/set', self.date_caption)

date_caption = DateCaption()

while True:
    time.sleep(5.0)
```
I've not tested this in one chunk but you should be able to figure it out hopefully!

In you previous code snippet the problem was possibly that using the `if x in y` syntax in python will check through the keys of y to see if any of them match x. In your case the two keys were 'latitude' and 'longitude' so None would never match. And "None" is a string whereas None is interpreted specially in python. To do what you were intending you would need to do `if None in gpsdata.values()`. Note you have changed the name of the variable from one line to the next. However your error message sounds like something different.

Paddy

Duzeper, 2025-04-24:
Firstly, I see the change of variable name. This is just a typo in this post while I was making test changes to the script syntax. I'm afraid I need to test pretty much any change I make every time. For example, I have just tested your suggestion above re. `if None in gpsdata.values()` and, firstly I didn't notice the parentheses after 'values', then I left in the double quotes around 'None'.
I'm pretty sure that I had originally made the assumption that the return of`{'latitude': None, 'longitude': None}` was a string.

My problem is, I think, that I don't 'think in code'. I did write a fairly lengthy database program in VBA about 15 years ago before I retired, but if I looked at it now, I doubt I could even follow my own code.

Also, even when I have the script written and working correctly, it still throws up the `xmp loading has failed:` message before the`print` line. It isn't described as an 'error' and, as I said, it doesn't appear to cause any problems, and, as far as I know, I have no need to access XMP data anyway.
Many thanks for your suggestion above. I will firstly give it a try and then try to understand it!

Martin

paddywwoof, 2025-04-24:
Martin, I'm just pleased that you're having a go. I tend to scribble stuff down from a general memory of how I think it should work, then go through tfixing he error messages. There's really no way to know if `values` is a function or a property or a function "@decorated" to look like a property!

I will look at the xmp loading message. It's probably coming from a try catch block in get_image_meta.py I noticed an extra bit of processing that could be tied up at the same time.

Paddy

Duzeper, 2025-04-26:
Hi Paddy. Sorry, only just had time to try that script and, unfortunately, it doesn't work - nothing happens when I press the button.
Where you have used `date_caption` I have changed that to `dat_cap_loc` in order to prevent conflict with the original script/MQTT panel, and I hope I haven't introduced any errors there. I have also, of course, changed the MQTT topic to match.

```
import time
import json
import paho.mqtt.client as mqtt

class DatCapLoc:
    def __init__(self):
        self.dat_cap_loc = 'OFF'
        self.has_gps = False
        self.client = mqtt.Client()
        self.client.on_connect = self.on_connect
        self.client.on_message = self.on_message
        self.client.connect('localhost', 1883, 60)
        self.client.loop_start()
        
    def on_connect(self, client, userdata, flags, rc):
        client.subscribe("homeassistant/sensor/picframe_image/attributes", qos=0)
        client.subscribe("dat_cap_loc", qos=0)
    
    def on_message(self, client, userdata, message):
        msg = message.payload.decode("utf-8")
        if message.topic == "dat_cap_loc":
            self.dat_cap_loc = msg
        elif message.topic == "homeassistant/sensor/picframe_image/attributes":
            attr = json.loads(message.payload)
            if attr["latitude"] is None or attr["longitude"] is None:
                self.has_gps = False
            else:
                self.has_gps = True
            self.publish()

    def publish(self):
        if self.has_gps:
            self.client.publish('homeassistant/switch/picframe_location_toggle/set', self.dat_cap_loc)
        else:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.dat_cap_loc)
            time.sleep(0.1)
            self.client.publish('homeassistant/switch/picframe_caption_toggle/set', self.dat_cap_loc)

dat_cap_loc = DatCapLoc()

while True:
    time.sleep(5.0)
```
Probably not relevant, but when I look at MQTT Explorer, I noticee that the image attributes for an image with no gps data include 
'"latitude": null & "longitude": null', not 'None'. I presume these are exactly equivalent, but I thought I'd mention it.

Incidentally, I notice when I run the script (also with the original date_caption script), there is a message in terminal:
```
pi@pi:~ $ /home/pi/venv_picframe/bin/python /home/pi/dat_cap_loc.py
/home/pi/dat_cap_loc.py:9: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  self.client = mqtt.Client()
```
As I said, this happens also with the working script, so does not appear to be an issue, but I wonder what it is that I need to update.

paddywwoof, 2025-04-27:
Martin, I get that warning too, not something about passing some argument to the constructor, but seems to work despite. Hard to know what isn't working but I would stick some print() lines in the `on_message` to see if the broker is responding and see what is in message.payload and message.topic

Duzeper, 2025-04-27:
I had thought along those lines last night just as I hit 'shutdown'
Anyway, I have inserted 
```
        print("Payload", message.payload)
        print("Topic", message.topic)
```
immediately after line 20 `msg = message.payload.decode("utf-8")`
Hitting the switch (ON) firstly returns this:
```
pi@pi:~ $ /home/pi/venv_picframe/bin/python /home/pi/dat_cap_loc.py
/home/pi/dat_cap_loc.py:9: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  self.client = mqtt.Client()
Payload b'ON'
Topic dat_cap_loc
```
followed, after a few seconds, by:
```
Payload b'{"latitude": null, "longitude": null, "location": null, "EXIF FNumber": 4.0, "EXIF ExposureTime": "1/500", "EXIF ISOSpeedRatings": 100.0, "EXIF FocalLength": "38.0", "EXIF DateTimeOriginal": 1236601979.0, "Image Model": "Canon EOS 350D DIGITAL", "Image Make": "Canon", "IPTC Caption/Abstract": "Lanzarote March 2009", "IPTC Object Name": null, "IPTC Keywords": "Gables Windows doors etc,Hotel,Landscape,Lanzarote,Rocks,Volcanic,"}'
Topic homeassistant/sensor/picframe_image/attributes
```
Then hitting (OFF):
```
Payload b'OFF'
Topic dat_cap_loc
```
I have now spotted, however, that the script is actually working, only 1 image late.

When I hit 'ON', the information (either 'location' or 'date/caption', they both work) doesn't appear until the next image displays.
If I leave the feature ON, it does follow though to subsequent images, but then, 'OFF' also waits for the next image to appear before switching off.

Martin

paddywwoof, 2025-04-27:
The `picframe_image/attributes` is only sent when the new image is chosen in controller so maybe the mqtt instruction is sent too soon to be picked up by the current image. Though that doesn't make sense when sending the `dat_cap_loc` topic message.

... Ah I've spotted it: I accidentally indented the last line of the on_message method. s`elf.publish()` should always run for either topic

Duzeper, 2025-04-27:
Brilliant. That's got it.
That's the sort of coding error that I would make. The difference is that I would probably never have spotted it!

Many thanks

Martin

Duzeper, 2025-04-27:
Just one more query.
Are there any other location settings?
Although I have `load_geoloc: True`, most of my locations are really very vague, eg 'Las Palmas, Spain' for somewhere in the Canaries. Some do go down to 'Town' level, but there is certainly nothing like the detail shown in the example in Wolfgangs article `How to show the place where your photo was taken with the Raspberry Pi digital picture frame geolocation feature`

Martin

Duzeper, 2025-04-27:
Having tried some images that I do get detailed locations in PictureFrame2020, I am seeing the same, more detailed, location info, so, I presume it is about map data of the specific location.
I am also pleased to see that the Icelandic alphabet is catered for.