helgeerbe/picframe #453: installing picframe in my own virtual environment

jnsilvius, 2025-03-21:
Hi, 
I have a number of virtual environments running on my raspberry pi and added picframe to one of them. So the location where i'm installing picframe is different from /home/pi/picframe.
I managed to get it installed in my own environment, but there's one modification i need to make in the installed picframe library:
in model.py you have this definition: 
DEFAULT_CONFIGFILE = "~/picframe_data/config/configuration.yaml"

in my case "~" points to "/home/pi", 
My workaround is that i change this line with the full pathname and after that everything works fine.

i'm not too experienced in python, but here's what i tried as well:

get the project folder using os:
import os
project_folder = os.path.dirname(os.path.dirname(os.path.dirname(os.path.dirname(script_path))))
its ugly, but it works

perhaps storing the project folder in an environment variable would be a cleaner approach

helgeerbe, 2025-03-21:
Hi @jnsilvius,

I'm not sure, if I understand correctly what you are trying. But you can easily provide an alternative path and filename for the configuration file. 

`picframe -h` is your friend:

````sh
UserWarning: Using SDL2 binaries from pysdl2-dll 2.30.10
INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe', '-h']
usage: picframe [-h] [-i DESTINATION_DIRECTORY | -v | configfile]

positional arguments:
  configfile            /path/to/configuration.yaml

options:
  -h, --help            show this help message and exit
  -i DESTINATION_DIRECTORY, --initialize DESTINATION_DIRECTORY
                        creates standard file structure for picframe in
                        destination directory
  -v, --version         print version information

````

jnsilvius, 2025-03-21:
Hi @helgeerbe ,

Thanks for that. I have used the option -i and that indeed creates a standard file structure in the target directory, but my issue is with one of the files in python library:

when running run_start.py i get the following warning:
```
INFO:start.py:starting ['run_start.py']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/scripts/picframe/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/home/pi/scripts/picframe/.venv/lib/python3.12/site-packages/picframe/start.py", line 131, in main
    m = model.Model()
        ^^^^^^^^^^^^^
  File "/home/pi/scripts/picframe/.venv/lib/python3.12/site-packages/picframe/model.py", line 159, in __init__
    with open(configfile, 'r') as stream:
         ^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/picframe_data/config/configuration.yaml'
```

It doesnt point to the destination directory that was created.
i can fix this by changing **model.py** in the installed library by changing below statement to the full path where picframe is installed
DEFAULT_CONFIGFILE = "~/picframe_data/config/configuration.yaml"
perhaps that clarifies?

running picframe -v gives these version:
```
picframe version:  2024.11.01

Checking required packages......
PIL :  11.1.0
pi3d :  2.54
yaml :  6.0.2
paho.mqtt :  2.1.0
iptcinfo3 :  2.1.4
numpy :  2.2.4
ninepatch : installed, but no version info
pi_heif :  0.22.0
defusedxml :  0.7.1
```

paddywwoof, 2025-03-22:
Hi @jnsilvius I think the point @helgeerbe is making is that if you start the app with no command line arguments, which is the vanilla method, it will default to the hard coded default you found in model.py.

However if you want to have a different location for the configuration.yaml file you can simply add that to the startup command
```bash
picframe /home/pi/alternative/location1/picframe_data/config/configuration.yaml
```
You can see me use that to run two versions of picframe at the same time with different settings https://github.com/helgeerbe/picframe/issues/438#issuecomment-2634208619 answering a question about multiple screens.

jnsilvius, 2025-03-25:
Hi @paddywwoof that solved my issue, many thanks to both!

closed 2025-03-25 by jnsilvius