helgeerbe/picframe #455: Lose the initial set-up image after reboot

demonboy, 2025-03-25:
Hi, I've just installed this on a Zero 2 W and have run into the same issue twice, having reflashed my card twice. I've been following the [one-click guide]([url](https://www.thedigitalpictureframe.com/install-the-pi3d-pictureframe-software-with-one-click-2025-edition-raspberry-pi-2-3-4-5/)) and have had no problems installing. My problems start after that. If I do a reboot I no longer see the Have A break image (it's still there in the correct directory). I've tried adding some images to the Pictures folder, rebooting, restarting the service, but nothing displays. The images I'm testing with are small files (25kb). I've tried changing the image location too, to no avail.

Unfortunately, I don't know how to troubleshoot or where to begin troubleshooting. I'd be grateful if someone could give me some pointers, thanks.

paddywwoof, 2025-03-26:
Quite hard to suggest things without more idea about what's happening. What's on the screen? Normal command line or blank screen with mouse pointer? I would try to see what processes are running using top. Does anything happen when you stop picframe.service then restart it? I would run journalctl and go to the End of the listing to see if there were any errors.

demonboy, 2025-03-26:
Hi Paddy, yeah, I'm sorry, I haven't used a Pi or Linux in a while so I'm rusty. All I see is a black screen with the mouse pointer. Should I always see the initial image until I post my own, or does this only show on first install? Nothing happens when I stop and then restart picframe service, still the black screen with mouse pointer. Power supply is the official Pi adapter. In both installs that initial picture disappeared after instigating wsdd and smb and then rebooting. I'll try journalctl next.

paddywwoof, 2025-03-26:
The coffee cup picture shows whenever the available pictures are none, either before any are there or if you set a filter that excluded everything. Wolfgang's one click script installs smb (user pi, password pi by default), are you installing manually as well? Journalctl is quite long-winded but might give clues.

demonboy, 2025-03-26:
No, you're right, I mislead you. Just wsdd after the one-click install. I might try a clean install and then record every step after that. I don't know journalctl, I'm sure I can look it up, but anything in particular I should look out for?

paddywwoof, 2025-03-26:
journalctl records everything systemd thinks interesting, so some is very obscure. See if there any python type errors, most are caught in try/except blocks so might describe something different from the actual problem, but good to show where to look. If you do reinstall, try running without wsdd to see if that is a problem. You should be able to copy pictures to the pi over smb or scp (I think it's in windows command line, certainly on Linux and Mac) or copy to the SD card when plugged into PC

helgeerbe, 2025-03-26:
Hi @demonboy, 

If you log in over ssh to the console you can check with `ps axu | grep picframe` if there is a picframe process running. If so, you could also add a logfile path in `configuration.yam` and check if picframe has a problem. But you will have only a logfile, if picframe is going to start.

demonboy, 2025-03-26:
Hi @helgeerbe 

I've just booted the machine up Your first suggestion returns:

`pi           688  0.0  0.4   6088  1920 pts/0    S+   10:03   0:00 grep --color=auto picframe`

Does this mean it is running?

demonboy, 2025-03-26:
Here is a sample from my journal covering a boot this morning and the last thing I did last night:

`Mar 25 22:18:58 piphoto systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Mar 25 22:19:02 piphoto labwc[578]: INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe']
Mar 25 22:19:02 piphoto labwc[578]: INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Mar 25 22:19:02 piphoto labwc[578]: Traceback (most recent call last):
Mar 25 22:19:02 piphoto labwc[578]:   File "/home/pi/venv_picframe/bin/picframe", line 8, in <module>
Mar 25 22:19:02 piphoto labwc[578]:     sys.exit(main())
Mar 25 22:19:02 piphoto labwc[578]:              ^^^^^^
Mar 25 22:19:02 piphoto labwc[578]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/start.py", line 133, in main
Mar 25 22:19:02 piphoto labwc[578]:     v = viewer_display.ViewerDisplay(m.get_viewer_config())
Mar 25 22:19:02 piphoto labwc[578]:         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 25 22:19:02 piphoto labwc[578]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py", line 61, in __init__
Mar 25 22:19:02 piphoto labwc[578]:     self.__text_bkg_hgt = config['text_bkg_hgt'] if 0 <= config['text_bkg_hgt'] <= 1 else 0.25
Mar 25 22:19:02 piphoto labwc[578]:                                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 25 22:19:02 piphoto labwc[578]: TypeError: '<=' not supported between instances of 'int' and 'str'
Mar 25 22:19:02 piphoto labwc[578]: Exception in thread Thread-1 (__loop):
Mar 25 22:19:02 piphoto labwc[578]: Traceback (most recent call last):
Mar 25 22:19:02 piphoto labwc[578]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Mar 25 22:19:02 piphoto labwc[578]:     self.run()
Mar 25 22:19:02 piphoto labwc[578]:   File "/usr/lib/python3.11/threading.py", line 975, in run
Mar 25 22:19:02 piphoto labwc[578]:     self._target(*self._args, **self._kwargs)
Mar 25 22:19:02 piphoto labwc[578]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 56, in __loop
Mar 25 22:19:02 piphoto labwc[578]:     time.sleep(self.__update_interval)
Mar 25 22:19:02 piphoto labwc[578]: TypeError: 'str' object cannot be interpreted as an integer
Mar 25 22:19:13 piphoto nmbd[520]: [2025/03/25 22:19:13.586493,  0] ../../source3/nmbd/nmbd_become_lmb.c:398(become_local_master_stage2)
Mar 25 22:19:13 piphoto nmbd[520]:   *****
Mar 25 22:19:13 piphoto nmbd[520]: 
Mar 25 22:19:13 piphoto nmbd[520]:   Samba name server PIPHOTO is now a local master browser for workgroup WORKGROUP on subnet 192.168.254.164
Mar 25 22:19:13 piphoto nmbd[520]: 
Mar 25 22:19:13 piphoto nmbd[520]:   *****
Mar 25 22:19:13 piphoto systemd[1]: systemd-fsckd.service: Deactivated successfully.
Mar 26 10:02:15 piphoto systemd-timesyncd[322]: Contacted time server 81.179.123.60:123 (2.debian.pool.ntp.org).
Mar 26 10:02:15 piphoto systemd-timesyncd[322]: Initial clock synchronization to Wed 2025-03-26 10:02:15.685883 GMT.
Mar 26 10:02:15 piphoto systemd[1]: Starting dpkg-db-backup.service - Daily dpkg database backup service...
Mar 26 10:02:15 piphoto systemd[1]: Starting logrotate.service - Rotate log files...
Mar 26 10:02:16 piphoto systemd[1]: logrotate.service: Deactivated successfully.
Mar 26 10:02:16 piphoto systemd[1]: Finished logrotate.service - Rotate log files.
Mar 26 10:02:16 piphoto systemd[1]: dpkg-db-backup.service: Deactivated successfully.
Mar 26 10:02:16 piphoto systemd[1]: Finished dpkg-db-backup.service - Daily dpkg database backup service.
Mar 26 10:02:17 piphoto systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Mar 26 10:03:00 piphoto sshd[669]: Accepted password for pi from 192.168.254.250 port 53540 ssh2
Mar 26 10:03:00 piphoto sshd[669]: pam_unix(sshd:session): session opened for user pi(uid=1000) by (uid=0)
Mar 26 10:03:00 piphoto systemd-logind[356]: New session 3 of user pi.
Mar 26 10:03:00 piphoto systemd[1]: Started session-3.scope - Session 3 of User pi.
Mar 26 10:03:00 piphoto sshd[669]: pam_env(sshd:session): deprecated reading of user environment enabled
Mar 26 10:17:01 piphoto CRON[705]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Mar 26 10:17:01 piphoto CRON[706]: (root) CMD (cd / && run-parts --report /etc/cron.hourly)
Mar 26 10:17:01 piphoto systemd[1]: Starting systemd-tmpfiles-clean.service - Cleanup of Temporary Directories...
Mar 26 10:17:01 piphoto CRON[705]: pam_unix(cron:session): session closed for user root
Mar 26 10:17:01 piphoto systemd[1]: systemd-tmpfiles-clean.service: Deactivated successfully.
Mar 26 10:17:01 piphoto systemd[1]: Finished systemd-tmpfiles-clean.service - Cleanup of Temporary Directories.
Mar 26 10:17:01 piphoto systemd[1]: run-credentials-systemd\x2dtmpfiles\x2dclean.service.mount: Deactivated successfully.`

helgeerbe, 2025-03-26:
Thanks, this looks like picframe crashes on startup. 

I see that it tries to read the configuration.yaml entry `text_bkg_hgt`. This value should be a number. Like `text_bkg_hgt: 0.25`       `

Can you post your configuration.yam file please.

helgeerbe, 2025-03-26:
This one: `/home/pi/picframe_data/config/configuration.yaml:`

paddywwoof, 2025-03-26:
Bit odd that it showed the missing pics image OK. I didn't think that altered behaviour in viewer_display, maybe it does.

demonboy, 2025-03-26:
Config file below. BTW, I did a clean install and this time I followed the long-winded set-up as opposed to the one-button version. After starting the service and rebooting, my screen now just shows a prompt, not the black screen with cursor. I am doing all of this via Putty, no keyboard input on the screen.

viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 20.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: "title caption name date folder location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  display_power: 2                        # default=0. choices={0, 1, 2}, 0 will use legacy `vcgencmd`, 1 will use `xset`, 2 will use 'wlr-randr' to blank the display
  use_glx: False                          # default=False. Set to True on linux with xserver running. NB use_sdl2 might need to be False for this to work.
  use_sdl2: True                          # default=True. pysdl2 can use display without xserver, it should be installed as a dependency of pi3d
                                          # but might need `sudo apt install libsdl2-dev` if picframe gives errors about missing sdl2

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%-I:%M"                  # default="%-I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay
  clock_top_bottom: "T"                   # default="T" ("T", "B"), whether to display clock at top or bottom of screen
  clock_wdt_offset_pct: 3.0               # default=3.0 (1.0-10.0), used to calclate pixels between clock text and side of screen
[... cut ...]
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"

demonboy, 2025-03-26:
> Can you check if the systemd picframe.service has started (pidof python, or ps axu as @helgeerbe suggests)

That's what I posted previously:

ps axu | grep picframe = 

`pi           587  0.0  0.4   6088  1920 pts/0    S+   14:26   0:00 grep --color=auto picframe
`
I do not know whether this tells me it is running or not.

I'm having problems accessing the config file. I set my install up to provide a web interface, which I did successfully first time around. After a reboot, however, when I type 'python config.py', it returns:

`Traceback (most recent call last):
  File "/home/pi/config.py", line 1, in <module>
    from flask import Flask, render_template, request, redirect, url_for
ModuleNotFoundError: No module named 'flask'
`

I've then tried accessing my config file via the prompt by typing:

`picframe_data/config/configuration.yaml`

And it replies:

`
-bash: picframe_data/config/configuration.yaml: Permission denied
`

paddywwoof, 2025-03-26:
I think it mens it is running but I don't really know what ps axu does differently from top say. If you use the systemctl stop and restart what do you get written to journalctl? You probably want to use nano editor
```bash
nano /home/pi/picframe_data/config/configuration.yaml
Then after editing use CtrlS to save and CtrlX to exit

demonboy, 2025-03-26:
If I type 'systemctl --user enable picframe.service', it returns 'inactive'.
I then enter 'systemctl --user enable picframe.service' and then check running services and it still returns 'inactive'.

Journal shows:

Mar 26 15:18:18 piphoto labwc[1315]: Some friendly trouble-shooting help
Mar 26 15:18:18 piphoto labwc[1315]: ===================================
Mar 26 15:18:18 piphoto labwc[1315]: If a seat could not be created, this may be caused by lack of permission to the
Mar 26 15:18:18 piphoto labwc[1315]: seat, input and video groups. If you are using a systemd setup, try installing
Mar 26 15:18:18 piphoto labwc[1315]: polkit (sometimes called policykit-1). For other setups, search your OS/Distro's
Mar 26 15:18:18 piphoto labwc[1315]: documentation on how to use seatd, elogind or similar. This is likely to involve
Mar 26 15:18:18 piphoto labwc[1315]: manually adding users to groups.
Mar 26 15:18:18 piphoto labwc[1315]: If the above does not work, try running with `WLR_RENDERER=pixman labwc` in
Mar 26 15:18:18 piphoto labwc[1315]: order to use the software rendering fallback
Mar 26 15:18:18 piphoto labwc[1316]: 00:00:10.050 [ERROR] [libseat] [common/terminal.c:162] Could not open target tty: Permission de>
Mar 26 15:18:18 piphoto labwc[1316]: 00:00:10.051 [ERROR] [libseat] [seatd/seat.c:86] Could not open terminal to clean up VT 0: Perm>
Mar 26 15:18:18 piphoto systemd[541]: picframe.service: Main process exited, code=exited, status=1/FAILURE
Mar 26 15:18:18 piphoto systemd[541]: picframe.service: Failed with result 'exit-code'.
Mar 26 15:18:19 piphoto systemd[541]: picframe.service: Scheduled restart job, restart counter is at 298.
Mar 26 15:18:19 piphoto systemd[541]: Stopped picframe.service - PictureFrame on Pi.
Mar 26 15:18:19 piphoto systemd[541]: Started picframe.service - PictureFrame on Pi.
lines 28789-28811/28811 (END)
Mar 26 15:18:08 piphoto labwc[1316]: 00:00:00.008 [ERROR] [libseat] [seatd/seat.c:72] Could not open terminal for VT 0: Permission d>
Mar 26 15:18:08 piphoto labwc[1316]: 00:00:00.008 [ERROR] [libseat] [seatd/seat.c:461] Could not open VT for client
Mar 26 15:18:08 piphoto labwc[1316]: 00:00:00.008 [ERROR] [libseat] [common/terminal.c:162] Could not open target tty: Permission de>
Mar 26 15:18:08 piphoto labwc[1316]: 00:00:00.008 [ERROR] [libseat] [seatd/seat.c:86] Could not open terminal to clean up VT 0: Perm>
Mar 26 15:18:18 piphoto labwc[1315]: 00:00:10.019 [ERROR] [backend/backend.c:105] Timeout waiting session to become active
Mar 26 15:18:18 piphoto labwc[1315]: 00:00:10.019 [ERROR] [backend/backend.c:407] Failed to start a DRM session
Mar 26 15:18:18 piphoto labwc[1315]: 00:00:10.019 [ERROR] [../src/server.c:354] unable to create backend
Mar 26 15:18:18 piphoto labwc[1315]: Some friendly trouble-shooting help
Mar 26 15:18:18 piphoto labwc[1315]: ===================================
Mar 26 15:18:18 piphoto labwc[1315]: If a seat could not be created, this may be caused by lack of permission to the
Mar 26 15:18:18 piphoto labwc[1315]: seat, input and video groups. If you are using a systemd setup, try installing
Mar 26 15:18:18 piphoto labwc[1315]: polkit (sometimes called policykit-1). For other setups, search your OS/Distro's
Mar 26 15:18:18 piphoto labwc[1315]: documentation on how to use seatd, elogind or similar. This is likely to involve
Mar 26 15:18:18 piphoto labwc[1315]: manually adding users to groups.
Mar 26 15:18:18 piphoto labwc[1315]: If the above does not work, try running with `WLR_RENDERER=pixman labwc` in
Mar 26 15:18:18 piphoto labwc[1315]: order to use the software rendering fallback
Mar 26 15:18:18 piphoto labwc[1316]: 00:00:10.050 [ERROR] [libseat] [common/terminal.c:162] Could not open target tty: Permission de>
Mar 26 15:18:18 piphoto labwc[1316]: 00:00:10.051 [ERROR] [libseat] [seatd/seat.c:86] Could not open terminal to clean up VT 0: Perm>
Mar 26 15:18:18 piphoto systemd[541]: picframe.service: Main process exited, code=exited, status=1/FAILURE
Mar 26 15:18:18 piphoto systemd[541]: picframe.service: Failed with result 'exit-code'.
Mar 26 15:18:19 piphoto systemd[541]: picframe.service: Scheduled restart job, restart counter is at 298.
Mar 26 15:18:19 piphoto systemd[541]: Stopped picframe.service - PictureFrame on Pi.
Mar 26 15:18:19 piphoto systemd[541]: Started picframe.service - PictureFrame on Pi.

helgeerbe, 2025-03-29:
Martin, it seems your configuation.yaml is not correctly formatted. You can test it with https://www.yamllint.com 

Otherwise can you upload your config file?

paddywwoof, 2025-03-29:
@demonboy that journalctl looks to indicate that picframe.service isn't managing to start. I don't know what the errors to do with libseat indicate but I would suspect that one of the lower level required components didn't install correctly. You could try `sudo apt update` and `sudo apt upgrade` then go through the dependencies and try reinstalling. I think you need:
```sudo apt-get install libsdl2-dev xwayland labwc wlr-randr``` you could see if any of those give an unexpected error. And obviously reboot. I've used Wolfgang's one click script quite a few times now and it has always worked (at least until I started tweaking things during development and scrambled the setup!) so I might be tempted to start from that, not enable mqtt or http, not install any other applications then smb, scp or copy to SD:/root/home/pi/Pictures one or two images and see how it went.