helgeerbe/picframe #458: sensor.picframe_image_date missing

SebJas, 2025-04-02:
Hi,

I looked at the description https://www.thedigitalpictureframe.com/how-to-show-your-current-picture-frame-photo-in-home-assistant-2024-edition/ and in the code example there is a sensor sensor.picframe_image_date. But I don't see this sensor when looking at the MQTT entities.

Hope you can help me here

paddywwoof, 2025-04-02:
I can't see that either. I think it would need a functions adding to `controller.py` something like
```python
    @property
    def image_date(self):
        pic = self.__model.get_current_pics()[0]
        if pic.exif_datetime > 0:
            fdt = time.strftime(self.__model.__show_text_fm, time.localtime(pic.exif_datetime))
        else:
            fdt = None
        return fdt
```
with an appropriate line adding in `interface_mqtt.py` function `publish_state()`

helgeerbe, 2025-04-02:
No, that isn't a sensor from picframe. It's a templete sensor in home assistant:

````yaml
template:
  - sensor:
      - name: picframe_image_date
        unique_id: picframe_image_date
        state: "{{ state_attr('sensor.picframe_image', 'EXIF DateTimeOriginal') | int(default=0) | timestamp_custom('%d. %b. %Y %H:%M') }}"
````

So its extracts from sensor.picframe_image the `EXIF DateTimeOriginal` (unix timestamp) and converts it in your human readable time string.

helgeerbe, 2025-04-02:
Topic of the image attributes is `homeassistant/sensor/picframe_image/attributes`.  The message is a json string. Where `picframe`is the picframe id from the configuration file.

SebJas, 2025-04-02:
@helgeerbe Thanks you for the quick response. I got it working now.

closed 2025-04-02 by SebJas