helgeerbe/picframe #460: Metadata info used for "title" displayed on screen

ddavistx82, 2025-04-02:
Do you have list of the sequence for metadata fields used to find the "title" used if displayed on-screen?

In the 2020 version, it seemed to default to the xmp:dc_Title field.  Only info found there would be displayed on-screen. 
I just installed the new version for Bookworm.  This version seems to use the iptc:headline field.  However, if that field is blank it seems to use the exif:ImageDescription or the idfo:Title or the xmp:dc_Title field.  I have not been able to confirm the preferred check order for the title and don't know where to put the desired text to be displayed.

Thanks

helgeerbe, 2025-04-04:
I had a short glimpse at the code for title:
First `xmp Headline` then `IPTC Object name` else `None`

On Display there is no placeholder for an empty title. So if there is no title but a description. The description is shown at titles position.

ddavistx82, 2025-04-04:
I have done a bit of testing and those 2 metadata fields do seem to be used for "title" in that order.  Unfortunately, those 2 fields are not accessible with a simple Windows app like File Explorer (properties), Photo Viewer or Photo Gallery.  They can only be edited with a more sophisticated tool like LightRoom or ExiftoolGUI.  Since I'm trying to make using picframe as simple as possible for some relatives, I will have to switch to "caption" vs "title".  From what I can tell, the "caption" option in configuration.yaml uses the xmp Description field as it's source.  Luckily, if you edit the Title field in File Explorer Properties, any change is written to the exif Image Description, xmp-dc Title, xmp Description, and the ifd0 XPTitle.  Since this includes the xmp Description field, a simple edit in File Explorer Properties will appear on the display.  Do you know if the code will check other fields if the xmp Description field is empty?  Thanks