helgeerbe/picframe #461: How to mount a USB drive that picframe can see

ddavistx82, 2025-04-02:
How do you mount an external USB on an RPi4B that picframe can see? 

I used Wolfgang's single script install.  I ran the initial OS install for an RPi4B which worked successfully.  I then ran the software update/upgrade process manually since I had issues with it running inside the script (probably network errors).  This completed successfully.  I could see and access a USD drive plugged-in to RPi.  I then ran the single script install which complete successfully.  I add some photos to the Pictures directory and made some minor updates to the configuration.yaml file.  After a reboot the picframe seems to be working correctly.  However, I can not see the USB drive.  I get the same results if I run the install script on a RPi3B+ card.  I have tried to follow some Raspberry Pi forum posts on how to mount a USB drive but have been unsuccessful in getting a process that has it mount in the /media/pi/drivename so that I can access it in the configuration.yaml file.  Any suggestions?

helgeerbe, 2025-04-04:
Hi @ddavistx82  this is quite old, but may still work: https://github.com/helgeerbe/picframe/issues/76#issuecomment-808919644

ddavistx82, 2025-04-04:
Wolfgang had another process to automatically and dismount any USB drive plugged in while picframe is running.  I think this will solve this issue.  Thanks

LeoColman, 2025-04-06:
https://dev.to/leocolman/expand-your-digital-picture-frame-use-cases-with-pi3d-integrated-usb-media-stick-support-using-2737

https://www.thedigitalpictureframe.com/expand-your-digital-picture-frame-use-cases-with-pi3d-integrated-usb-media-stick-support/

LeoColman, 2025-04-08:
Sorry, the script isn't mine, it's wolfgang's. I believe their contact can be found on https://www.thedigitalpictureframe.com

I use udisks2 / udiskie which will get directly from the USB AFAIK. I explain how I do it in the dev.to article