helgeerbe/picframe #463: Peripherals in configuration.yaml does not provide a clean Exit out of picframe

ddavistx82, 2025-04-05:
I have tried to use the peripherals in the configuration.yaml file but with only partial success.  I have tried the mouse option but that defaults with the menu visible and the mouse on screen.  You have to manually move the mouse pointer off the screen to get rid of the pointer.  Prior version started with the menu off and you had to manually initialize and move the mouse for the menu to appear.  If I set the section to keyboard, the Pause and PowerOff buttons work correctly.  However, the Exit seems to shell out to a black window (labwc?).  A mouse click will then small box with Reconfigure/Exit options.  Reconfigure does nothing.  An Exit here closes the window to the console but then picframe restarts.  Is there any way to cleanly exit and shutdown picframe, returning to the console.  This would really help if something goes wrong with the network connection and you can't ssh back in.  A keyboard plugged directly in could get you back to the console to troubleshoot the network connection.

paddywwoof, 2025-04-06:
@ddavistx82 yes I think there is a problem there, and I will have a play around to see what we can do. Certainly the mouse pointer should be fixable. As you suggested, the black window is related to only stopping the python process but leaving the the window manager running Wayland. I'm a conflicted about baking specific hacks into the code to cope with specific configurations, but to manually do what you suggest (plug in keyboard for fixing or debugging):
1 plug keyboard in USB
2 Ctrl-Alt-F4 to open new terminal
3 log in as pi
4 systemctl --user stop picframe.service
5 Ctrl-Alt-F1 to return to previously locked terminal

ddavistx82, 2025-04-06:
Thanks.  That works.  I have a small Bluetooth enabled micro keyboard that functions as an alternate remote control for the picframe running on the wall.  For the copies that are running at family member's homes, this lets me talk them through how to update their wifi passwords when that happens and they lose network connectivity to the picframe.  Generally, I have them use the iOS Shortcuts app and MQTT to use as a remote control.  While not high on an ask list, adding to the Peripherals option for Back and Next would be nice.  Using the small keyboard as a remote control, it can currently do Pause and PowerOff.  Adding Back and Next would round out the most often used functions for a remote control.

paddywwoof, 2025-04-06:
@ddavistx82 It's a bit of life-saver when hacking things on picframe that are to do with the wayland display (i.e. shutting down and restarting). If you try running picframe from the desktop you sometimes 'lose' it! In that case you have to work your way through the function keys until you find it. I think the current version has the default terminal in F1 but to actually see the desktop you need to go to F7.

You can do Back and Next with the setup now. With the mouse you click  on the left or right side (below the menu activation area at the top of the screen) With the keyboard there are shortcuts 'a' for back and 'd' for next (don't ask me what they stand for I didn't write this bit originally) I added code so that the RIGHT and LEFT keys did back and next. Also ESC with top python (but you will get systemctl restarting as you discovered before).

I fixed the mouse pointer easily enough with `sdl2.SDL_ShowCursor(sld2.SDL_DISABLE)` then a timer for mouse movement and `self.__gui.show_pointer = False` or `True`. I will put that in the next release.

closed 2025-05-05 by github-actions[bot]