helgeerbe/picframe #464: Installation Issues with on Raspberry Pi 4B with Raspbian

Quietscher, 2025-04-06:
Installation Issues with on Raspberry Pi 4B with Raspbian.

It breaks when configuring opencv.

```
 Copying files from CMake output
      Traceback (most recent call last):
        File "/home/admin/myenv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
          main()
        File "/home/admin/myenv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/admin/myenv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 280, in build_wheel
          return _build_backend().build_wheel(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-3oottz8i/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 230, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-3oottz8i/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-build-env-3oottz8i/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 541, in <module>
          main()
        File "setup.py", line 270, in main
          setup(
        File "/tmp/pip-build-env-3oottz8i/overlay/lib/python3.11/site-packages/skbuild/setuptools_wrap.py", line 700, in setup
          _classify_installed_files(
        File "setup.py", line 415, in _classify_installed_files_override
          with open(config_py, 'w') as opencv_init_config:
               ^^^^^^^^^^^^^^^^^^^^
      FileNotFoundError: [Errno 2] No such file or directory: '_skbuild/linux-aarch64-3.11/cmake-install/python/cv2/config-3.py'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for opencv-python
Failed to build opencv-python
ERROR: Failed to build installable wheels for some pyproject.toml based projects (opencv-python)
(myenv) admin@raspberrypi:~ $ sudo pip3 install picframe
```

Quietscher, 2025-04-06:
https://github.com/opencv/opencv-python/issues/1093

paddywwoof, 2025-04-06:
That command line instruction looks wrong to me. I would expect
```bash
python -m pip install picframe
```
If you invoke sudo I think you will muddle the installations for `root` and `pi`. Are you following Wolfgang's manual install instructions?

Paddy

Quietscher, 2025-04-06:
I tried many different ones.
Also yours or just 
Pip install ...
Or also pip3 install

Quietscher, 2025-04-06:
Furthermore I had to activate python venv otherwise nothing was running

paddywwoof, 2025-04-06:
Yes you *definitely* have to activate `venv` before you do any installs. Are you using the lite 64bit standard raspberry pi OS. There seem to be missing binaries for some of the PIL functionality in the 32bit version and when python tries to build them it fails with a similar error. Have you installed `libsdl2-dev`? That seems to bring in a lot of graphics related dependencies, though I don't really see open-cv checking for any of that in your error message.

Quietscher, 2025-04-06:
[image]

Quietscher, 2025-04-06:
I did Not install libsdl2-dev so far

paddywwoof, 2025-04-06:
Ah, that might be a problem then. I tried for ages to get picframe to work with the 32bit OS because that's the standard version distributed that supposedly works on all RPis. However the only way I could manage it was to git clone picframe then change the project.toml file to not require  `"pi_heif>=0.8.0",` and you might be able to do the same with `"opencv-python"` but it will break the current video playing code and I don't think it's all `try: except:` blocked.

Is there a reason you need to use the 32bit OS?

paddywwoof, 2025-04-06:
There are some instructions specific to installing open-cv on 32bit Pi OS https://qengineering.eu/install-opencv-on-raspberry-pi.html

Quietscher, 2025-04-07:
I got pi heif to work, but not Opencv

paddywwoof, 2025-04-07:
@Quietscher well done with heif, did you have to compile it or is a 32bit wheel available now?

It's probably worth thinking how to make the requirement for these 'edge case' modules more optional.

Quietscher, 2025-04-09:
ok with 64 bit it installs

Quietscher, 2025-04-09:
but i dont get it to run automatically. which documentation do you mean?

paddywwoof, 2025-04-09:
Hi, Wolfgang's documentation is [all here](https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-wayland-2025-edition-pi-2-3-4-5/) [and one-click version here](https://www.thedigitalpictureframe.com/install-the-pi3d-pictureframe-software-with-one-click-2025-edition-raspberry-pi-2-3-4-5/) To get it to run automatically with the lite (non-desktop) OS, the best way is for `systemctl` to run a script to start the `labwc` window manager and that has an autostart file that runs a bash script to start venv and run picframe. All seems a bit convoluted but simpler versions fail to allow the display to be turned off and on again.

If you have the full desktop version of the OS then you don't need labwc and you can follow the [suggestion here](https://github.com/helgeerbe/picframe/issues/438#issuecomment-2634517015) using `/home/pi/.config/autostart/start_picframe.desktop`. A disadvantage of this technique is that videos playing using vlc will be offset down slightly (which isn't a major problem in reality).

Let me know how you get on

Paddy

Ascendin81, 2025-04-09:
I start picframe using ./picframe in /home/pi/.local/.bin/ and my vlc videos are being slid  down a little when they play.  How do I prevent this? is there another way to start picframe manually, where videos do not slide down? Thank you.

Quietscher, 2025-04-10:
ah i found it

Quietscher, 2025-04-10:
thanks. is there documentation on how to get .mov video files to work?
i cant find any

paddywwoof, 2025-04-10:
@Quietscher .mov should play fine, at least the ones I've used do. What happens if you `vlc video file.mov` in a terminal?

Quietscher, 2025-04-12:
and you didnt need to configure anything?

Quietscher, 2025-04-12:
`vlc Pictures/IMG_3760.mov
VLC media player 3.0.21 Vetinari (revision 3.0.21-0-gdd8bfdbabe8)
[00000055a2002030] main interface error: no suitable interface module
[00000055a1f18560] main libvlc error: interface "globalhotkeys,none" initialization failed
[00000055a1f18560] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00000055a2002030] skins2 interface error: cannot initialize OSFactory
[00000055a2002030] [cli] lua interface: Listening on host "*console".
VLC media player 3.0.21 Vetinari
Command Line Interface initialized. Type `help' for help.
> [0000007f84020220] avcodec decoder error: cannot start codec (h264_v4l2m2m)
[0000007f78002ce0] gles2 generic error: parent window not available
[0000007f78002ce0] xcb generic error: window not available
[0000007f780013f0] gles2 vout display error: parent window not available
[0000007f780013f0] gl vout display error: parent window not available`

Quietscher, 2025-04-12:
```

ffmpeg -codecs | grep 'h264_omx\|h264_v4l2m2m'
ffmpeg version 5.1.6-0+deb12u1+rpt3 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+deb12u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
 DEV.LS h264                 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (decoders: h264 h264_v4l2m2m h264_cuvid ) (encoders: libx264 libx264rgb h264_nvenc h264_omx h264_v4l2m2m h264_vaapi )

```

paddywwoof, 2025-04-12:
Yes, it played without any other configuration. Your messages look like there was an error running VLC but ffmpeg thought you had the codec installed, is that right? Does it look like VLC won't play **any** video, or just the .mov you want it to? Try some from here say https://www.sample-videos.com/ (or anywhere else from internet really)

Quietscher, 2025-04-12:
It does run, I can hear the movies audio. But since picframe is running I won't see a video output so I think the problem is somewhere else. And yes I've also tried installing ffmpeg

Quietscher, 2025-04-12:
But when picframe is showing a video it won't run. It just displays a still image of the video

Quietscher, 2025-04-12:
I have the basic picframe config on a pure raspios 64Bit system.
Idk why it does not work

paddywwoof, 2025-04-12:
That's the 64bit desktop or the 64bit lite version as per Wolfgang's help pages?

Quietscher, 2025-04-12:
Desktop. Just the normal thing from the website via raspi imager

Ascendin81, 2025-04-12:
Follow the instructions on the website, use bookworm 64bit lite version.  Just run the script via SSH on different PC.  It works great!

Ascendin81, 2025-04-14:
I followed the instructions on the one install script step-by-step.  
The only thing that I had to do was install VLC via the SSH manually. 
VLC does not get installed by the script, as it was written when picframe played only pictures.
Good Luck!

paddywwoof, 2025-04-15:
@sapnho now that picframe main branch has video included, does the one click script `# Step 4:` (and the non-automatic instructions) need `sudo apt install vlc` adding?

sapnho, 2025-04-15:
> [@sapnho](https://github.com/sapnho) now that picframe main branch has video included, does the one click script `# Step 4:` (and the non-automatic instructions) need `sudo apt install vlc` adding?

Yes, I will add it this week. Is this the only difference needed for the new version?

sapnho, 2025-04-15:
Ok, I just added vlc to the packages that need to be installed. The script runs a minute and a half longer but everything seems to be fine. Haven't tried the new video functionality yet, though.

closed 2025-05-14 by github-actions[bot]