helgeerbe/picframe #465: Video playback - Portrait mode (but start and end picture in landscape)

jensvonbulow, 2025-04-17:
Hi,

I love the Video playback addition.

I have wanted something like this for a long time. Always thought moving pictures like in the Hogwarts hallways would be great.

Thank you.

I am running picframe under "Wayland" on "Bookworm" on a "Pi 4b" in 64bit mode. (*1)

When the video is found and you select the first and last picture to fade in and out, if the video is in "Portrait" mode then still that you extracted for the fade in is displayed in "landscape" mode. The jump from landscape to portrait is very jarring, never mind that the still is on its side.

Think of a video taken on a cell phone and held normally. If the video is taken while holding the phone in a "landscape" mode then the still and video transition well.

Are you able to extract the rotation from the video and apply it to the still before applying?

I tried reading your code and see where you decide to scale and add the "black bars" to the canvas (in VideoFrameExtractor) but my ability to actually change the code is "constrained".

I look forward to your response.

Regards
Jens

PS: The "Version 2" update to the playback could include the matting and playback inside a frame so that the whole picture frame's look and feel doesn't change between stills and videos... (but for now, the video playback is just amazing)... oh, another thought I had would be to consider the "fade in" time, the "display time" and the "fade out" time and only select a (random) portion of the video to playback... in other words if the video is longer than the total time configured for stills that the video time not be more than this (and if the video is shorter, then the last still is displayed until its time for the next "story") but that can wait till "Version 3".

Something like this

[Fade in] -[Start of video not displayed this time] - [Video Playback] - [End of video not displayed this time] - [Fade out]

or

[Fade in] - [Video Playback] - [Last frame displayed until enough time has passed] - [Fade out]

(did that make any sense)

*1) The reason I mention it is when running under Wayland I see a lot of "unhandled" X11 warnings

<snip>
[0000007f041f6e30] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
00:12:51.995 [xwayland/xwm.c:1384] **Unhandled** NET_WM_STATE property change 322 (_NET_WM_STATE_ABOVE)
00:12:51.996 [xwayland/xwm.c:1384] Unhandled NET_WM_STATE property change 344 (_NET_WM_STATE_BELOW)
00:12:51.997 [xwayland/xwm.c:1384] Unhandled NET_WM_STATE property change 322 (_NET_WM_STATE_ABOVE)
00:12:51.998 [xwayland/xwm.c:1384] Unhandled NET_WM_STATE property change 344 (_NET_WM_STATE_BELOW)
[0000007f54567510] avcodec decoder: Using DRM Video Accel for hardware decoding
00:12:52.184 [render/gles2/renderer.c:158] Created GL FBO for buffer 3840x2160
00:12:52.190 [types/wlr_pointer_constraints_v1.c:383] unconstrained 0x558fdbd7d0
00:12:52.194 [../src/view-impl-common.c:55] [map] identifier=python3.11, title=(null)
00:12:52.203 [xwayland/xwm.c:1660] **unhandled** X11 event: FocusOut (10)
00:12:52.351 [render/gles2/renderer.c:158] Created GL FBO for buffer 3840x2160
00:12:52.397 [render/gles2/renderer.c:158] Created GL FBO for buffer 3840x2160
00:13:18.594 [types/wlr_pointer_constraints_v1.c:373] constrained 0x558fdbd7d0
00:13:18.594 [xwayland/xwm.c:1660] unhandled X11 event: FocusOut (10)
00:13:18.594 [xwayland/xwm.c:1660] unhandled X11 event: FocusOut (10)
00:13:18.596 [xwayland/xwm.c:887] **unhandled** X11 property 279 (WM_STATE) for window 6291478
00:13:18.596 [xwayland/xwm.c:887] unhandled X11 property 279 (WM_STATE) for window 6291478
00:13:30.497 [types/wlr_compositor.c:771] New wlr_surface 0x558fdc2530 (res 0x558fdc3ee0)
00:13:30.498 [xwayland/xwm.c:1752] New xwayland surface: 0x558fdc2530
00:13:30.500 [xwayland/xwm.c:887] unhandled X11 property 279 (WM_STATE) for window 6291478
</snip>

helgeerbe, 2025-04-17:
Yes, there is definitely room for improvements. You're right, I haven't thought of portrait videos, as I don't use videos from mobiles. I will check this, and see what I could do.

Your version 2 request is hard to fulfill. I had hard times to scale and position videos with vlc and use gpu for decoding. The only reliable scaling is fullscreen (cropped, or with black borders) if the video aspect ratio doesn't fit to the display.  Any brilliant ideas are more than welcome. But at the moment I would say, matting and video will not work. 

Writing this a new idea comes to my mind: 

***
NOTE:  Test, if we can resize the video's sdl window container  and use the matting pic as slide in the background.

***

Version 3. Maybe be can set temporarily the display time to video length. So the complete video will be played.

Regarding your X warning. I have no idea what they ment. I assume, they appear while the video is played. So they are coming from vlc.

helgeerbe, 2025-04-22:
Ok, I thought portrait video mode is an easy task, but it's not.

Mobile phones record portrait videos in landscape mode(!) and store orientation in meta data.  It has 4 possible values (0, 90, 180 and 270) and it stores how the device was held when starting the recording

The best solution I found is to use `ffprobe` with `stream_tags=rotate`. 

I will prepare a test branch and so you can test it.

jensvonbulow, 2025-04-22:
Hi.VLC handles the playback perfectly. It's the extracted pictures that are the problem. I am put and about picking up kids from school. Hi can help when I get home?

helgeerbe, 2025-04-22:
I know. But there is no defined general orientation flag for all kind of video formats. When I extract the frames, I have to find the orientation on my own.

You can try this, if you like:

Activate python virtual environment:  like `source path_to_my_virt_env/bin/activate`

and update picframe to the testbranch:

`pip install git+https://github.com/helgeerbe/picframe@video_orientation --upgrade`

helgeerbe, 2025-04-22:
This seems to work for my iPhone

helgeerbe, 2025-04-22:
I don't know, if `ffmpeg` comes with the lite os version. Otherwise you might have to install ist with `sudo apt install ffmpeg`

jensvonbulow, 2025-04-22:
Hi Helge,

Vielen Dank für diene muhe.

This is exactly what was needed. One small change, please…

This just happens to be my first video file, about 40 seconds long and 70 megabytes in size…

<snip>
***@***.***:~ $ ll /home/jens/Pictures/20190519_162358.mp4
-rw-r--r-- 1 jens jens 71637370 Apr 22 15:52 /home/jens/Pictures/20190519_162358.mp4
***@***.***:~ $

***@***.***:~# time ffprobe -hide_banner -loglevel error -select_streams v:0 -show_entries stream_tags=rotate -show_entries side_data=rotation -of json /home/jens/Pictures/20190519_162358.mp4 > json

real    0m39.935s
user    0m39.641s
sys     0m0.264s
***@***.***:~#
</snip>

The ffprobe takes almost 50 seconds to process all the data and creates 117kilobytes of json data that needs to be parsed.

But if you change “side_data” to “stream_side_data” you it only takes 0.4 of a second and returns 203 bytes of data.

<snip>
--- video_streamer.py.jens.20250422     2025-04-22 18:20:21.157700607 +0200
+++ video_streamer.py   2025-04-22 18:21:42.185436149 +0200
@@ -80,7 +80,7 @@
         try:
             cmd = [
                 "ffprobe", "-v", "error", "-select_streams", "v:0",
-                "-show_entries", "side_data=rotation", "-of", "json", self.video_path
+                "-show_entries", "stream_side_data=rotation", "-of", "json", self.video_path
             ]
             result = subprocess.run(
                 cmd,
</snip>

This is the data it returns…

<snip>
***@***.***:~# cat json.stream
{
    "streams": [
        {
            "tags": {

            },
            "side_data_list": [
                {
                    "rotation": -90
                }
            ]
        }
    ]
}
***@***.***:~#
</snip>

I hope that helps.

Regards
Jens

paddywwoof, 2025-04-22:
For what it's worth I found that running `ffmpeg` with Popen and piping the stderr to a variable allowed me to get more or better info. Obviously you have to do hacky parsing to get at it... Unfortunately I can't remember what item of info I was trying to get, probably when we were looking at using duration or fps or something (I can guarantee it wasn't the rotation).

On the same subject, when I looked at the problems installing opencv (on 32 bit system), and realized what a monster it was, I tried using ffmpeg to grab frames (0.1 second from start and end) it was very simple and quick. I think most modules use ffmpeg under the hood anyway!

Paddy

helgeerbe, 2025-04-23:
Hi Jens,

thanks for the hint. You are absolutely right. I will change this.

helgeerbe, 2025-04-23:
Paddy, `ffprobe` is at the end `ffmpeg`. To get rid of opencv we could use something like this:

````python
import subprocess
import numpy as np

def get_frame_as_numpy(video_path, seek_time=None):
    # Build ffmpeg command
    cmd = [
        "ffmpeg",
        "-ss", str(seek_time) if seek_time else "0",  # seek time if specified
        "-i", video_path,
        "-vframes", "1",
        "-f", "image2pipe",
        "-pix_fmt", "rgb24",
        "-vcodec", "rawvideo",
        "-"
    ]

    # Run ffmpeg and capture output
    process = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.PIPE)

    # Get video dimensions (width x height)
    probe = subprocess.run([
        "ffprobe", "-v", "error",
        "-select_streams", "v:0",
        "-show_entries", "stream=width,height",
        "-of", "csv=s=x:p=0",
        video_path
    ], stdout=subprocess.PIPE, text=True)

    width, height = map(int, probe.stdout.strip().split('x'))

    # Convert raw bytes to numpy array
    frame = np.frombuffer(process.stdout, dtype=np.uint8).reshape((height, width, 3))
    return frame

def get_first_and_last_frames(video_path):
    # Get duration using ffprobe
    result = subprocess.run([
        "ffprobe", "-v", "error", "-show_entries",
        "format=duration", "-of", "default=noprint_wrappers=1:nokey=1",
        video_path
    ], stdout=subprocess.PIPE, text=True)
    duration = float(result.stdout.strip())

    first_frame = get_frame_as_numpy(video_path, seek_time=0)
    last_frame = get_frame_as_numpy(video_path, seek_time=duration - 0.1)
    return first_frame, last_frame

# Example usage
first, last = get_first_and_last_frames("input.mp4")
print(first.shape, last.shape)  # Should print something like (720, 1280, 3)

````

But then I have to learn how to scale and rotate the images as numpy arrays. Or should I use PIL instead?

helgeerbe, 2025-04-23:
To get all metadata in a single call:
````python
import subprocess
import json

def get_video_info(path):
    cmd = [
        "ffprobe", "-v", "error",
        "-select_streams", "v:0",
        "-show_entries", "stream=width,height",
        "-show_entries", "stream_side_data=rotation",
        "-show_entries", "format=duration",
        "-of", "json",
        path
    ]
    result = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.PIPE, text=True)
    info = json.loads(result.stdout)

    stream = info["streams"][0]
    width = stream.get("width")
    height = stream.get("height")
    duration = float(info["format"]["duration"])
    rotation = 0
    for item in stream.get("side_data_list", []):
        if "rotation" in item:
            rotation = int(item["rotation"])
            break

    return width, height, duration, rotation
````

I had to retrieve the meta info as json:

````json
{
    "programs": [

    ],
    "streams": [
        {
            "width": 1920,
            "height": 1080,
            "side_data_list": [
                {

                },
                {
                    "rotation": 90
                }
            ]
        }
    ],
    "format": {
        "duration": "1.733333"
    }
}
````

For what ever reason ffprobes failed to return the rotation as csv (width, height, rotation, duration)

jensvonbulow, 2025-04-23:
How would you feel about getting the location information at the same time and populating the database... (I see you are currently hard coding the dimensions at 100x100 and bailing out early)

Is that cheeky of me to ask?

helgeerbe, 2025-04-23:
Well video files and meta data.  Looks more like a mess than in image files. For location on iPhones i found `"com.apple.quicktime.location.ISO6709"`. I guess, this will never be complete for all the cameras and video file formats.

But yes, if we heading into this direction, it makes complete sense to store the meta info in the database, like we do it for images.

paddywwoof, 2025-04-23:
Helge, agree that it makes sense, and we should probably add that in at some stage...

On the ffprobe meta gathering, I think my earlier questions were due to the almost impenetrable documentation. But,, I did `-show_entries stream=width,height,duration -show_entries stream_side_data=rotation` and it produced
```bash
2224x1080x6.990178x-90
6.997333
```
which is quite parsable but maybe not so reliable as json. The duration always arrived after a line feed stuck on the end, though without the rotation being asked for there was a 'dividing' x added i.e.
```bash
2224x1080x6.990178x
6.997333
```
still  perfectly usable to do it that way, I would have thought.

Rotating multiples of 90 degrees with numpy is superficially trivial, but there is a gocha in that it will try to make do with a 'view' and not actually recreate the array in the new orientation. At a later stage (when loading into the GPU array buffer) the result will look unaltered. You just have to make sure you copy the array. i.e.
```python
im = np.array([[1,2,3,4,5,6],[7,8,9,10,11,12],[13,14,15,16,17,18]]) # w=6, h=3
#rotate 90 degrees anticlockwise
im90 = np.rot90(im).copy()
im180 = np.flip(im, (0, 1)).copy()
im270 = np.rot90(np.flip(im, (0, 1))).copy()
# and you can do the same thing with square brackets, which I think copies automatically
im90 = im.T[::-1,:]
```
But... though numpy will be a lot faster than PIL or opencv, it probably doesn't matter here at all. When I did the test before with frame extraction I got ffmpeg to write the images to `/dev/shm/temp1.jpg` then read them using PIL Image and it took a tiny fraction of a second. So I might go down the PIL Image route for the start and end frames and rotate them using `Image.rotate()`

helgeerbe, 2025-04-24:
I removed opencv and use ffmpeg to extract the video frames. Scaling is handled by Pillow.

I had to learn, that for videos in portrait mode, you have only to switch height and width. So no rotation is needed.

Can you test again.

helgeerbe, 2025-04-25:
I added a video meta info object. Containing:

````txt
 A dataclass representing video metadata.

    Attributes:
    -----------
    width : int
        The width of the video in pixels.
    height : int
        The height of the video in pixels.
    duration : float
        The duration of the video in seconds.
    rotation : int
        The rotation angle of the video (e.g., 0, 90, 180, 270).
    title : Optional[str]
        The title of the video from metadata.
    caption : Optional[str]
        The caption/subtitle of the video.
    description : Optional[str]
        The description/comment of the video.
    creation_date : Optional[datetime]
        The creation date of the video or file creation date as fallback.
    gps_coords : Optional[Tuple[float, float]]
        The GPS coordinates (latitude, longitude) where the video was taken.
    location : Optional[str]
        The location name derived from GPS coordinates.
````

Meta data is and creation time for start and end date is logged actually by default.  Please feel free to test with various video files.  Let me know, if you detect other tags e. g. for gps

paddywwoof, 2025-04-26:
Helge, tested this for a while and it seems to work well. Dataclass is nice, I often want to use python dict like a rust struct or javascript object with dot syntax, and for the Pic data we define a class in model.py so we can do that.

Before pushing new version: I noticed that in get_image_meta the method to return the size reloads the file even thought that's done in the constructor, so we should probably tidy that up to save a bit of IO load. Also maybe something needs doing about video playing in the http interface as per alexy's issue post.

helgeerbe, 2025-04-27:
Paddy, in python you can do the same thing in many ways. I use what comes first to my mind. But unfortunately I'm not so consistent in my choices. 

I fixed the size operator in get_image_meta. Size is now a property of the class. 

I will also store the meta info in the database. So we can use the same mechanism like we use for image data. 

For the http interface I'm not sure, if we should return video url or should we return the first frame?

helgeerbe, 2025-04-28:
Merge all changes to dev. Have in mind that you have to run `touch *.mp4`, so that the cache is updated. And you have to run `picframe.start -i ~/` again to get the new webpage.

closed 2025-04-28 by helgeerbe