helgeerbe/picframe #466: FAQ on use of videos in picframe

sapnho, 2025-04-17:
So, I just tried out videos and it's amazing to see it work. Thanks @paddywwoof and @helgeerbe and whoever helped get it done! Wonderful.

I am preparing a blog article on this subject and have a few questions:

- What dimensions and durations work reliably? (Currently testing on a Pi5, iPhone 4k videos are super smooth)
- Which video formats are supported? (MP4, MOV)
- For which screen dimensions does it work? 4k?
- Is matting supported with videos?
- When does a transition happen? When the video ends, or according to the time delay?

sapnho, 2025-04-17:
Just tried on the Pi4: iPhone videos don't work and the CPU utlilisation is almost 4x 100%.

helgeerbe, 2025-04-17:
Hm, I believe on P3 max resolution is 1080.

4-series devices support resolutions up to 1080p at a 60Hz refresh rate, or two 4K displays at a 30Hz refresh rate. You can also drive a single display at 4K with a 60Hz refresh rate if you connect the display to the HDMI0 port and set the hdmi_enable_4kp60=1 flag in /boot/firmware/config.txt.

5-series devices support up to two displays at 4K resolution at a 60hz refresh rate with no additional configuration.

sapnho, 2025-04-17:
Ah, actually, I meant the Pi3D video resolutions that run reliably on the various Pi models. It would be interesting to get some field data.

paddywwoof, 2025-04-17:
Hi, anything that you can run using VLC should run OK. The script looks for extensions in `['.mp4', '.mkv', '.flv', '.mov', '.avi', '.webm', '.hevc']` which I've tested using a few different ones (downloaded from online test video sites and from own sources).

On the Pi5 I'm currently running videos up to 3840×2160 H.264 (High Profile) 30FPS 27s long and they seem to be fine with reasonable processor load.

On the Pi4 that biggest video clocks up 400% CPU and never starts playing. However a smaller version of the identical video 2224×1080 (taken on my phone) will run perfectly smoothly using about 160% CPU

The monitor on both Pis is 1920x1080

I've got some of the test video mixed in with the pictures on the picframe in my kitchen which is using a Pi3 and there is one that plays perfectly smoothly and it's 1920x1080 ITU H.264 30FPS, but it's an animated planet earth so it's mostly black, maybe easier on processing than some videos. However VLC seems to be able to cope with the video dimensions on the Pi3 OK.

Paddy

sapnho, 2025-04-21:
Is this the length or the resolution of a video that is the issue? I uploaded 4-minute videos as well as 10-second clips and I try to establish a relationship with CPU usage.

paddywwoof, 2025-04-21:
I think the resolution is most important. Different compression systems are more demanding so that will make a difference as well. Don't know if the memory capacity of the Pi will make a difference. The length of the video shouldn't matter too much as VLC will just carry on decoding it with whatever CPU load it needs.

We've not talked about audio, and at the moment there's a `--no-audio` argument passed to VLC, but I can't see any reason why someone might not have sound that could be played. I think that would need some kind of audio system running, as well as setting the OS to send audio to the right place., I had an HDMI screen with audio but I couldn't get sound out after fairly minimal testing.

closed 2025-05-20 by github-actions[bot]