helgeerbe/picframe #467: iPhone videos and live photo videos are not shown

ylexus, 2025-04-26:
Many (or possibly all) of my .mov videos from my iPhone are not shown by picframe. Some of them end up in the database, some not, but none are shown - I checked by disabling randomisation, limiting the dates to one day and going next/previous - the files are just skipped. I suspect this is possibly due to the fact that the "time taken" and other EXIF data in those videos is not properly extracted. For example, for a certain live photo (video + photo), the database entries look like this:

| fname                                                                          | last_modified | file_id | orientation | exif_datetime | f_number         | exposure_time | iso  | focal_length      | make  | model             | lens                                                | rating | latitude | longitude | width | height | title | caption | tags | is_portrait | location |
|--------------------------------------------------------------------------------|---------------|---------|-------------|---------------|------------------|---------------|------|-------------------|-------|-------------------|-----------------------------------------------------|--------|----------|-----------|-------|--------|-------|---------|------|-------------|----------|
| /var/storage/data/google-photos/Takeout/incremental/2025/2025-03/IMG_5746.heic |  1743431903.0 |   96578 |           1 |  1743431903.0 | 1.77999997138807 | 1/2232        | 80.0 | 6.764999866370901 | Apple | iPhone 16 Pro Max | iPhone 16 Pro Max back triple camera 6.765mm f/1.78 |        |  <redacted> |   <redacted> |  4284 |   5712 |       |         |      | 1           |          |
| /var/storage/data/google-photos/Takeout/incremental/2025/2025-03/IMG_5746.mov  |  1743431903.0 |   96629 |           1 |           0.0 |              0.0 |               |  0.0 |                   |       |                   |                                                     |        |          |           |   100 |    100 |       |         |      | 0           |          |

The picture (.heic) is displayed dOK, but not the .mov file. Attaching [the actual .mov](https://github.com/user-attachments/assets/28b9969b-aadd-496d-8887-4c47bc90c5ec) file - it does have the date taken data in it.

It is possible that I am simply not able to find it as I am filtering out by date. How can I make picframe display it to diagnose this further?

There are definitely multiple issues here, as some other .mov or .mp4 files shot on same iPhone are not even in the database. I have obliterated the database to eliminate any prior scanning  issues, and it's scanning again now, will update whether the previously missing .mov/.mp4 files are now in the database or not. 

```
(venv_picframe) pi@pframe:~ $ /home/pi/venv_picframe/bin/picframe -v
INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe', '-v']
picframe version:  2025.04.06

Checking required packages......
PIL :  11.1.0
pi3d :  2.54
yaml :  6.0.2
paho.mqtt :  2.1.0
iptcinfo3 :  2.1.4
numpy :  2.2.3
ninepatch : installed, but no version info
pi_heif :  0.21.0
defusedxml :  0.7.1
vlc :  3.0.21203
cv2 :  4.11.0
```

paddywwoof, 2025-04-26:
@ylexus often what I do when testing is put the new files into a subdirectory, but picframe will show the most recent few images (videos) first after a restart of the systemd service, so that sometimes suffices. I saved the video from your link to a Pi4 and renamed it from `.qt` to `.mov`. The still appeared OK but the video struggled, maybe playing a frame or two then some scramble. I couldn't play it properly with VLC on the command line either (just a flash, not rotated). When I put the SD card into a Pi5 it seemed to play fine.

I think there is a real possibility that users will have to convert videos from mobile phones to codec/container/size that they have found to work reliably on their Pi, say H264, MP4, 1920x1440, or whatever. Doing this on laptop with ffmpeg isn't too bad, but obviously a tedious extra step!

Paddy

PS I will have a look at the video playing on http later if I get chance.

ylexus, 2025-04-27:
The problem of the quality of playback is a separate one. Here what I see is that picframe is not even noticing the file. Although, it looks like this only happened because the files existed on the disk before picframe gained the ability to play videos. Once upgraded, it failed to notice re-scan the videos and add them to the database. Videos that appeared after the picframe version upgrade did get scanned and were added to the database. So this is problem 1 to fix, although of course it can be classified as a one-off problem for people that upgrade from the older version with pre-existing videos. The workaround is to delete the database file.

Problem 2 is that the EXIF data from videos is not retrieved, so the date filters don't work and I assume the caption text would not be showing anything interesting about videos files, if that is ever supported.

helgeerbe, 2025-04-27:
@ylexus at the moment, we are handling just the plain video file and don't care about the meta info. That's why the filters don't work. I'm actually working on a meta data extractor for video data. When done, also filter should work.

ylexus, 2025-04-27:
thanks @helgeerbe, feel free to close this if you consider "Problem 1" not important enough to spend time on.

helgeerbe, 2025-04-28:
@ylexus I think this is more a temporary problem. I created some code, that will populate the db with video meta.  If you have the video files already on disk, instead of deleting the db you can run something like  `touch *.mp4`. This will update the timestamps of the files and the cache will be refreshed only for those files.

helgeerbe, 2025-04-28:
Will closed this issue.

closed 2025-04-28 by helgeerbe