helgeerbe/picframe #468: Current image display via http does not work with videos

ylexus, 2025-04-26:
When a video is requested via http://<host>:9000/current_image, the browser simply downloads it as "current_image" file without extension instead of displaying it. I suspect, looking at the code, this is because the "Content-Type" is set to "image" for everything. Not sure whether setting it to "video" will fix it. 

/home/pi/venv_picframe/bin/

I have created an iOS shortcut that uses quick preview action for the content downloaded like this and while for images it works OK, it just displays an empty screen on videos. This is actually my ultimate purpose - it's very convenient to say "siri current photo" and have the currently shown photo immediately shown on the phone with the ability to share/send it.

```
(venv_picframe) pi@pframe:~ $ /home/pi/venv_picframe/bin/picframe -v
INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe', '-v']
picframe version:  2025.04.06

Checking required packages......
PIL :  11.1.0
pi3d :  2.54
yaml :  6.0.2
paho.mqtt :  2.1.0
iptcinfo3 :  2.1.4
numpy :  2.2.3
ninepatch : installed, but no version info
pi_heif :  0.21.0
defusedxml :  0.7.1
vlc :  3.0.21203
cv2 :  4.11.0
```

helgeerbe, 2025-04-27:
@ylexus video is at the moment more or less experimental. At the moment, I don't know what would the best solution. I use the url in my home assistant app to show the actual image. 

 I can think 
- return the first frame as image
- return the movie url with correct content type (haven't tested yet, if home assistant could play the movies as well in a media player card).
- just return a placeholder image (e. g. popcorn icon) - as picframe is not a media server

At the moment I tend more to show the first frame to give you an impression of the content.

ylexus, 2025-04-27:
Taking what you said into account about current_image returning something simple and also accommodating my use case where the original needs to be downloaded, I dare to propose the change: https://github.com/helgeerbe/picframe/pull/469. I would appreciate for an opinion on this. Obviously it's not the final solution, but at least the current_image is backwards compatible in terms of heic->jpg conversion (I suspect this was done because HA does not support HEIC?)

helgeerbe, 2025-04-28:
Thanks for the PR. Guess this is a good compromise at the moment.

ylexus, 2025-05-01:
@helgeerbe or @paddywwoof  - I've raised a small PR related to my latest one that handles larger media downloads: https://github.com/helgeerbe/picframe/pull/476

ylexus, 2025-05-24:
@helgeerbe or @paddywwoof - there are a number of fixes on the dev branch, is it worth merging to main and releasing? how often do you usually release?