helgeerbe/picframe #472: Display freezes when playing some videos

ylexus, 2025-04-29:
On certain videos (not sure if this is repeatable) I can see errors in the log and the display freezes on a frame of that video forever. I can see the mages are rotated in the log/MQTT, but the display no longer updates. Multiple errors seen in the log when this happens, see below.

I can send the videos that caused both occurrences, but since they are private I'd prefer e-mailing them personally.

Video 1:
```
Apr 28 22:23:36 pframe labwc[35525]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Apr 28 22:23:36 pframe labwc[35525]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': Non>
Apr 28 22:23:36 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'Clip #2 (fcp1).mov'}
Apr 28 22:23:36 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58704', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0,>
Apr 28 22:23:36 pframe labwc[35525]: DEBUG:video_streamer:Stopping video
Apr 28 22:23:36 pframe labwc[35525]: DEBUG:video_streamer:Releasing media
Apr 28 22:23:36 pframe labwc[35525]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2012/Clip #2 (fcp1).mov
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7ffee5cbce50>
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7ffee5cbe6d0>
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.util.FixedString.FixedString object at 0x7ffeec44d610>
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x7ffeec44c910>
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x7ffee5cbe1d0>
Apr 28 22:23:37 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7ffee5cbc350>
Apr 28 22:23:38 pframe labwc[35525]: DEBUG:video_streamer:Playing video: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2012/Clip #2 (fcp1).mov
Apr 28 22:23:38 pframe labwc[35525]: [h264_v4l2m2m @ 0x7ffea4328270] Could not find a valid device
Apr 28 22:23:38 pframe labwc[35525]: [h264_v4l2m2m @ 0x7ffea4328270] can't configure decoder
Apr 28 22:23:38 pframe labwc[35525]: [00007ffea431f260] avcodec decoder error: cannot start codec (h264_v4l2m2m)
Apr 28 22:23:38 pframe labwc[35525]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7ffee5cbce50>
Apr 28 22:23:39 pframe labwc[35525]: [00007ffe980062c0] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Apr 28 22:23:50 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
Apr 28 22:23:51 pframe labwc[35525]: [00007ffea431f260] main decoder error: buffer deadlock prevented
Apr 28 22:23:52 pframe labwc[35525]: [00007ffe800db9e0] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Apr 28 22:23:57 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1002 ms)
Apr 28 22:24:03 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1012 ms)
Apr 28 22:24:16 pframe labwc[35525]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Apr 28 22:24:16 pframe labwc[35525]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 4.0, 'EXIF ExposureTime': '10/1250', 'EXIF ISOSpeedRatings': 100.0, 'EXIF FocalLengt>
Apr 28 22:24:16 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'DSC01407.JPG'}
Apr 28 22:24:16 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58704', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0,>
Apr 28 22:24:17 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1021 ms)
Apr 28 22:24:17 pframe labwc[35525]: [h264 @ 0x7ffea44e79a0] Missing reference picture, default is 2147483647
Apr 28 22:24:17 pframe labwc[35525]: [h264 @ 0x7ffea44e79a0] Missing reference picture, default is 2147483647
Apr 28 22:24:17 pframe labwc[35525]: [h264 @ 0x7ffea44e79a0] Missing reference picture, default is 2147483647
Apr 28 22:24:17 pframe labwc[35525]: [h264 @ 0x7ffea44e79a0] Missing reference picture, default is 2147483647
Apr 28 22:24:17 pframe labwc[35525]: [h264 @ 0x7ffea432bc70] Missing reference picture, default is 2147483647
Apr 28 22:24:18 pframe labwc[35525]: [00007ffea431f260] main decoder error: buffer deadlock prevented
Apr 28 22:24:28 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1041 ms)
Apr 28 22:24:40 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1058 ms)
Apr 28 22:24:42 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1058 ms)
Apr 28 22:24:47 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1063 ms)
Apr 28 22:24:56 pframe labwc[35525]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Apr 28 22:24:56 pframe labwc[35525]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': Non>
Apr 28 22:24:56 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'IMG_3783.MOV'}
Apr 28 22:24:57 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58704', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0,>
Apr 28 22:25:11 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1070 ms)
Apr 28 22:25:11 pframe labwc[35525]: [00007ffea431f260] main decoder error: buffer deadlock prevented
Apr 28 22:25:12 pframe labwc[35525]: [00007ffe74017200] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Apr 28 22:25:17 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1074 ms)
Apr 28 22:25:18 pframe labwc[35525]: [00007ffea431f260] main decoder error: buffer deadlock prevented
Apr 28 22:25:19 pframe labwc[35525]: [00007ffebc1897a0] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Apr 28 22:25:24 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1084 ms)
Apr 28 22:25:36 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1087 ms)
Apr 28 22:25:37 pframe labwc[35525]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Apr 28 22:25:37 pframe labwc[35525]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': 51.587, 'longitude': -0.3815, 'location': 'Pinner, London, England, United Kingdom', 'EXIF FNumber': 2.0, 'EXIF ExposureTime': '1/122', 'EXIF I>
Apr 28 22:25:37 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'IMG_3639.HEIC'}
Apr 28 22:25:37 pframe labwc[35525]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58704', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0,>
Apr 28 22:25:42 pframe labwc[35525]: [0000000010b6c5b0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1092 ms)
Apr 28 22:25:46 pframe labwc[35525]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
[... cut ...]
Apr 28 20:07:59 pframe labwc[27319]: 06:57:13.517 [ERROR] [xwayland/xwm.c:1633] xcb error: op ChangeProperty (no minor), code Window (no extension), sequence 1527, value 4194312
Apr 28 20:07:59 pframe labwc[27319]: 06:57:13.517 [ERROR] [xwayland/xwm.c:1633] xcb error: op ChangeProperty (no minor), code Window (no extension), sequence 1530, value 4194312
Apr 28 20:07:59 pframe labwc[27319]: 06:57:13.517 [ERROR] [xwayland/xwm.c:1633] xcb error: op DeleteProperty (no minor), code Window (no extension), sequence 1531, value 4194312
```

helgeerbe, 2025-04-29:
Hm, looks like vlc is struggling. On thing that comes to my mind, is that we create a window handle for the vlc player. MacOs is quite restrictive and doesn't allow to gui operations in different threads. Linux is more relaxed, but it is recommended to do the same. Unfortunately pi3d is a single window lib.

To come around this limitation I'm thinking about to write a small player program that can run in its own process context. Picframe will then jus send simple commands like: load flile, play, pause, stop and quit.

helgeerbe, 2025-04-29:
Do you see the problems, wenn playing the videos in vlc as standalone player or only in combination with picframe?

ylexus, 2025-04-29:
When executed directly, vlc struggles, chops the video sometimes, but plays it until the end. In fact, picframe also played it until the end, but the display froze on the last frame.

vlc output when run directly is:
```
pi@pframe:~ $ DISPLAY=:0 cvlc --fullscreen "/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2012/Clip #2 (fcp1).mov"
VLC media player 3.0.21 Vetinari (revision 3.0.21-0-gdd8bfdbabe8)
xcb_connection_has_error() returned true
[000055567bdd2ce0] main interface error: no suitable interface module
[000055567bce9560] main libvlc error: interface "globalhotkeys,none" initialization failed
[000055567bdd2ce0] dummy interface: using the dummy interface module...
[h264_v4l2m2m @ 0x7ffefc044400] Could not find a valid device
[h264_v4l2m2m @ 0x7ffefc044400] can't configure decoder
[00007ffefc0150b0] avcodec decoder error: cannot start codec (h264_v4l2m2m)
[00007ffef0002d70] gles2 generic error: parent window not available
[00007ffef0002d70] xcb generic error: window not available
[00007ffef00013f0] gles2 vout display error: parent window not available
[00007ffef00013f0] gl vout display error: parent window not available
[00007ffef00013f0] xcb vout display error: window not available
[00007ffef00013f0] xcb vout display error: window not available
[00007ffef00013f0] drm_vout vout display: <<< OpenDrmVout: I420 1440x1090(1440x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1440x1090(1440x1080 @ 0,0 4/3)
[00007ffef00013f0] drm_vout vout display error: Failed to get xlease
[00007ffef00013f0] drm_vout vout display error: Failed to set atomic cap
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
[00007ffefc0150b0] main decoder error: buffer deadlock prevented
[00007ffef003ade0] gles2 generic error: parent window not available
[00007ffef0033420] xcb generic error: window not available
[00007ffef0039c60] gles2 vout display error: parent window not available
[00007ffef0039c60] gl vout display error: parent window not available
[00007ffef0039c60] xcb vout display error: window not available
[00007ffef0039c60] xcb vout display error: window not available
[00007ffef0039c60] drm_vout vout display: <<< OpenDrmVout: I420 1440x1090(1440x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1440x1090(1440x1080 @ 0,0 4/3)
[00007ffef0039c60] drm_vout vout display error: Failed to get xlease
[00007ffef0039c60] drm_vout vout display error: Failed to set atomic cap
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1011 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1012 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1015 ms)
[00007ffefc0150b0] main decoder error: buffer deadlock prevented
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1028 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1031 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1038 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1039 ms)
[00007ffefc0150b0] main decoder error: buffer deadlock prevented
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1046 ms)
[00007ffefc0150b0] main decoder error: buffer deadlock prevented
[00007ffef005a880] gles2 generic error: parent window not available
[00007ffef005a880] xcb generic error: window not available
[00007ffef003ade0] gles2 vout display error: parent window not available
[00007ffef003ade0] gl vout display error: parent window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] drm_vout vout display: <<< OpenDrmVout: I420 1440x1090(1440x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1440x1090(1440x1080 @ 0,0 4/3)
[00007ffef003ade0] drm_vout vout display error: Failed to get xlease
[00007ffef003ade0] drm_vout vout display error: Failed to set atomic cap
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1047 ms)
[00007ffefc0150b0] main decoder error: buffer deadlock prevented
[00007ffef0033420] gles2 generic error: parent window not available
[00007ffef0033420] xcb generic error: window not available
[00007ffef003ade0] gles2 vout display error: parent window not available
[00007ffef003ade0] gl vout display error: parent window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] drm_vout vout display: <<< OpenDrmVout: I420 1440x1090(1440x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1440x1090(1440x1080 @ 0,0 4/3)
[00007ffef003ade0] drm_vout vout display error: Failed to get xlease
[00007ffef003ade0] drm_vout vout display error: Failed to set atomic cap
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1054 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1063 ms)
[00007fff0c000ca0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1070 ms)
[... cut ...]
[00007ffef003ade0] gl vout display error: parent window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] xcb vout display error: window not available
[00007ffef003ade0] drm_vout vout display: <<< OpenDrmVout: I420 1440x1090(1440x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1440x1090(1440x1080 @ 0,0 4/3)
[00007ffef003ade0] drm_vout vout display error: Failed to get xlease
[00007ffef003ade0] drm_vout vout display error: Failed to set atomic cap
```

ylexus, 2025-04-29:
I think the first thing to do is probably add support for disabling the video playback completely (skip all videos) as I can imagine people can be struggling with the new version. Once I made picframe rescan all my library (with may videos), it does not take more than a few hours for it to discover a "bad" video and hang, so it's unusable for me as is really.

Actually realised that I could go back to the version without video support, although I hate rolling back versions, I would rather help with the fix somehow. Shame I am not a python developer (I am java).

helgeerbe, 2025-04-30:
If you don't want to move ore rename your video files, you could change line 38 in `video_streamer.py` in something like:

````python
# VIDEO_EXTENSIONS = ['.mp4', '.mkv', '.flv', '.mov', '.avi', '.webm', '.hevc']
VIDEO_EXTENSIONS = []
````

To find `video_streamer.py`:

````sh
# activate your virtual environment 
source venv_picframe/bin/activate
# find you active site-packages folder
python -c "import sysconfig; print(sysconfig.get_paths()['purelib'])"
/home/pi/venv_picframe/lib/python3.11/site-packages
cd /home/pi/venv_picframe/lib/python3.11/site-packages
````

If pure vlc struggles with some of your video files, picframe will fail also. So it might be a good idea to to reencode those files with `ffmpeg`

I notice also some conflicts with the movie player and the pi3d library. On some occasions I see some flickering and the still image is shown while the video is played. My assumption that there is somme resource conflicts with the window handles. That's the reason why I want the movie player run in a complete separate prozess. The actual solution seems not to be threadsafe.

ylexus, 2025-04-30:
Thanks, I will probably re-encode the files, although I am yet to determine what exactly makes vlc struggle possibly it's the lack of hardware acceleration for some of the codecs. 

But the struggling vlc should not make the whole process stop displaying any further pictures. So externalising vlc in a separate process sounds like a good idea, or otherwise maybe errors from the calls to vlc are not handled correctly? I have no idea how this all works obviously, but the general idea that bad or unsupported user data should not break software is a good principle.

ylexus, 2025-05-01:
Interestingly, VLC struggles to play this video even on my MacBook Pro. it plays choppy and media info window goes haywire. It plays really smoothly in QT player though, so suspect a bug in VLC. I found many other H264 videos with exactly same resolution and some play absolutely fine on RPi5 and some struggle. 

I can see you've done some work on externalising the vlc process, so instead of re-coding all my h264 videos I'll just wait for this change to be released.

ylexus, 2025-05-16:
I thought I have seen commits related to a vlc in a separate process?

ylexus, 2025-05-24:
@helgeerbe I found that all the videos that VLC struggles with mpv plays extremely smoothly and with no problems. However, I found some short MP4 videos that vlc plays fine but mpv just shows blue screen. Would it be relatively easy to, for now, map an extension to a player, now that you start the player as an external process anyway?

ylexus, 2025-05-25:
Basically, it's this VLC issue: https://code.videolan.org/videolan/vlc/-/issues/8391. it's 12 years old (!).

helgeerbe, 2025-05-25:
12 years, this is some kind of a history.

I had a short glimpse on the mpv docu. It should be quite easy to get the player state over sockets. But mpv always manage its own window. So if we stick with only fullscreen, one could replace the video_player.py with an alternative implementation. 

But vlc offers to render directly into a window handle. So in future, we could make the window smaller and place it everywhere on the screen. That means picframe can render the matting and vlc is only an overlay just hiding the real image. This will not work with mpv, at least not reliable what I have read so far.

helgeerbe, 2025-05-30:
I don't see any error or exception in the log. Looks like the player just doesn't send the Ended state. But I don't know why.

Using visual studio code, it's quite easy to debug via ssh the code on the pi. Heap or stack trace would of no help. Looks like the player is running and in play state.

Does this video play well in standalone vlc?

ylexus, 2025-05-30:
I use both http and MQTT. 

To continue my log extract from the above, the main process continued going through pictures and other videos, but the screen only showed the last frame from that video (`Photos from 2010/Clip #120 (fcp1).mov`):
```
May 30 08:59:52 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 1 ms for update
May 30 08:59:52 pframe labwc[57471]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': 34.5895, 'longitude': 32.9386, 'location': 'Akrotiri British Sovereign Base Area, Cyprus', 'EXIF FNumber': 6.3, 'EXIF ExposureTime': '1/640', 'E
XIF ISOSpeedRatings': 200.0, 'EXIF FocalLength': '150.0', 'EXIF DateTimeOriginal': 1225204410.0, 'Image Model': 'NIKON D70', 'Image Make': 'NIKON CORPORATION', 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
May 30 08:59:52 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'DSC_5549.jpg'}
May 30 08:59:52 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58882', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 
'fade_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
May 30 08:59:57 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 08:59:59 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 08:59:59 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:00:08 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:00:10 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:00:10 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:00:19 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:00:21 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:00:21 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:00:29 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:00:31 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:00:31 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:00:32 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
May 30 09:00:32 pframe labwc[57471]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 4.0, 'EXIF ExposureTime': '10/10000', 'EXIF ISOSpeedRatings': 100.0, 'EXIF FocalLengt
h': '20.3', 'EXIF DateTimeOriginal': 1064931254.0, 'Image Model': 'CYBERSHOT', 'Image Make': 'SONY', 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
May 30 09:00:32 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'DSC07519.JPG'}
May 30 09:00:32 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58882', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 
'fade_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
May 30 09:00:39 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:00:41 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:00:41 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:00:50 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:00:52 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:00:52 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:01:01 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:01:03 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Updating cache
May 30 09:01:03 pframe labwc[57471]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
May 30 09:01:11 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
May 30 09:01:12 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Update location: Wait for db 0 ms and need 1 ms for update
May 30 09:01:12 pframe labwc[57471]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
May 30 09:01:12 pframe labwc[57471]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': 50.6186, 'longitude': -2.244, 'location': 'West Lulworth, Dorset, England, United Kingdom', 'EXIF FNumber': 2.0, 'EXIF ExposureTime': '1/358', 'EXIF ISOSpeedRatings': 20.0, 'EXIF FocalLength': '6.0', 'EXIF DateTimeOriginal': 1626279544.0, 'Image Model': 'iPhone 11 Pro Max', 'Image Make': 'Apple', 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
May 30 09:01:12 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'IMG_6557.HEIC'}
May 30 09:01:12 pframe labwc[57471]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58882', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 'fade_time': 1.0, 'brightness': 1 [... cut ...]
[00007fff345205a0] drm_vout generic error: Failed to set atomic cap
[00007fff348d0540] mmal_xsplitter vout display: Drm no go
[00005555c49ba8b0] main playlist: end of playlist, exiting
pi@pframe:~ $ 
```

I tried moving the video to a subdirectory and switching picframe to that sub-directory via web interface, and it then happily loops through this video, i.e. plays it continuously. I then added a photo to this directory it then loops through the photo and the video happily. So the issue is intermittent.

helgeerbe, 2025-05-30:
Thanks for the valuable input. Problem seems to be, that the player does not detect "end of paying". The rest seems to work. When you click next, picframe sends a stop event to the player, which seems to be received and the actuaul video is stopped. 

So the big question is, why does the player doesn't recognize the end of the playing. I have to review the code

helgeerbe, 2025-05-30:
I found this on the net:

The issue you're seeing — that vlc.State.Ended is not reliably detected — is a known limitation of polling player.get_state() in python-vlc. Instead of polling get_state(), use player.event_manager() to register a callback for the MediaPlayerEndReached event.

I will give it a try, as I don't see that I miss a player state.

ylexus, 2025-06-01:
Thanks, I've installed this branch, will keep it running for a while. I'd say if you do not hear from me for a week then this is fixed.

Quick question  - why is there a long delay (something like 5 seconds) between displaying the first frame (form the .frame file) and the start of the video playback? Is it some workaround or is it there by design? Looks awkward to me, like the video is frozen before it starts.

ylexus, 2025-06-01:
OK so it happened again, same symptoms. The video froze on the last frame, and in the background picframe seems to have kept walking through the next images. No more logging from `video_streamer` until I pressed Next and then everything resumed.

```
Jun 01 13:06:50 pframe labwc[1573]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': None, '
EXIF DateTimeOriginal': 0.0, 'Image Model': None, 'Image Make': None, 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 01 13:06:50 pframe labwc[1573]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'Clip #165.mov'}
Jun 01 13:06:50 pframe labwc[1573]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58885', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 'fa
de_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
Jun 01 13:06:50 pframe labwc[1573]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2010/Clip #165.mov
Jun 01 13:06:50 pframe labwc[1573]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2010/Clip #165.mov took 0.394 seconds
Jun 01 13:06:50 pframe labwc[1573]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '21.1s', 'rotation': 0, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2019, 12, 19, 23, 42, 59), 'gps': None, 'f_number'
: None, 'make': None, 'model': None, 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7fff1b0d0710>
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7fff1b2bb110>
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.util.FixedString.FixedString object at 0x7fff1b2e4ed0>
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x7fff19a717d0>
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x7fff19a70590>
Jun 01 13:06:52 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7fff234e4950>
Jun 01 13:07:03 pframe labwc[1573]: DEBUG:video_streamer:[player] [h264_v4l2m2m @ 0x7fff78536a00] Could not find a valid device
Jun 01 13:07:03 pframe labwc[1573]: DEBUG:video_streamer:[player] [h264_v4l2m2m @ 0x7fff78536a00] can't configure decoder
Jun 01 13:07:03 pframe labwc[1573]: DEBUG:video_streamer:[player] [00007fff78536230] avcodec decoder error: cannot start codec (h264_v4l2m2m)
Jun 01 13:07:03 pframe labwc[1573]: DEBUG:video_streamer:[player] [00007fff60150050] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Jun 01 13:07:03 pframe labwc[1573]: INFO:video_streamer:Video player started in 0.100 seconds.
Jun 01 13:07:03 pframe labwc[1573]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7fff1b0d0710>
Jun 01 13:07:05 pframe labwc[1573]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
Jun 01 13:07:07 pframe labwc[1573]: DEBUG:image_cache.ImageCache:Updating cache
Jun 01 13:07:07 pframe labwc[1573]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
Jun 01 13:07:07 pframe labwc[1573]: DEBUG:video_streamer:[player] [000000003cecbab0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
Jun 01 13:07:10 pframe labwc[1573]: DEBUG:video_streamer:[player] [000000003cecbab0] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1007 ms)
Jun 01 13:07:10 pframe labwc[1573]: DEBUG:video_streamer:[player] [00007fff78536230] main decoder error: buffer deadlock prevented
Jun 01 13:07:11 pframe labwc[1573]: DEBUG:video_streamer:[player] [00007fff44c090d0] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Jun 01 13:07:20 pframe labwc[1573]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
Jun 01 13:07:22 pframe labwc[1573]: DEBUG:image_cache.ImageCache:Updating cache
Jun 01 13:07:22 pframe labwc[1573]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
Jun 01 13:07:31 pframe labwc[1573]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Jun 01 13:07:31 pframe labwc[1573]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 5.6, 'EXIF ExposureTime': '1/320', 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': '21.
3125', 'EXIF DateTimeOriginal': 1153659430.0, 'Image Model': 'Canon PowerShot S45', 'Image Make': 'Canon', 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 01 13:07:31 pframe labwc[1573]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'Picture 086.jpg'}
[... cut ...] tion/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 01 13:11:31 pframe labwc[1573]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'P1010283.JPG'}
Jun 01 13:11:31 pframe labwc[1573]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58885', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 'fade_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
```

helgeerbe, 2025-06-02:
I add more logs. But in the event the state changes are logged. I see PLAYING but no ENDED.

I will also log all states in main loop which are not playing, ended, stoped  when my internal state tells me vlc is playing. When vlc struggling it might not fire the ended event?

At the moment, I have no real idea. I could kill the player after duration but then I have also to keep pausing into account.

helgeerbe, 2025-06-02:
OK, I log every callback call like this `self.logger.error("VLC event: MediaPlayerEncounteredError")`

and in main loop I check for state mismatch with 

````python
 if (
      state is not None and
      state not in [vlc.State.Error, vlc.State.Stopped, vlc.State.Ended] and
      self.last_state == "PLAYING"
  ):
      self.logger.debug("Current VLC state: %s, but my state is PLAYING", state)
````

Please retry.

ylexus, 2025-06-03:
It happened again but this time it's strange as can't see the world ENDED or any logging about the state anymore across the last 24 hours of logging where a large amount of videos were played. In any case, here's  how the gripped log looks when it "hung":

` journalctl --user-unit picframe.service --since "1 day ago" | grep video`:
```
Jun 03 18:44:53 pframe labwc[98319]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2009/Clip #47 (fcp1).mov
Jun 03 18:44:53 pframe labwc[98319]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2009/Clip #47 (fcp1).mov took 0.292 seconds
Jun 03 18:44:53 pframe labwc[98319]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '144.0s', 'rotation': 0, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2019, 12, 19, 0, 18, 21), 'gps': None, 'f_number
': None, 'make': None, 'model': None, 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
Jun 03 18:45:05 pframe labwc[98319]: DEBUG:video_streamer:[player] [h264_v4l2m2m @ 0x7fff680e7be0] Could not find a valid device
Jun 03 18:45:05 pframe labwc[98319]: DEBUG:video_streamer:[player] [h264_v4l2m2m @ 0x7fff680e7be0] can't configure decoder
Jun 03 18:45:05 pframe labwc[98319]: DEBUG:video_streamer:[player] [00007fff680d5040] avcodec decoder error: cannot start codec (h264_v4l2m2m)
Jun 03 18:45:05 pframe labwc[98319]: DEBUG:video_streamer:[player] [00007fff78035ed0] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Jun 03 18:45:05 pframe labwc[98319]: INFO:video_streamer:Video player started in 0.100 seconds.
Jun 03 18:45:10 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
Jun 03 18:45:10 pframe labwc[98319]: DEBUG:video_streamer:[player] [00007fff680d5040] main decoder error: buffer deadlock prevented
Jun 03 18:45:10 pframe labwc[98319]: DEBUG:video_streamer:[player] [00007fff5c281080] gles2 gl: Initialized libplacebo v4.208.0 (API v208)
Jun 03 19:17:13 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
Jun 03 19:17:14 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
Jun 03 19:17:15 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
Jun 03 19:17:16 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
Jun 03 19:17:17 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
Jun 03 19:17:18 pframe labwc[98319]: DEBUG:video_streamer:[player] [0000000022011240] main input error: input control fifo overflow, trashing type=0
```

Since then, there was nothing else except this "main input error:" message. I think this is all VLC output, and no video-related logging from picframe itself. Also strange that the video length is 2m 24s, but the errors started showing more than half an hour after the playback started. It may be related to my automation turning off the display via MQTT at 19:15:31.

helgeerbe, 2025-06-04:
You are right. Those "main input errors" are messages from vlc, struggling with playing the video. 

Just to get it right. You see "PLAYING" but no "ENDED"? But can you see something like "VLC event: MediaPlayerStopped"?

helgeerbe, 2025-06-04:
Please try again

ylexus, 2025-06-04:
@helgeerbe where did you commit the fix, the lates commit in any branch was 2 days ago.

helgeerbe, 2025-06-05:
Sorry, I forgot to push to github. It's the vlc-event branch.

I add also for vlc "--quiet", and "--verbose0". It's just a shot in the dark. Assumption, vlc was so chatty (due to the errors), it blocks the pipes for the interprocess communication.

ylexus, 2025-06-05:
It happened again. Unfortunately, looks like there's zero useful info in the log. The end of `journalctl --user-unit picframe.service --since "1 day ago" | grep video | less` looks like this:

```
Jun 05 19:48:55 pframe labwc[167260]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Мини-отпуск на Озере и Море/IMG_0474.MP4
Jun 05 19:48:55 pframe labwc[167260]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Мини-отпуск на Озере и Море/IMG_0474.MP4 took 0.276 seconds
Jun 05 19:48:55 pframe labwc[167260]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '3.0s', 'rotation': -90, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2018, 4, 14, 15, 39, 13), 'gps': (51.5842, -0.4318), 'f_number': None, 'make': 'Apple', 'model': 'iPhone 7 Plus', 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
Jun 05 19:49:07 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying
Jun 05 19:49:07 pframe labwc[167260]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: PLAYING
Jun 05 19:49:07 pframe labwc[167260]: INFO:video_streamer:Video player started in 0.102 seconds.
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerEndReached
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:Current VLC state: State.Ended, but my state is PLAYING
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: ENDED
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2011/Clip #44 (fcp3).mov
Jun 05 19:49:10 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 05 19:49:11 pframe labwc[167260]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2011/Clip #44 (fcp3).mov took 0.288 seconds
Jun 05 19:49:11 pframe labwc[167260]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '21.1s', 'rotation': 0, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2019, 12, 20, 0, 1, 25), 'gps': None, 'f_number': None, 'make': None, 'model': None, 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
Jun 05 19:49:23 pframe labwc[167260]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying
Jun 05 19:49:23 pframe labwc[167260]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: PLAYING
Jun 05 19:49:23 pframe labwc[167260]: INFO:video_streamer:Video player started in 0.101 seconds.
```

The non-gripped output after the last line above looks like this:

```
[... cut ...]
Jun 05 19:50:30 pframe labwc[167260]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58884', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, '
fade_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
```

So picframe keeps happily gong through next pictures. I always wondered why it does it by the way. If it thinks the video is playing, why is it changing through next pictures? This is also reflected in MQTT history - it looks like it keeps cycling though pictures.

ylexus, 2025-06-06:
Yes, and if get_time() call fails consider it a dead player and kill immediately

ylexus, 2025-06-08:
Thanks, it plays now, will be watching for freezes. 

There's still a significant delay at the beginning of each video where all I can see is the first frame for about 10 seconds. It's hard to believe this is due to VLC initialising as if I play the same video from the command line it starts immediately. I guess it's a different issue to raise, I will raise it once this one is properly fixed 😎.

ylexus, 2025-06-12:
OK it's definitely working better and happily killing players :), but just now it got stuck again, but this time it not only got stuck, but did not even react to the MQTT display off - as if, MQTT showed that display was off and picframe PAUSED, but the display was ON and showing a frame from the last video played. After an hour (at 23:52:45) I came along pressed display off via MQTT again and it switched off. Then I pressed display ON, it turned on and un-froze. In the log it looked like this (with my comments):

`journalctl --user-unit picframe.service --since "1 day ago" | grep video_streamer | less`:
```
Jun 12 22:50:16 pframe labwc[921]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: ENDED
Jun 12 22:50:16 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 12 22:50:16 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 12 22:51:09 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying

--- this is the video that got stuck starting to play ---
Jun 12 22:51:10 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:Video started playing.
Jun 12 22:51:10 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:vlc current time: 511
Jun 12 22:51:10 pframe labwc[921]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: PLAYING
Jun 12 22:51:10 pframe labwc[921]: INFO:video_streamer:Video player started in 1.047 seconds.

--- 22:51:13 : MQTT command to turn display off was sent by my automation ---

Jun 12 22:51:14 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying
Jun 12 22:51:14 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:Video started playing.
Jun 12 22:51:14 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:vlc current time: 3991

--- no more logging from video_streamer until I press display OFF/ON ---

Jun 12 23:52:59 pframe labwc[921]: DEBUG:video_streamer:[player] ERROR:video_player:vlc is stuck while playing for more than 3 seconds. Stopping it!
Jun 12 23:52:59 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying
Jun 12 23:52:59 pframe labwc[921]: DEBUG:video_streamer:[player] DEBUG:video_player:vlc current time: 5981, last time: 5981
```

It's unclear why the display failed to actually turn off and why video_streamer watchdog itself "froze" and only un-froze a hour later.

ylexus, 2025-06-13:
I have installed a new monitor today and not turning off thing is very likely related to that. I will be experimenting. I even noticed once that if I manually tell home assistant to switch display off, it switches off and then a few moments later switches back on. MQTT/HA status remains OFF. Really weird.

ylexus, 2025-06-15:
I think this branch is no quite stable and all known issues are fixed.

helgeerbe, 2025-06-16:
Ok, I merged it into dev and will prepare a release.

ylexus, 2025-06-16:
Unfortunately, it just happened again. The player got killed, but that did not remove it from display - it just stuck there.

```
Jun 16 18:43:03 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerEndReached
Jun 16 18:43:03 pframe labwc[936]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: ENDED
Jun 16 18:43:03 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 16 18:43:03 pframe labwc[936]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Jun 16 18:43:03 pframe labwc[936]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': None, 'E
XIF DateTimeOriginal': 1576799503.0, 'Image Model': None, 'Image Make': None, 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 16 18:43:03 pframe labwc[936]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'Clip #253.mov'}
Jun 16 18:43:03 pframe labwc[936]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58996', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 'fad
e_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
Jun 16 18:43:03 pframe labwc[936]: DEBUG:viewer_display.ViewerDisplay:Loading video frames: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2010/Clip #253.mov
Jun 16 18:43:03 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerStopped
Jun 16 18:43:04 pframe labwc[936]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2010/Clip #253.mov took 0.293 seconds
Jun 16 18:43:04 pframe labwc[936]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '89.8s', 'rotation': 0, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2019, 12, 19, 23, 51, 43), 'gps': None, 'f_number':
 None, 'make': None, 'model': None, 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7fff1abe4ed0>
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x7fff1b4e6110>
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.util.FixedString.FixedString object at 0x7fff1abc61d0>
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x7fff1b4e5410>
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x7fff23916550>
Jun 16 18:43:05 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7fff1b4c4b90>
Jun 16 18:43:13 pframe labwc[936]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
Jun 16 18:43:15 pframe labwc[936]: DEBUG:image_cache.ImageCache:Updating cache
Jun 16 18:43:15 pframe labwc[936]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
Jun 16 18:43:16 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:VLC event: MediaPlayerPlaying
Jun 16 18:43:18 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:Video started playing.
Jun 16 18:43:18 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:vlc current time: 1388
Jun 16 18:43:18 pframe labwc[936]: DEBUG:video_streamer:[player] INFO:video_player:State changed to: PLAYING
Jun 16 18:43:18 pframe labwc[936]: INFO:video_streamer:Video player started in 1.802 seconds.
Jun 16 18:43:18 pframe labwc[936]: DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x7fff1abe4ed0>
Jun 16 18:43:30 pframe labwc[936]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
Jun 16 18:43:30 pframe labwc[936]: DEBUG:video_streamer:[player] ERROR:video_player:vlc is stuck while playing for more than 3 seconds. Stopping it!
Jun 16 18:43:30 pframe labwc[936]: DEBUG:video_streamer:[player] DEBUG:video_player:vlc current time: 10693, last time: 10693
Jun 16 18:43:32 pframe labwc[936]: DEBUG:image_cache.ImageCache:Updating cache
Jun 16 18:43:32 pframe labwc[936]: DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
Jun 16 18:43:42 pframe labwc[936]: DEBUG:image_cache.ImageCache:Found 0 new files on disk
Jun 16 18:43:44 pframe labwc[936]: DEBUG:image_cache.ImageCache:Update file stats: Wait for 0 ms and need 0 ms for update
Jun 16 18:43:44 pframe labwc[936]: DEBUG:interface_mqtt.InterfaceMQTT:Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 2.4, 'EXIF ExposureTime': '1/228', 'EXIF ISOSpeedRatings': 50.0, 'EXIF FocalLength': '4.2
8', 'EXIF DateTimeOriginal': 1440672187.0, 'Image Model': 'iPhone 4S', 'Image Make': 'Apple', 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 16 18:43:44 pframe labwc[936]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'IMG_2652.JPG'}
[... cut ...]
XIF DateTimeOriginal': 1740394810.317, 'Image Model': None, 'Image Make': None, 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
Jun 16 18:44:24 pframe labwc[936]: INFO:interface_mqtt.InterfaceMQTT:Send image state: {'image': 'IMG_1639.JPG'}
Jun 16 18:44:24 pframe labwc[936]: INFO:interface_mqtt.InterfaceMQTT:Send sensor state: {'directory': 'Takeout', 'image_counter': '58996', 'date_from': -2147385600, 'date_to': 2145916800, 'location_filter': '', 'tags_filter': '', 'time_delay': 40.0, 'fad
e_time': 1.0, 'brightness': 1.0, 'matting_images': 0.01}
```

ylexus, 2025-06-17:
Could you please re-open this?

helgeerbe, 2025-06-17:
Totally strange. This never happened to my system. But I shut down picframe at around midnight.  
@paddywwoof reported a bug, where picframes consumed 100% cpu time. I prepared a fix for it. Don't know, if this influenced the interprocess communication.

ylexus, 2025-06-17:
lmk which branch to try with the 100% cpu fix - think I saw the official release that said this in the release notes?

ylexus, 2025-06-19:
Basically, when this happens, the player is hanging on the vlc.stop:
```
(venv_picframe) pi@pframe:~ $ py-spy dump --pid 127179
Process 127179: /home/pi/venv_picframe/bin/python3 /home/pi/venv_picframe/lib/python3.11/site-packages/picframe/video_player.py --x 0 --y 0 --w 3840 --h 2160 --log_level debug
Python v3.11.2 (/usr/bin/python3.11)

Thread 127179 (idle): "MainThread"
    libvlc_media_player_stop (vlc.py:10049)
    stop (vlc.py:3239)
    check_video_progress (picframe/video_player.py:236)
    run (picframe/video_player.py:285)
    main (picframe/video_player.py:379)
    <module> (picframe/video_player.py:383)
Thread 127216 (idle): "Thread-1 (_stdin_reader)"
    _stdin_reader (picframe/video_player.py:208)
    run (threading.py:975)
    _bootstrap_inner (threading.py:1038)
    _bootstrap (threading.py:995)
```

It may be worth adding something like `'--verbose=2', '--file-logging', '--logfile=vlc-log.txt'`, optionally, to the vlc args, hoping that we discover why exactly VLC hangs. It always hangs on those "struggling" videos, which hints at that VLC itself is not in the good shape. Ultimately the video_streamer should really terminate and restart the player process - I think this was the main purpose of moving it to another process?

StonyManager, 2025-07-14:
I'm running version 2025.5.30. It doesn't seem to play videos at all. If I want to start playing videos, do I upgrade to the latest main version is is there a development version/branch I should use? Thanks.

ylexus, 2025-08-04:
Commenting to keep this alive. Any chance of a fix?

ylexus, 2025-09-20:
Reviving, I am actively watching this issue and hoping for a fix.

ylexus, 2025-10-06:
Not stale

paddywwoof, 2025-10-21:
@ylexus does your patch above still work? Although I really like `Rust`, the py-spy dependency looks very heavyweight for what should be such a simple check. Ideally users wouldn't have to install anything in addition to the python modules needed by picframe. Maybe we could achieve the same thing by getting `video_player` to `rm /dev/shm/video_stopped` when load command runs (and vlc stops properly?) and `touch /dev/shm/video_stopped` when the stop command runs. An external process could just check the age of the ram file.

Does the whole of `picframe` hang, or just `video_player.py`, or just `vlc`? At one stage you said the video was stuck but the slide show carried on behind. If picframe itself is still running then a thread in there might be able to restart the service. If this isn't a big problem for most people then maybe we could include the `monitor_video_player.py` script and allow users to set up the systemd service to start it if they run into problems.

ylexus, 2025-11-06:
Reviving. @paddywwoof @helgeerbe what is the purpose of auto-staling the issues?

paddywwoof, 2025-11-29:
Alexey,

I've added a few lines, and an additional `thread` to `VideoStreamer` as below. It doesn't (hasn't yet) cause problems with the few videos on my system. I will push it up to the `dev` branch for you to see it actually helps with your problem.

Paddy

```diff
--- a/src/picframe/video_streamer.py
+++ b/src/picframe/video_streamer.py
@@ -480,6 +480,7 @@ class VideoStreamer:
         self._proc_stderr = None
         self._stderr_thread = None
         self._state_thread = None
+        self._state_check_tm = None
 
         # Determine log level name for child process
         log_level = logging.getLevelName(self.__logger.getEffectiveLevel())
@@ -517,6 +518,12 @@ class VideoStreamer:
         self._stderr_thread = threading.Thread(target=self._listen_stderr, daemon=True)
         self._stderr_thread.start()
 
+        # Start a thread to check _listen_state not waiting too long for STATE message
+        # listening for lines from stdout is a blocking read so that thread can't spot
+        # that it's got stuck!! PG Nov 2025
+        self._stderr_thread = threading.Thread(target=self._player_killer, daemon=True)
+        self._stderr_thread.start()
+
         if video_path is not None:
             self.play(video_path)
 
@@ -556,6 +563,12 @@ class VideoStreamer:
                 self._is_playing = True
             elif line == "STATE:ENDED":
                 self._is_playing = False
+            self._state_check_tm = time.time() # STATE message should happen at < 3s intervals
+
+    def _player_killer(self):
+        if self._state_check_tm is not None and (time.time() - self._state_check_tm) > 10.0:
+            self.__logger.debug("Suspicious lapse in STATE messages from VideoPlayer")
+            self.kill()
 
     def _listen_stderr(self):
         if not self._proc_stderr:
@@ -581,7 +594,7 @@ class VideoStreamer:
             return
         self._send_command(f"load {video_path}")
 
-        timeout = 15  # seconds
+        timeout = 60  # seconds
         start_time = time.time()
         try:
             while not self.is_playing()
```

ylexus, 2025-12-04:
I think I never published here the thread dump of a player script when this happens, here it is:

```
(venv_picframe) pi@pframe:~ $ py-spy dump --pid 127179 
(venv_picframe) pi@pframe:~ $ py-spy dump --pid 127179
Process 127179: /home/pi/venv_picframe/bin/python3 /home/pi/venv_picframe/lib/python3.11/site-packages/picframe/video_player.py --x 0 --y 0 --w 3840 --h 2160 --log_level debug
Python v3.11.2 (/usr/bin/python3.11)

Thread 127179 (idle): "MainThread"
    libvlc_media_player_stop (vlc.py:10049)
    stop (vlc.py:3239)
    check_video_progress (picframe/video_player.py:236)
    run (picframe/video_player.py:285)
    main (picframe/video_player.py:379)
    <module> (picframe/video_player.py:383)
Thread 127216 (idle): "Thread-1 (_stdin_reader)"
    _stdin_reader (picframe/video_player.py:208)
    run (threading.py:975)
    _bootstrap_inner (threading.py:1038)
    _bootstrap (threading.py:995)
```

So yes if it never tries to call stop() but instead SIGKILLs itself it should work. Let me try this patch.

P.S. while restarting the service hit this again: #557, any chance to fix that one too?

ylexus, 2025-12-17:
@paddywwoof can this now be officially released?

paddywwoof, 2025-12-17:
There was one issue I meant to sort out on the touch screen behaviour. I might get hold of a small and cheap 7inch touch screen as it might be useful for a home assistant display (as I gradually migrate my ropey DIY to a proper system)

But then it can be rolled into main as far as I'm concerned.