helgeerbe/picframe #474: having an issue rotating the screen

siosios, 2025-04-29:
hello,

I'm trying to rotate the screen with display_rotate=3 and disabling  dtoverlay=vc4-kms-v3d, the issue I'm having is when I reboot the RPI the screen rotates but the software does not start so i manually run the sh file and get:

pi@picframe:~ $ ./start_picframe.sh
pi@picframe:~ $ INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Exception in thread Thread-1 (__loop):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 56, in __loop
    self.update_cache()
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 86, in update_cache
    self.__modified_files = self.__get_modified_files(self.__modified_folders)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 388, in __get_modified_files
    found = self.__db.execute(sql_select, (base, extension.lstrip("."), dir, mod_tm)).fetchone()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeEncodeError: 'utf-8' codec can't encode character '\udca0' in position 32: surrogates not allowed
gl_id  b'GL' b'4.5 (Compatibility Profile) Mesa 24.2.8-1~bpo12+rpt2'

Any help would be appreciated!

siosios, 2025-04-29:
going to try this pull request https://github.com/helgeerbe/picframe/pull/414

helgeerbe, 2025-04-30:
Does this solve your issue?

paddywwoof, 2025-04-30:
@siosios do you have a non-standard setup apart from wanting the screen rotated. On the tests I've done, using the 64bit lite OS and following [Wolfgang's instructions](https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/) works OK. That method uses wayland with `wlr-randr` in the startup script. I would be interested to see what happens with the 'rotate everything' system when you set image pairs on and then show some landscape images... presumably you get pairs on landscape images one above the other?

The biggest issue is that picframe thinks it's the right way up so things like text, clock, overlay, menus etc are all 90<sup>o</sup>

siosios, 2025-04-30:
i was able to fix the issue by upgrading from a pi 3b to a pi 5. the 3b would not connect to the monitor with `wlr-randr` no matter what i tried and i tried a lot of different things. so the final result was exactly what was here https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/ after swapping to a pi 5, i also created a cron job to dim the frame in the evenings and brighten it back up in the morning on a set schedule using the commands below

`pi@picframe:~ $ ddcutil setvcp 10 - 100
pi@picframe:~ $ ddcutil setvcp 10 + 100`

so far so good, just wish i didn't need to use the 8gb pi for that

closed 2025-05-29 by github-actions[bot]