helgeerbe/picframe #475: external video player

helgeerbe, 2025-04-30:
@paddywwoof I just created the external player branch with an inital commit `external_player`.

The player itself is started as a stand alone player controlled by simple commands over stdout, stdin from picframe. I also was able to play videos on mac os.
For a quick test, I deployed it on my pi 3 and it runs as well. Only the player window does not fit the display size.

I will be off the next 2 weeks. But if you have some time, feel free to test it, or make some improvements, fixes.

helgeerbe, 2025-05-02:
Just did my last commit for the next 2 weeks. 

- Player runs standalone in its own sdl2 window as a separate program
- On my pi3d the load is around 0.5 - 0.7 while playing video.
  - using `player.setfullscreen(True)` consumes 300% even when no window was shown, so we don't use it
  - make the player a little bit more robust. play and stop timeout at 5 sec, if there is not state change
  - before calling play(), there is a check, if player is still alive. If not, player is restarted. 
    -  I don't have any corrupted video files that makes vlc crash, but this might be an interesting test case.

Further improvements:
- if image is not using full screen (e. g. for matting) we can create the player on position and size of the image. So matting will be seen, while video is playing.

Not in scope:
- text overlay on video. This is handled by pi3d, even than you have to wait a little before video playing starts

Limitation:
- Tested on rasbian  pi 3B+, wayland with xwayland. At the moment vlc does not support sdl2 on wayland so we use X11 calls.
- Player runs natively on mac os with sdl2, but pi3d requires a X11 server
- I doubt, that the player will run on windows, but haven't tested it

sangh, 2025-05-03:
What do you think of doing text overlay by creating a subtitle file for vlc to play directly?

helgeerbe, 2025-05-22:
I refactored the video player functionally.  Now a separate process is started. It runs quite smooth on my pi 3.  I'm curious how it behaves on pi 4 and pi 5. Feel free to test it.

` pip install git+https://github.com/helgeerbe/picframe@external_player --upgrade'

paddywwoof, 2025-05-23:
Helge, I've run this on my Pi5 and would say it seems good. It ran both the 3840 × 2160 H.264 (High Profile) (20460 kbps) and the 960x400 hevc without any issues. Pause, Next, Back etc. all seem to work OK.

helgeerbe, 2025-05-25:
Thanks for the feedback. 

@ylexus I noticed the blink of the last frame as well on my pi 3. Picframe switch from first to last frame, while the video is playing. I test that the video player window is shown, before the player sends its "playing" state. But I had to learn, that this is not guaranteed  by the window manager. On slow systems, there can be a visible delay between. The only solution I can think of is to wait 2-5 seconds after playing has started before picframe switches the frames in the background.

On other issue is, that I see the mousepointer while video is played. I have to check when picframe hast to hide it.

ylexus, 2025-05-26:
I was not able to reproduce the orientation issue with a new frontal camera video I created, so must be only some older ones. I found all the `.frame` files and there were surprisingly a very small amount of them, definitely smaller than the amount of videos played since I switched to this branch. All of the generated frames had correct orientation and aspect ratio. I will be keeping an eye on this.

I did find this error in the log, but this is only a single occurrence:

```
May 24 23:45:30 pframe labwc[349628]: DEBUG:get_video_info:Video metadata extraction for /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv took 0.304 seconds
May 24 23:45:30 pframe labwc[349628]: DEBUG:get_video_info:Video metadata: {'dimensions': '1440x1080', 'duration': '0.0s', 'rotation': 0, 'title': None, 'caption': None, 'creation_date': datetime.datetime(2008, 12, 7, 20, 2, 58), 'gps': None, 'f_number
': None, 'make': None, 'model': None, 'lens': None, 'iso': None, 'exposure_time': None, 'focal_length': None, 'rating': None, 'tags': None}
May 24 23:45:30 pframe labwc[349628]: ERROR:VideoFrameExtractor:Error: Invalid video duration.
May 24 23:45:30 pframe labwc[349628]: WARNING:viewer_display.ViewerDisplay:Failed to retrieve frames from video: /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv
```

helgeerbe, 2025-05-26:
Dealing with video metadata is really a mess. Can you send me the output of:

`ffprobe -v quiet -print_format ini -show_format -show_streams /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv` and `exiftool var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv`

If exiftool is not installed you can install it with `sudo apt install libimage-exiftool-perl`

paddywwoof, 2025-05-27:
I've just checked the Pi I left running with the latest version and it seemed fine apart from a couple of the transition stills. I looked at journalctl and could see that there were repeated logs for problems extracting info on the two webm files then error due to duration 0.0

When I run ffprobe on the files the info is all present and correct. I will look into this more later.

helgeerbe, 2025-05-27:
I pushed the changed to this branch. Now the frames from this video type renders well in picframe.

ylexus, 2025-05-27:
@helgeerbe if it's still relevant,

```
pi@pframe:~ $ ffprobe -v quiet -print_format ini -show_format -show_streams "/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv"
# ffprobe output

[streams.stream.0]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High
codec_type=video
codec_tag_string=[0][0][0][0]
codec_tag=0x0000
width=1440
height=1080
coded_width=1440
coded_height=1080
closed_captions=0
film_grain=0
has_b_frames=1
sample_aspect_ratio=4\:3
display_aspect_ratio=16\:9
pix_fmt=yuv420p
level=40
color_range=unknown
color_space=unknown
color_transfer=unknown
color_primaries=unknown
chroma_location=left
field_order=tt
refs=1
is_avc=true
nal_length_size=4
id=N/A
r_frame_rate=50/1
avg_frame_rate=50/1
time_base=1/1000
start_pts=80
start_time=0.080000
duration_ts=N/A
duration=N/A
bit_rate=N/A
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=N/A
nb_read_frames=N/A
nb_read_packets=N/A
extradata_size=65

[streams.stream.0.disposition]
default=1
dub=0
original=0
comment=0
lyrics=0
karaoke=0
forced=0
hearing_impaired=0
visual_impaired=0
clean_effects=0
attached_pic=0
timed_thumbnails=0
captions=0
descriptions=0
metadata=0
dependent=0
still_image=0

[streams.stream.0.tags]
BPS-eng=13156809
DURATION-eng=00\:00\:38.880000000
NUMBER_OF_FRAMES-eng=1944
NUMBER_OF_BYTES-eng=63942093
_STATISTICS_WRITING_APP-eng=mkvmerge v42.0.0 ('Overtime') 64-bit
_STATISTICS_WRITING_DATE_UTC-eng=2020-01-12 19\:35\:28
_STATISTICS_TAGS-eng=BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES

[streams.stream.1]
index=1
codec_name=ac3
codec_long_name=ATSC A/52A (AC-3)
profile=unknown
codec_type=audio
codec_tag_string=[0][0][0][0]
codec_tag=0x0000
sample_fmt=fltp
sample_rate=48000
channels=2
channel_layout=stereo
bits_per_sample=0
id=N/A
r_frame_rate=0/0
avg_frame_rate=0/0
time_base=1/1000
start_pts=0
start_time=0.000000
duration_ts=N/A
duration=N/A
bit_rate=256000
max_bit_rate=N/A
bits_per_raw_sample=N/A
nb_frames=N/A
nb_read_frames=N/A
nb_read_packets=N/A

[streams.stream.1.disposition]
default=1
dub=0
original=0
comment=0
lyrics=0
karaoke=0
forced=0
hearing_impaired=0
visual_impaired=0
clean_effects=0
attached_pic=0
timed_thumbnails=0
captions=0
descriptions=0
metadata=0
dependent=0
still_image=0

[streams.stream.1.tags]
BPS-eng=256000
DURATION-eng=00\:00\:38.880000000
NUMBER_OF_FRAMES-eng=1215
NUMBER_OF_BYTES-eng=1244160
_STATISTICS_WRITING_APP-eng=mkvmerge v42.0.0 ('Overtime') 64-bit
_STATISTICS_WRITING_DATE_UTC-eng=2020-01-12 19\:35\:28
_STATISTICS_TAGS-eng=BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES

[format]
filename=/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv
nb_streams=2
nb_programs=0
format_name=matroska,webm
format_long_name=Matroska / WebM
start_time=0.000000
duration=38.880000
size=65209504
bit_rate=13417593
probe_score=100

[format.tags]
encoder=libebml v1.3.10 + libmatroska v1.5.2
creation_time=2008-12-07T20\:02\:58.000000Z
```

```
pi@pframe:~ $ exiftool "/var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008/20081207200258.mkv"
ExifTool Version Number         : 12.57
File Name                       : 20081207200258.mkv
Directory                       : /var/storage/data/google-photos/Takeout/Google Photos/Photos from 2008
File Size                       : 65 MB
File Modification Date/Time     : 2025:02:24 12:29:36+00:00
File Access Date/Time           : 2025:05:27 15:05:56+01:00
File Inode Change Date/Time     : 2025:03:16 13:29:50+00:00
File Permissions                : -rwxrwxr-x
File Type                       : MKV
File Type Extension             : mkv
MIME Type                       : video/x-matroska
EBML Version                    : 1
EBML Read Version               : 1
Doc Type                        : matroska
Doc Type Version                : 4
Doc Type Read Version           : 2
Timecode Scale                  : 1 ms
Muxing App                      : libebml v1.3.10 + libmatroska v1.5.2
Writing App                     : mkvmerge v42.0.0 ('Overtime') 64-bit
Duration                        : 0:00:39
Date/Time Original              : 2008:12:07 20:02:58Z
Video Codec ID                  : V_MPEG4/ISO/AVC
Image Width                     : 1440
Image Height                    : 1080
Display Width                   : 1920
Display Height                  : 1080
Video Frame Rate                : 50
Track Number                    : 2
Track UID                       : c030a39308475816
Track Type                      : Audio
Audio Codec ID                  : A_AC3
Default Duration                : 32 ms
Track Language                  : und
Audio Sample Rate               : 48000
Audio Channels                  : 2
Image Size                      : 1440x1080
Megapixels                      : 1.6
```

ylexus, 2025-05-27:
BTW the cursor still shows on videos even with the latest code (`Successfully installed picframe-0+untagged.609.g01e5d6a`, which is the commit made an hour ago)

ylexus, 2025-05-28:
I have found a video that definitely generates .frame files with the wrong aspect ratio. I have just tried re-generating with the very latest version. Unfortunately, the video includes faces of people, so I cannot publicly publish it. @helgeerbe happy to e-mail it to you or something like that, or alternatively let me know what you want me to do.

helgeerbe, 2025-05-29:
Thanks. I dont take `sample_aspect_ratio` into account. My videos have all `sample_aspect_ratio=1\:1` your file has `sample_aspect_ratio=4\:3`.

But solution should be easy. I will prepare a fix.

helgeerbe, 2025-05-30:
Good to know. I will push this branch to dev and prepare a new release.

I will close this issue. If someone hast more issues with video playing. Feel free to open new issues. This is easier to track.