helgeerbe/picframe #477: Ask ChatGPT for location

sapnho, 2025-05-11:
I am currently working on a Python script that would send the current photo to an image lookup in ChatGPT to determine the location if no GPS data is contained in the metadata and if triggered by the user via MQTT.

I am not sure how accurate this will be, but testing will provide the answer.

Once the information has been retrieved, what would be the best way to display the information in PictureFrame? Some sort of overlay?

paddywwoof, 2025-05-11:
Probably easiest to write text to`/dev/shm/clock.txt` but it would be nice to save the result so the chatgpt didn't have to be run again and again. That would mean adding to the `sqlite` dB. See what the results look like first then we can see if worth pursuing.

sapnho, 2025-05-12:
That would be even better. I'll do some testing. Thanks.

closed 2025-06-10 by github-actions[bot]