helgeerbe/picframe #480: Mobile app project

robzonbi, 2025-05-14:
I've taken the idea of developing a mobile app to make a one stop shop for cropping, editing and uploading, to students at the mobile app development, Algonquin college.

They have accepted the idea as their senior project, and have begun research and development. When complete I will offer the app free to the Pi3D picture frame community.

To be absolutely sure to get things right, I'm looking for info on the specific meta data containers that picture frame examines for Date, location and caption. If there are other containers that are relevant we could add those too.

helgeerbe, 2025-05-14:
This sounds pretty cool. Actually I'm away from home and have no access to to my development system. But have a look at 
`get_image_meta.py`

I will be back next week.

helgeerbe, 2025-05-14:
Picframe's actual webserver does not support websockets. But I'm thinking about to replace it and offer the same Mqtt functionallity as a rest api. We could implement also an image upload service. 

Just thinking.

paddywwoof, 2025-05-15:
Yes I would be reluctant to expose the Pi to the internet - probably *because* I'm not less IT savvy. For my mum, I have a very simple python app that checks a few times per hour for emails at given address wtih 'picframe' in the subject and loads attachments that are images into a subdirectory of Pictures.

jgodfrey, 2025-05-15:
Additionally, you can use something like "syncthing" to periodically sync one or more sources (such as a Google Photos album) to multiple picture frames.

robzonbi, 2025-05-15:
Google drive is the method I've suggested for this project. I'll have to get them to look into whether or not there is an API for uploading to google photos. If so I think between Google drive, Google photos and email, most users would find the app useful. I suppose the only issue with email, is anyone with your email could spam your picture frame with gawd knows what images. Lol.

sapnho, 2025-05-15:
It seems like Google changed the API in Google Photos and the automatic sync is no longer possible: https://github.com/rclone/rclone/issues/8434

robzonbi, 2025-05-15:
Ah, well there you go. I think Google photos is the ticket then. They also said apple cloud could be an option if the pi can be synched to it.

helgeerbe, 2025-05-15:
Good points. As always there are many ways. In LAN you could mount the picture folder and simply save the image. I personally use WireGuard on mobiles to create a VPN.  So saving images would work seamless over the internet. 

In Germany most people uses a Fritz Box router for internet access. It has a build in support for WireGuard. To setup the VPN you just have to scan a QR code.

robzonbi, 2025-05-15:
Oh that's cool. I guess trying to make a one size fits all solution means find non-regional solutions. But also giving options for a wide range of users. Ftp transfer was also suggested. And when on the same network as the pi, direct transfer.

tipp0025, 2025-05-17:
I'm the development lead for the project mentioned by @robzonbi. Initially, we plan to focus on transferring the images via cloud, along with allowing cropping and editing metadata through the app, which will likely be cross-platform. We're still in the planning stages, so your feedback is welcome to help shape the final product.
 
I've included a link below to a brief (5-10mins) survey we've created to gain more insight from potential users. Feel free to share any questions or comments about the project, as I'll be checking back regularly on the posts here.

[Survey Link](https://forms.office.com/Pages/ResponsePage.aspx?id=JNkb7GoKqUqqicmAMWwESROZ7FgoNnZNtKAU0uwc2b9URjJaQUZPV1VXWUtCQllBRUZKT1lXT1dQMC4u)

ylexus, 2025-05-24:
Google photos API is very restricted these days. Basically the API does not see anything else other than what it itself uploaded. rclone has a hacky way of bypassing that via https://github.com/rclone/gphotosdl, which starts a browser in the background and pretends to click download button there. It has its issues, but generally works. I use exactly that to download new photos to my storage once a month, and the storage is then seen by picframe via an NFS network mount.