helgeerbe/picframe #481: Monitor power

Duzeper, 2025-05-14:
I am trying to set up timer/service pairs to turn off/on picframe and the monitor at set times as I have done with PictureFrame2020.

Given that `vcgencmd display_power 1(0)`  no longer works, I have, firstly, configured `display_power: 2` in `configuration.yaml`, and then created 2 scripts as per Wolfgang's article (monitor_off.py and monitor_on.py)
In case it's relevant, I am using a RPi 3B+

```
#/home/pi/venv_picframe/bin/python /home/pi/monitor_off.py
import subprocess  # for command execution

def turn_off_monitor():
    try:
        subprocess.call("wlr-randr --output HDMI-A-1 --off", shell=True)
    except Exception as e:
        print(f"An error occurred: {e}")

turn_off_monitor()
```
and
```
#/home/pi/venv_picframe/bin/python /home/pi/monitor_on.py
import subprocess  # for command execution

def turn_on_monitor():
    try:
        subprocess.call("wlr-randr --output HDMI-A-1 --on --mode 1920x1080", shell=True)
    except Exception as e:
        print(f"An error occurred: {e}")

turn_on_monitor()
```
These both run successfully  in terminal so next I created 2 service files to execute them:

```
[Unit]
Description=Frame off service

[Service]
Type=oneshot
#ExecStart=systemctl --user stop picframe.service
ExecStart=/home/pi/venv_picframe/bin/python /home/pi/monitor_off.py

[Install]
WantedBy=default.target
```
and
```
[Unit]
Description=Frame on service

[Service]
Type=oneshot
ExecStart=/home/pi/venv_picframe/bin/python /home/pi/monitor_on.py
#ExecStart=systemctl --user start picframe.service

[Install]
WantedBy=default.target
```
Again, these both start successfully in terminal, with the `picframe.service` lines commented out.
However, when I include `ExecStart=systemctl --user stop picframe.service` in `picframe_off.service`, although `picframe.service`  does stop, the monitor does not power off. Instead it displays the RPI CLI

`journalctl --user -u picframe_off.service` shows no errors

When I include `ExecStart=systemctl --user start picframe.service` in `picframe_on.service` it appears to work as I expected.

So my question is, how do I get picframe to stop and the monitor to power off?
When I say that they appear to be working, that is when run from terminal. This is not entirely the case (see below) when I used the timers

FYI, I ran `journalctl --user -u picframe.service`, I see multiple instances of the following:
`May 14 16:42:04 pi labwc[1977]: WARNING:viewer_display.ViewerDisplay:Unsupported setting for display_power=2.`

However, this is the journal output for the frame_off time (23:30) and frame_on time (09:00)
[journal.txt](https://github.com/user-attachments/files/20211161/journal.txt)
So, although `picframe.service` stopped at 23:30, as I said, the monitor then displayed the CLI and at 09:00, nothing happened, `picframe.service` did not start

Martin

helgeerbe, 2025-05-14:
No, you cannot have two ExecStart= lines in a single [Service] section of a systemd service unit file. If you do, only the last ExecStart= line takes effect, and the others are ignored.

Create a shell script that runs all the commands you need, then call that script from ExecStart=:

Or: If one command is a prerequisite (e.g., setting up something before the main command), use ExecStartPre=:

[Service]
ExecStartPre=/usr/bin/command1
ExecStart=/usr/bin/command2

paddywwoof, 2025-05-15:
Martin, I turn picframe (including the monitor) using a *very* simple python script that I start with systemd. I use a PID connected to the GPIO pin at home, but here (at my mum's house) I just use the time
```python
import time
import paho.mqtt.client as mqtt
from datetime import datetime

CHECKTM = 300
ON_TIME = 3600 * 6.0 # switch on after 6am
OFF_TIME = 3600 * 22.0 # off after 10pm

client = mqtt.Client()
client.connect('localhost', 1883, 60)
client.loop_start() # is this even needed when only publishing?
        
on_status = True

while True:
    # seconds after midnight, this is a very crude method but fine for many purposes
    dtm_now = datetime.now()
    now = dtm_now.hour * 3600 + dtm_now.minute * 60 + dtm_now.second 
    if now > ON_TIME and now < OFF_TIME and not on_status:
        client.publish('homeassistant/switch/picframe_display/set', 'ON')
        on_status = True
    elif (now <= ON_TIME or now >= OFF_TIME) and on_status:  
        client.publish('homeassistant/switch/picframe_display/set', 'OFF')
        on_status = False
    time.sleep(CHECKTM)
```
If you already have something set up as per our discussion on  https://github.com/helgeerbe/picframe/issues/452 then you can just add the time checking into your existing loop and make the display on and off a new method in your `DatCapLoc` class. That way the changes needed are tiny.
```python
        ...
        self.client.loop_start()
        self.on_status = True # in __init__()
        ....
            self.client.publish('homeassistant/switch/picframe_caption_toggle/set', self.dat_cap_loc)

    def picframe_display(self, status): # new method
        self.client.publish('homeassistant/switch/picframe_display/set', status)
        self.on_status = True if status == 'ON' else False

dat_cap_loc = DatCapLoc()

while True:
    time.sleep(5.0)
    dtm_now = datetime.now() # need to import datetime.datetime also setup ON_TIME etc
    now = dtm_now.hour * 3600 + dtm_now.minute * 60 + dtm_now.second 
    if now > ON_TIME and now < OFF_TIME and not on_status:
        dat_cap_loc.picframe_display('ON')
    elif (now <= ON_TIME or now >= OFF_TIME) and on_status: 
        dat_cap_loc.picframe_display('OFF')

```

Paddy

Duzeper, 2025-05-16:
Thanks Paddy.
Can I just clarify:
Firstly, the first half of your code, including `ON_TIME` & `OFF_TIME` is a python script that I should start from a systemd service in the same way as my `dat_cap_loc.py`?

Secondly, the second half should be inserted into `dat_cap_loc.py`, and have I put the code in correct the locations as indicated by my `#New code here` comments? Sorry, but I don't know how to highlight text here to make it clearer.

```
import time
import json
import paho.mqtt.client as mqtt

class DatCapLoc:
    def __init__(self):
        self.dat_cap_loc = 'OFF'
        self.has_gps = False
        self.client = mqtt.Client()
        self.client.on_connect = self.on_connect
        self.client.on_message = self.on_message
        self.client.connect('localhost', 1883, 60)
        self.client.loop_start()

#New code here
        self.on_status = True # in __init__()
#To here
        
    def on_connect(self, client, userdata, flags, rc):
        client.subscribe("homeassistant/sensor/picframe_image/attributes", qos=0)
        client.subscribe("dat_cap_loc", qos=0)
    
    def on_message(self, client, userdata, message):
        msg = message.payload.decode("utf-8")
        if message.topic == "dat_cap_loc":
            self.dat_cap_loc = msg
        elif message.topic == "homeassistant/sensor/picframe_image/attributes":
            attr = json.loads(message.payload)
            if attr["latitude"] is None or attr["longitude"] is None:
                self.has_gps = False
            else:
                self.has_gps = True
        self.publish()

    def publish(self):
        if self.has_gps:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.dat_cap_loc)
            time.sleep(0.1)            
            self.client.publish('homeassistant/switch/picframe_location_toggle/set', self.dat_cap_loc)
        else:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.dat_cap_loc)
            time.sleep(0.1)
            self.client.publish('homeassistant/switch/picframe_caption_toggle/set', self.dat_cap_loc)
            
#New code here  
    def picframe_display(self, status): # new method
        self.client.publish('homeassistant/switch/picframe_display/set', status)
        self.on_status = True if status == 'ON' else False
#To here  
      
dat_cap_loc = DatCapLoc()

while True:
    time.sleep(5.0)
#New code here    
    dtm_now = datetime.now() # need to import datetime.datetime also setup ON_TIME etc
    now = dtm_now.hour * 3600 + dtm_now.minute * 60 + dtm_now.second 
    if now > ON_TIME and now < OFF_TIME and not on_status:
        dat_cap_loc.picframe_display('ON')
    elif (now <= ON_TIME or now >= OFF_TIME) and on_status: 
        dat_cap_loc.picframe_display('OFF')
#To here
```       
Martin

paddywwoof, 2025-05-16:
Martin, kind of. The first example was an approximation for what I already do, and it runs in its own service. The second was what you might need to do to get your existing code to do what you want.

To make the code markup look like python (or other language) you can  add that after the triple back-ticks. I've modded your code in line with what I think would work. To put it into effect you just need to edit it then restart the systemd service that call the python script.

Paddy

Duzeper, 2025-05-16:
Thanks.
As an aside, where in that script does it 'see' that a new image has come up?
I ask because, if I switch 'Date/Caption/Location' on, although it works correctly until an image is displayed where the GPS data, or lack of, changes ie. if I leave the switch active and a previous image contained GPS data, then it still sees `self.has_gps = True` and it displays Date, Caption & Location whether or not there is any GPS data. I assume therefore, that I need to reset `self.has_gps = False`, but I am not sure where.

Martin

Duzeper, 2025-05-16:
Actually they are different. 
The `if self.has_gps:` condition  publishes `picframe_date_toggle/set` and `picframe_location_toggle/set` and the else publishes `picframe_date_toggle/set` and `picframe_caption_toggle/set`
So what appears to happen is that, once the caption_toggle has been set, it never turns off until the Date/Caption/Location switch is turned off, so all subsequent images display Caption even if `if self.has_gps:` is true

paddywwoof, 2025-05-16:
Ah yes. You probably need to turn `location_toggle` on or off in each half of the condition, as well as `caption_toggle`. The best way would be to set local variables for each combination of the two variables in if else blocks then run all three `publish` commands afterwards.

pyrced, 2025-05-17:
I run a python script to turn my monitor on and off using ddcutil. The script also includes controls to adjust the brightness of my display automatically using ddcutil with lux readings from an adafruit bh1750 light sensor.
I chose to use ddcutil because it seemed more universal for display power management as it didn't matter if picframe was running in wayland or x.

To start and stop the picframe service I use some simpy crontab commands:
31 21 * * * XDG_RUNTIME_DIR=/run/user/1000 systemctl --user stop picframe.service
29 06 * * * XDG_RUNTIME_DIR=/run/user/1000 systemctl --user restart picframe.service

paddywwoof, 2025-05-17:
I was fairly sure I tried ddcutil when we were trying to get the screen turn off to work but it wouldn't work. Maybe there was a fix in the latest upgrade in Feb this year. I will check it out again.

Duzeper, 2025-05-17:
Hi Paddy. Sorry that script doesn't work.
I tried running it from a service and nothing happened at the 'OFF' time so I have tried running it from a command line with this result:
```pi@pi:~ $ /home/pi/venv_picframe/bin/python /home/pi/dat_cap_loc.py
/home/pi/dat_cap_loc.py:13: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  self.client = mqtt.Client()
Traceback (most recent call last):
  File "/home/pi/dat_cap_loc.py", line 58, in <module>
    elif (now <= ON_TIME or now >= OFF_TIME) and on_status:
                                                 ^^^^^^^^^
NameError: name 'on_status' is not defined
```
This is the script I am running:
```
import time
import json
import paho.mqtt.client as mqtt
from datetime import datetime

ON_TIME = 3600 * 19.0 # switch on after 7pm For testing
OFF_TIME = 3600 * 18.75 # off after 6:45pm

class DatCapLoc:
    def __init__(self):
        self.dat_cap_loc = 'OFF'
        self.has_gps = False
        self.client = mqtt.Client()
        self.client.on_connect = self.on_connect
        self.client.on_message = self.on_message
        self.client.connect('localhost', 1883, 60)
        self.client.loop_start()
        self.on_status = True # in __init__()
        
    def on_connect(self, client, userdata, flags, rc):
        client.subscribe("homeassistant/sensor/picframe_image/attributes", qos=0)
        client.subscribe("dat_cap_loc", qos=0)
    
    def on_message(self, client, userdata, message):
        msg = message.payload.decode("utf-8")
        if message.topic == "dat_cap_loc":
            self.dat_cap_loc = msg
        elif message.topic == "homeassistant/sensor/picframe_image/attributes":
            attr = json.loads(message.payload)
            if attr["latitude"] is None or attr["longitude"] is None:
                self.has_gps = False
            else:
                self.has_gps = True
        self.publish()

    def publish(self):
        if self.has_gps:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.dat_cap_loc)
            time.sleep(0.1)            
            self.client.publish('homeassistant/switch/picframe_location_toggle/set', self.dat_cap_loc)
        else:
            self.client.publish('homeassistant/switch/picframe_date_toggle/set', self.dat_cap_loc)
            time.sleep(0.1)
            self.client.publish('homeassistant/switch/picframe_caption_toggle/set', self.dat_cap_loc)
             
    def picframe_display(self, status):
        self.client.publish('homeassistant/switch/picframe_display/set', status)
        self.on_status = True if status == 'ON' else False
      
dat_cap_loc = DatCapLoc()

while True:
    time.sleep(5.0) 
    dtm_now = datetime.now() 
    now = dtm_now.hour * 3600 + dtm_now.minute * 60 + dtm_now.second 
    if now > ON_TIME and now < OFF_TIME and not on_status:
        dat_cap_loc.picframe_display('ON')
    elif (now <= ON_TIME or now >= OFF_TIME) and on_status: 
        dat_cap_loc.picframe_display('OFF')
```
Martin

paddywwoof, 2025-05-17:
You've probably figured it out now but the simplest way would be to put the line `on_status = True` after creating the `DatCapLoc()` instance.

I see you didn't try setting variables then follow with the three `client.publish` statements. Did the problem sort itself out? I would show you explicitly what I mean by the suggestion but I don't really know what your intended logic is... something like this:
```python
        date_toggle = 'OFF'
        caption_toggle = 'OFF'
        location_toggle = 'OFF'
        if self.dat_cap_loc == 'ON':
            date_toggle = 'ON'
            caption_toggle = 'ON'
            if self.has_gps:
                location_toggle = 'ON'

        self.client.publish('homeassistant/switch/picframe_date_toggle/set', date_toggle)
        time.sleep(0.1) 
        self.client.publish('homeassistant/switch/picframe_location_toggle/set', location_toggle)
        self.client.publish('homeassistant/switch/picframe_caption_toggle/set', caption_toggle)
```

Duzeper, 2025-05-17:
I think you credit me with rather more skill than I deserve so, no I hadn't got there yet! Who knows, I might actually have worked it out myself eventually, although I have my doubts. Anyway, I will give that a go.

No I haven't got round to setting variables yet as the timing issue is more important to me at the moment. Also, if I leave it long enough, then you do it for me! JUST KIDDING, but thanks anyway.

Martin

Duzeper, 2025-05-18:
The monitor powered of at 23:30 and then powered on at (about) 9:00 OK. I say 'about' 'cos  it took about a minute before it actually came on, but I suspect that that is a monitor power detection issue, nothing to do with picrame. However, when the monitor did come on, picframe was paused.

Also, the journal shows multiple instances of:
```
pi labwc[760]: WARNING:viewer_display.ViewerDisplay:Unsupported setting for display_power=2.
pi labwc[650]: 00:02:09.779 [ERROR] [../src/output.c:773] failed to get output layout box
```
and

` pi labwc[837]: unknown output HDMI-A-1`

Although, given that the monitor does actually power off & on, I assume these are not an issue.

I am, however, a little confused (it must be said that, wrt python programming, this is a pretty much permanent state), in that, in the while loop, although there is picframe_display ON & OFF, there is no reference to picframe_paused, so what is pausing the frame and, more to the point, not unpausing it.

Also, I wonder, is there not a more direct way to determine and interrogate the current time?

paddywwoof, 2025-05-18:
Martin, my glib reply earlier about the easiest way to set the `on_status` was wrong (well right but a good reason not to 'just' do the easiest thing without thinking about it). You already have the attribute in `DatCapLoc` so you should change the condition to `if now > ON_TIME and now < OFF_TIME and not dat_cap_loc.on_status:` (and obviously where it occurs the line below) and **not** create a new global scope variable.

It seems odd that the monitor was paused as that seems to be taken care of in `picframe.Controller.display_is_on()` see https://github.com/helgeerbe/picframe/blob/2c16b92617de7c10171db03315ddd7aea5fbadbd/src/picframe/controller.py#L183C1-L187C33 and I vaguely think that the `display_power=2` warnings were fixed. Have you got the latest version of picframe? (in a terminal `source venv_picframe/bin/activate' then `python` then `import picframe` then `picframe.__version__`)

The time checking here is very clunky, but it's easy to understand and lightweight for processing. The turning on and off should happen within 5.0 seconds using the sleep you put in your while loop. python datetime will get the time from the linux in some way, it probably only stays accurate if you have connection to the internet.

Duzeper, 2025-05-18:
Don't know what version I have 'cos:
```
pi@pi:~ $ source venv_picframe/bin/activate
(venv_picframe) pi@pi:~ $ python
Python 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import picframe
>>> picframe.version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'picframe' has no attribute 'version'. Did you mean: '_version'?
>>> picframe_version
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'picframe_version' is not defined
>>>
```

paddywwoof, 2025-05-18:
Martin, don't know why the backticks in the markup didn't work with `picframe.__version__` they seem to have worked here

Duzeper, 2025-05-18:
That worked. Result - '2024.11.01'

Duzeper, 2025-05-18:
PS. Yes the RPi does/will have internet connection

paddywwoof, 2025-05-18:
The latest version is 2025.4.something. I don't know if it will tidy up those warning or address the unpausing after turning the display back on but you can upgrade it by setting the source venv_picframe then
```python -m pip install picframe --upgrade```

Duzeper, 2025-05-21:
Paddy, FYI, I have confirmed that picframe pauses at the 'OFF' time and unpauses 'at th 'ON' time using MQTT Explorer.
Also, I have modified `dat_cap_loc.py` thus:
```
...
 def publish(self):
        date_toggle = 'OFF'
        caption_toggle = 'OFF'
        location_toggle = 'OFF'
        
        if self.dat_cap_loc == 'ON':
            date_toggle = 'ON'
            caption_toggle = 'ON'
            if self.has_gps:
                caption_toggle = 'OFF'
                location_toggle = 'ON'

        self.client.publish('homeassistant/switch/picframe_date_toggle/set', date_toggle)
        time.sleep(0.1) 
        self.client.publish('homeassistant/switch/picframe_location_toggle/set', location_toggle)
        time.sleep(0.1) 
        self.client.publish('homeassistant/switch/picframe_caption_toggle/set', caption_toggle)
....

```
I wasn't sure if that was where you were suggesting I place the code, however, it does seem to do want I want it to do. As always, however, I am not sure if there will be any unintended consequences if it is wrong, so I'll keep my eyes on it.
Also, I always assumed that the `time.sleep(0.1)` lines between the `client.publish` lines were there to, perhaps, avoid any timing conflict between them (I'm making this up now!), but, anyway, since you had included them originally in https://github.com/helgeerbe/picframe/issues/452, I included them here.
So far, anyway, all seems well.

I cannot thank you enough for all your help with this, there is no doubt that I would not have got anywhwere near this outcome without it.

Martin

closed 2025-06-19 by github-actions[bot]