helgeerbe/picframe #483: Network drive

Duzeper, 2025-05-24:
Is it possible to point `pic_dir` to a network drive, specifically a NAS drive?

Martin

ylexus, 2025-05-24:
@Duzeper sure, any locally mounted network drive will work, picframe does not care. I, for example, use an NFS mount from my other Raspberry Pi which works as a NAS. These days all you need to do is ask ChatGPT "how do I mount my NAS drive share on my raspberry pi so that it stays auto-mounted",  provide a model of your NAS, and it'll give you precise instructions.

ylexus, 2025-05-24:
I am not sure which version of my comment you saw - it looked like a 6 words to me just now, I re-saved it and the whole paragraph is now shown.

Duzeper, 2025-05-24:
No, I'm pretty sure I saw the whole paragraph, and it was a great help, thanks
Martin