helgeerbe/picframe #486: Filtering on date using preset buttons in Home Assistant

sapnho, 2025-05-31:
Hello @helgeerbe, I guess this is an area where I need your expertise.

I want to create four buttons in Home Assistant that allow me to add a filter to the date_from field, so that the frame will only show photos from today, the last 7 days, the last 30 days, and include a reset button.

Today, I use your [wonderful article](https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/) to control it manually, but adding presets would provide another level of user friendliness.

How can I best do this?

The payload for "Today" would be something like
`payload_template: "{{ now().timestamp() | timestamp_custom('%Y:%m:%d') }}"`
`topic: frame/date_from`

However, using this as an MQTT Publish action in HA is not supported resulting in `Failed to perform the action script/1590255929456. extra keys not allowed @ data['payload_template']`

Also, it would be great if this entry field in HA was updated when the MQTT command it send.

[image]

Any thoughts how to best implement that? Thanks!

sapnho, 2025-05-31:
The last issue was fixable by restarting Pi3D. However, when I use MQTT publishing commands in Home Assistant with picframe, the filtering stops working. MQTT analyser doesn't show anything unusual.

helgeerbe, 2025-05-31:
Just checked my installation. Tag, directory, location filtering works in home assistant. What's not working is the date filter. I also have additional buttons for today, last week, and last month. But they don't work. Have to check next week. I did updates on home assistant recently.

helgeerbe, 2025-05-31:
I created 3 input buttons for
- today
- last week
- last month

[image]

This is my automation

```YAML
alias: picframe_quick_date_select
description: ""
triggers:
  - entity_id:
      - input_button.picframe_date_today
    id: today
    trigger: state
  - entity_id:
      - input_button.picframe_date_last_week
    id: last_week
    trigger: state
  - entity_id:
      - input_button.picframe_date_last_month
    id: last_month
    trigger: state
conditions: []
actions:
  - if:
      - condition: trigger
        id: today
    then:
      - data:
          topic: picframe/date_to
          payload: "{{ now().strftime(\"%Y/%m/%d\") }}"
        action: mqtt.publish
      - data:
          topic: picframe/date_from
          payload: "{{ now().strftime(\"%Y/%m/%d\") }}"
        action: mqtt.publish
    else:
      - if:
          - condition: trigger
            id: last_week
        then:
          - data:
              topic: picframe/date_from
              payload: >-
                {% set first_day_of_week = (now().strftime("%w")) | int %} {{
                as_timestamp(now() - timedelta(days=first_day_of_week, weeks=1))
                | timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
          - data:
              topic: picframe/date_to
              payload: >-
                {% set first_day_of_week = (now().strftime("%w")) | int %} {{
                as_timestamp(now() - timedelta(days=first_day_of_week - 6,
                weeks=1)) | timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
        else:
          - condition: trigger
            id: last_month
          - data:
              topic: picframe/date_from
              payload: >-
                {% set day_of_month = (now().strftime("%d")) | int %}  {{
                as_timestamp(now() - timedelta(days=day_of_month)) |
                timestamp_custom("%Y/%m/01")}}
            action: mqtt.publish
          - data:
              topic: picframe/date_to
              payload: >-
                {% set day_of_month = (now().strftime("%d")) | int %}  {{
                as_timestamp(now() - timedelta(days=day_of_month)) |
                timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
mode: single

```

helgeerbe, 2025-05-31:
if you delete the data in date from and date to in home assistant, it should be resetted to min and max date

sapnho, 2025-06-07:
> Just checked my installation. Tag, directory, location filtering works in home assistant. What's not working is the date filter. I also have additional buttons for today, last week, and last month. But they don't work. Have to check next week. I did updates on home assistant recently.

The date filtering in Home Assistant works fine on my end. I just can't get the shortcuts to work...

helgeerbe, 2025-06-07:
First check, that under helpers, you have the input_button defined.
[image]

Second check your dashboard. Are the correct helpers referencenced?

[image]

This is my filter yaml-code:

```yaml
type: vertical-stack
cards:
  - type: entities
    entities:
      - entity: sensor.picframe_image_counter
        name: Bilder
      - entity: select.picframe_directory
        name: Directory
      - entity: input_text.picframe_tags_filter
        name: Stichwortsuche
      - entity: input_text.picframe_location_filter
        name: Ortssuche
      - entity: input_text.picframe_date_from
        name: Bilder ab (TT.MM.JJJJ)
      - entity: input_text.picframe_date_to
        name: Bilder bis (TT.MM.JJJJ)
    show_header_toggle: false
    theme: Backend-selected
    title: Filter
    state_color: true
  - show_name: false
    show_icon: true
    show_state: false
    type: glance
    entities:
      - entity: input_button.picframe_date_today
        tap_action:
          action: toggle
      - entity: input_button.picframe_date_last_week
        tap_action:
          action: toggle
      - entity: input_button.picframe_date_last_month
        tap_action:
          action: toggle
    state_color: true
```

Last check your automation. `payload_template` changed to `payload`

```yaml
alias: picframe_quick_date_select
description: ""
triggers:
  - entity_id:
      - input_button.picframe_date_today
    id: today
    trigger: state
  - entity_id:
      - input_button.picframe_date_last_week
    id: last_week
    trigger: state
  - entity_id:
      - input_button.picframe_date_last_month
    id: last_month
    trigger: state
conditions: []
actions:
  - if:
      - condition: trigger
        id: today
    then:
      - data:
          topic: picframe/date_to
          payload: "{{ now().strftime(\"%Y/%m/%d\") }}"
        action: mqtt.publish
      - data:
          topic: picframe/date_from
          payload: "{{ now().strftime(\"%Y/%m/%d\") }}"
        action: mqtt.publish
    else:
      - if:
          - condition: trigger
            id: last_week
        then:
          - data:
              topic: picframe/date_from
              payload: >-
                {% set first_day_of_week = (now().strftime("%w")) | int %} {{
                as_timestamp(now() - timedelta(days=first_day_of_week, weeks=1))
                | timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
          - data:
              topic: picframe/date_to
              payload: >-
                {% set first_day_of_week = (now().strftime("%w")) | int %} {{
                as_timestamp(now() - timedelta(days=first_day_of_week - 6,
                weeks=1)) | timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
        else:
          - condition: trigger
            id: last_month
          - data:
              topic: picframe/date_from
              payload: >-
                {% set day_of_month = (now().strftime("%d")) | int %}  {{
                as_timestamp(now() - timedelta(days=day_of_month)) |
                timestamp_custom("%Y/%m/01")}}
            action: mqtt.publish
          - data:
              topic: picframe/date_to
              payload: >-
                {% set day_of_month = (now().strftime("%d")) | int %}  {{
                as_timestamp(now() - timedelta(days=day_of_month)) |
                timestamp_custom("%Y/%m/%d")}}
            action: mqtt.publish
mode: single

```

closed 2025-06-08 by helgeerbe

sapnho, 2025-06-08:
I made a post about this: https://www.thedigitalpictureframe.com/how-to-add-a-date-filtering-shortcut-button-in-home-assistant-for-your-picture-frame/