helgeerbe/picframe #487: Add ability to turn video support off

ylexus, 2025-06-01:
While support for video playing is still under development/testing, it would be extremely handy to be able to make picframe skip all videos via a configuration setting, without patching any code. Currently I have to patch a source file to sent the list of supported video extensions to an empty value, which does disable videos, but I suspect causes strange behaviour where sometimes the picframe jumps back to the image it just displayed, while showing the wrong title belonging to another image. Also this means I need to reapply this patch on every update.

ylexus, 2025-06-28:
It would be super0helpful to have a way to disable videos. Commenting out extensions no longer works and likely causes this:

```
Jun 28 15:37:17 pframe labwc[942307]: WARNING:get_image_meta.GetImageMeta:Can't open file: "/var/storage/data/google-photos/Takeout/incremental/2025/2025-04/IMG_5890.mov"
Jun 28 15:37:17 pframe labwc[942307]: WARNING:get_image_meta.GetImageMeta:Cause: cannot identify image file '/var/storage/data/google-photos/Takeout/incremental/2025/2025-04/IMG_5890.mov'
Jun 28 15:37:17 pframe labwc[942307]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.util.FixedString.FixedString object at 0x7fff286e8e50>
Jun 28 15:37:17 pframe labwc[942307]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x7fff286e9290>
Jun 28 15:37:17 pframe labwc[942307]: DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x7fff2a89f710>
Jun 28 15:37:17 pframe labwc[942307]: Traceback (most recent call last):
Jun 28 15:37:17 pframe labwc[942307]:   File "/home/pi/venv_picframe/bin/picframe", line 8, in <module>
Jun 28 15:37:17 pframe labwc[942307]:     sys.exit(main())
Jun 28 15:37:17 pframe labwc[942307]:              ^^^^^^
Jun 28 15:37:17 pframe labwc[942307]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/start.py", line 137, in main
Jun 28 15:37:17 pframe labwc[942307]:     c.loop()
Jun 28 15:37:17 pframe labwc[942307]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/controller.py", line 332, in loop
Jun 28 15:37:17 pframe labwc[942307]:     (loop_running, skip_image, video_playing) = self.__viewer.slideshow_is_running(pics, time_delay, fade_time, self.__paused)
Jun 28 15:37:17 pframe labwc[942307]:                                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 28 15:37:17 pframe labwc[942307]:   File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py", line 647, in slideshow_is_running
Jun 28 15:37:17 pframe labwc[942307]:     wh_rat = (self.__display.width * self.__sfg.iy) / (self.__display.height * self.__sfg.ix)
Jun 28 15:37:17 pframe labwc[942307]:                                      ^^^^^^^^^^^^^
Jun 28 15:37:17 pframe labwc[942307]: AttributeError: 'NoneType' object has no attribute 'iy'
``z

ylexus, 2025-07-13:
leaving comment to un-stale, I am actively watching this issue

closed 2025-08-12 by github-actions[bot]

ylexus, 2025-08-12:
Reviving the issue - still interested

helgeerbe, 2025-08-12:
In the actual refactoring I added a `"media_type_filter": "all",  # one of: all, image, video` in the model section.
This filter could later also be used e. g. in home assistant to filter on the fly for media types.

But I need some more time, as I do a complete refactoring of the actual code base.