helgeerbe/picframe #496: Pictureframe  - black screen after reboot

Helforge, 2025-06-27:
Hi,

I followed the tutorial "Install the Pi3D PictureFrame software with one click (2025 Edition, Raspberry Pi 2/3/4/5)"
on thedigitalpictureframe.com and everything worked fine until i unplugged it last night and plugged it back in this morning,

Now i'm just getting a black screen with a mouse indicator. When i connect a mouse and i click it says:

* Reconfigure (which does nothing if i click on it)
* Exit (which brings me to the main screen and automatically back to the black screen again in about 1 second).

I tried the following:

- Rebooting the Rasberry PI.
- Power cycling.
- Changed power supply
- Booted without the screen and then connected the screen.
- Tried to start the picframe service with. (systemctl --user start picframe.service)
- Tried to restart the pictframe service with  (systemctl --user restart picframe.service)

Hardware:
- Raspberry PI 3 B
-  32 GB SD
- Powersupply from Raspberry
- Ethernet connection (no Wifi used)

Could someone tell me what i am doing wrong?

A little background before anyone answers:
This is my first Raspberry PI project, and since there was a guide @thedigitalpictrueframe.com i started with this as my first project.
I have no knowledge of Python and a very basic understanding of programming, so i know this was going to be a bit of a challenge for me :-)

paddywwoof, 2025-06-27:
I'm thinking maybe unplugging could have been the cause. The building of the database might have been interrupted mid write (maybe?) I would try deleting the database file, check whereabouts but from memory something like `picframe_data/data/pictureframe.db3`It is a newish release so there may be a bug, but I have been running it in a couple of systems for a few days OKPaddy

Helforge, 2025-06-27:
Hi Paddy,

Thank you for the quick response!
The digital pictureframe was working as intended before i unplugged it (i had already added some of my own pictures and it was displaying them). Shouldn’t the database already be completed by then?

If i just delete the database, does it make a new one automatically or do i need to do something myself?

Helforge

closed 2025-06-29 by Helforge

paddywwoof, 2025-06-29:
Good. Hopefully that was the issue. It might be worth having a note in the installation instructions on this subject (probably already is one as Wolfgang's write-up is amazingly comprehensive)