helgeerbe/picframe #497: change_picture_time is ignored + caption display

RicardoSwe2, 2025-07-03:
Fairly new here.  I have successfully installed picframe on my raspbery pi 5. This took may rounds to get it working. Some issues cannot be solved -- such as:

**change_picture_time** - this is ignored. I cannot control the picture time. It seems to default to more than 2 minutes. I cannot get a shorter display time.
By far this is the varialble I'd like to be able to control. I just cannot.

**caption display** - this seems rather weird. I cannot get a caption shorter than:
(as best I can make this out...)
jpg file name-jpg date of origin-file folder
This ends up displaying the year twice. Don't want that but cannot see how to control this.

For example, one photo displays as "Ital 2021-340.jpg * Oct 29, 2021*Italy"
I'd prefer it just show: "file folder - file title.jpg Oct 29, 2021"

paddywwoof, 2025-07-03:
@RicardoSwe2, it might be hard to debug without seeing how your setup has changed from the 'vanilla' version described in Wolfgang's one click system. But I will try to give some pointers:

The delay between subsequent pictures is controlled by the `time_delay: 200.0` setting in the `configuration.yaml` however things that can impact that are a) something else slowing the computer down while it's loading the file into memory and resizing and adding the frame etc. The resizing etc can be what slows the computer down if the images are very large, for instance mobile phone images can *much*, *much* larger than most monitors. There are suggestions for code to do resizing of images on www.thedigitalpictureframe.com I think, if that turns out to be a likely cause. b) the fading `fade_time: 10.0` and text display time `show_text_tm: 20.0` will need to be completed before the next picture comes up, so if those times are long the `time_delay` will get extended.

The captions are controlled in `configuration.yaml` by `show_text: "title caption name date folder location"` where `title` and `caption` are extracted from the image exif data (I think, I never use those!) `name` is the file name of the image, `date` is either extracted from the exif, or if there's nothing there is uses the file system date, `folder` is the subdirectory of the image stored in your Pictures directory and `locataion` gets the lat, lon from exif and looks up where that is. You might have to rename your images if their names currently contain information you don't want to be duplicated, such as the year and country. I can't see any way for pictureframe to solve that without a cludgy hack to the code.

Autostart is done in Wolfgang's script by a two stage process. First `systemd` is used to automatically `ExecStart=/usr/bin/labwc` and then `labwc` is set to automatically run `start_picframe.sh` when it starts (OK and that runs the *third* stage which is starting up the python virtual environment and running picframe!!) When I'm wanting to mess around with picframe and stop and restart it I usually open a new terminal on the RPi (either SSL or Ctrl+Alt+F4 on  a plugged in keyboard) then `systemctl --user stop picframe.service` and later restart by swapping `stop` for `start`. If you want to disable the autostart of the process use `systemctl --user disable picframe.service` and re-enable it with `enable`, you might be able to `start` a process that hasn't been `enable`d but it shouldn't start on boot.

Brittle. Hmm ChatGPT calling the kettle black? But fair enough picframe isn't a professionally written bit of software with a massive software house behind it - it's possible to break things by changing settings. Normally though there is a very sensible explanation and if you post the errors you get before changing yet more variables, one of us might be able to point you in the right direction.

Paddy

RicardoSwe2, 2025-07-03:
Paddy,
Thank you. Good explanations and ideas for paths forward. I'll try some of  your ideas. I think my impatience is often my worst enemy in things like this. 

Part of my frustration with PicFrame -- **_which I do like a lot_** -- is the lack of explanations like yours about the background info on delay between pictures. I never saw those causes explained elsewhere.

I agree with you about ChatGPT calling the kettle black. Interesting. However, I went there because ChatGPT was suggested to me by an authority on PicFrame to resolve some of these issues. ChatGPT certainly seems starts off sounding authoritative, eh? Part of the problem. I thought brittle was a good description.

Richard